emsesp/EMS-ESP32 #1929: Unexpected Application Error

mattreim, 2024-08-06:
### PROBLEM DESCRIPTION

Sometimes this error message appears when changing tabs:

[image]

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{
  "System": {
    "version": "3.7.0-dev.30",
    "uptime": "000+02:00:24.802",
    "uptime (seconds)": 7224,
    "platform": "ESP32",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "free mem": 154,
    "max alloc": 87,
    "free caps": 105,
    "used app": 1676,
    "free app": 308,
    "partition": "app0",
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network": {
    "network": "WiFi",
    "hostname": "EMS-ESP",
    "RSSI": -54,
    "BSSID": "set",
    "TxPower setting": 0,
    "static ip config": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": false,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP": {
    "NTP status": "connected",
    "enabled": true,
    "server": "192.168.178.1",
    "tz label": "Europe/Berlin"
  },
  "MQTT": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client id": "esp32-422a13d8",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 2,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 0,
    "publish time thermostat": 0,
    "publish time solar": 0,
    "publish time mixer": 0,
    "publish time water": 0,
    "publish time other": 0,
    "publish time sensor": 0,
    "publish single": true,
    "publish2command": true,
    "send response": false
  },
  "Syslog": {
    "enabled": false
  },
  "Sensor": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API": {
    "API calls": 776,
    "API fails": 0
  },
  "Bus": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 14707,
    "bus reads (tx)": 2790,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 21,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "CUSTOM",
    "locale": "de",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower min duration": 120,
    "shower alert": false,
    "phy type": 0,
    "rx gpio": 4,
    "tx gpio": 5,
    "dallas gpio": 14,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 6,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 25,
    "web log buffer": 25
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Logamax plus GB172i.2-15",
      "device id": "0x08",
      "product id": 195,
      "version": "05.06",
      "entities": 60,
      "handlers received": "0xBF 0xC2 0x14 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x2E 0x3B 0x04",
      "handlers fetched": "0xE6 0xEA 0x28",
      "handlers pending": "0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0xE7 0x02E0 0x02EA 0xD7 0xD8 0x02D6 0x029B"
    },
    {
      "type": "thermostat",
      "name": "Logamatic BC400",
      "device id": "0x10",
      "product id": 4,
      "version": "49.04",
      "entities": 42,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E 0xBB 0x023E 0x0269",
      "handlers ignored": "0xC4 0x047B 0xF7 0xBF 0x3B"
    },
    {
      "type": "thermostat",
      "name": "RC100H",
      "device id": "0x38",
      "product id": 200,
      "version": "40.04",
      "entities": 3,
      "handlers received": "0x042B 0x047B",
      "handlers pending": "0x0273 0x0A6A"
    }
  ]
}
```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2024-08-06:
Yes, known issue. Is fixed in another issue

mattreim, 2024-08-06:
Should I close the issue?

proddy, 2024-08-06:
I'll close it. The psram is the last point to fix in https://github.com/emsesp/EMS-ESP32/issues/1920 but it's waiting on some other 3rd party library to fix something before I can merge

closed 2024-08-06 by proddy

mattreim, 2024-08-06:
OK, thanks