emsesp/EMS-ESP32 #1932: HA dhw control is not working since 3.7.0-dev.30 [BC400]

Bingo2023, 2024-08-11:
In Home Assistance the climate control is not working anymore.

It always shows off state (see screenshot below). It should say "heating" because cooling is on.

[image]

SW: 3.7.0-dev.30
HP with BC400

Hint: was working w/ 3.7.0-dev.23
only changes I see below:
[image]

PS: nice improvement would be: to differentiate between heating and cooling.

proddy, 2024-08-11:
could you post the value of the MQTT topic `homeassistant/climate/ems-esp/thermostat_hc1/config` or similar. Look for climate (if using MQTT Explorer). We need to see if the modes are correct.

Bingo2023, 2024-08-11:
[image]

complete:
`{"~":"ems-esp","uniq_id":"thermostat_hc1","obj_id":"thermostat_hc1","name":"Hc1","mode_stat_t":"~/thermostat_data","mode_stat_tpl":"{%if value_json.hc1 is undefined or value_json.hc1.mode is undefined%}off{%elif value_json.hc1.mode=='Manuell'%}heat{%elif value_json.hc1.mode=='Tag'%}heat{%elif value_json.hc1.mode=='Nacht'%}off{%elif value_json.hc1.mode=='aus'%}off{%else%}auto{%endif%}","temp_cmd_t":"~/thermostat/hc1/seltemp","temp_stat_t":"~/thermostat_data","temp_stat_tpl":"{{value_json.hc1.seltemp if value_json.hc1 is defined and value_json.hc1.seltemp is defined else 0}}","curr_temp_t":"~/thermostat_data","curr_temp_tpl":"{{value_json.hc1.currtemp if value_json.hc1 is defined and value_json.hc1.currtemp is defined else 0}}","min_temp":"5","max_temp":"30","temp_step":"0.5","mode_cmd_t":"~/thermostat/hc1/mode","modes":["auto","heat","off"],"dev":{"name":"ems-esp Thermostat","ids":["ems-esp-thermostat"]},"avty":[{"t":"~/thermostat_data","val_tpl":"{{'online' if value_json.hc1 is defined and value_json.hc1.seltemp is defined else 'offline'}}"},{"t":"~/thermostat_data","val_tpl":"{{'online' if value_json.hc1 is defined and value_json.hc1.currtemp is defined else 'offline'}}"},{"t":"~/thermostat_data","val_tpl":"{{'offline' if value_json.hc1 is undefined or value_json.hc1.mode is undefined else 'online'}}"}],"avty_mode":"all"}`

proddy, 2024-08-11:
And please post also the acutal value from the MQTT topic `ems-esp/thermostat_data`. Looking for `hc1.mode`

Bingo2023, 2024-08-11:
it is off, but for BC400 we have to use **"hc1.coolingon"**
[image]
[image]

complete message:
`{"datetime":"11.08.2024 16:43","intoffset":0.0,"floordry":"aus","dampedoutdoortemp":28.3,"floordrytemp":0,"building":"Leicht","minexttemp":-12,"damping":"an","energycostratio":0.0,"pvenabledhw":"an","pvraiseheat":2,"pvlowercool":0,"hc1":{"seltemp":13.0,"currtemp":27.2,"haclimate":"Raumtemperatur","mode":"aus","modetype":"Komfort","ecotemp":19.0,"manualtemp":0.0,"comforttemp":21.0,"summertemp":16,"designtemp":35,"offsettemp":0,"minflowtemp":22,"maxflowtemp":55,"roominfluence":0,"roominflfactor":0.0,"curroominfl":0.0,"nofrostmode":"Außen","nofrosttemp":5,"targetflowtemp":10,"heatingtype":"Fussboden","hpoperatingmode":"Kühlen","hpoperatingstate":"Kühlen","program":"prog 1","tempautotemp":-1.0,"remoteseltemp":13.0,"fastheatup":0,"switchonoptimization":"aus","reducemode":"reduziert","noreducetemp":-31,"reducetemp":0,"dhwprio":"aus","cooling":"an","coolingon":"an","hpmode":"Heizen&Kühlen","dewoffset":2,"roomtempdiff":1,"hpminflowtemp":10,"control":"RC220/RT800","remotetemp":null,"remotehum":null,"heatondelay":1,"heatoffdelay":1,"instantstart":2,"boost":"aus","boosttime":1,"coolstart":23,"coolondelay":1,"cooloffdelay":18,"switchprogmode":"Level","rftemp":27.2,"dewtemperature":2.2,"airhumidity":20},"dhw":{"mode":"Komfort","circmode":"aus","chargeduration":60,"charge":"aus","extra":0,"disinfecting":"an","disinfectday":"Di","disinfecttime":660,"dailyheating":"aus","dailyheattime":120}}`

Bingo2023, 2024-08-11:
check also related ticket:
https://github.com/emsesp/EMS-ESP32/issues/1398

depending to what mode is set at HP the mode looks different (pics from Buderus App (MX300 Interface):
[image]
[image]

Bingo2023, 2024-08-11:
I think we need to adapt more for BC400, my feeling is that 20% of the signals are not fitting for the new BC400 systems.
See summer mode: https://github.com/emsesp/EMS-ESP32/issues/1904

As I understand the system, "summer/winter" also called "Betriebsart" (operating mode?) is used to activate cooling or heating functionality (or "auto" to switch on temperature conditions).
Mode selection for heating and cooling can be set independently of above setting.
--> therefor we should use the status of "summer/winter" to differentiate between heating & cooling.

[image]

Bingo2023, 2024-09-01:
the file system of my ESP32 was damaged/not compatible. OTA updates worked and everything looks OK, but the SW was doing unexpected things like not sending some messages randomly.

I fixed it by a fresh format with EMS-ESP-Flasher v1.0.3. tool.
It is running stable since some days now. Sorry for that confusion.

DHW is working but only shows heating, since summer/winter signal is not present in dev30 see https://github.com/emsesp/EMS-ESP32/issues/1904

my posted photos are valid, we could check this again once the summer/winter signal works again.

proddy, 2024-09-01:
Shall we close this one (which is about dhw control) and create a new issue, or re-open 1904 for the summer mode and more BC400 settings?

closed 2024-10-31 by proddy