emsesp/EMS-ESP32 #1935: RTC WAtchdog reboots and strange system/send with NAN

harvey637, 2024-08-14:
### PROBLEM DESCRIPTION

_A clear and concise description of what the problem is._

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

harvey637, 2024-08-14:
Since some versions I experience ems-esp reboots and strange sytem/send every few seconds.

Log says:
```
2024-08-14 07:59:40.000 INFO 1: [emsesp] Starting EMS-ESP version 3.7.0-dev.30 from app0 partition
2024-08-14 07:59:40.000 INFO 2: [emsesp] Last system reset reason Core0: RTC watch dog reset: CPU+RTC, Core1: APP CPU reset by PRO CPU
2024-08-14 07:59:40.000 INFO 3: [emsesp] Loading board profile CUSTOM
2024-08-14 07:59:40.000 INFO 4: [emsesp] Starting Scheduler service
2024-08-14 07:59:40.000 INFO 5: [emsesp] Starting Custom Entity service
2024-08-14 07:59:40.000 INFO 6: [emsesp] Starting Modules service
2024-08-14 07:59:40.000 INFO 7: [module] No external library modules found
2024-08-14 07:59:40.000 INFO 8: [emsesp] Loaded EMS device library (120 records)
2024-08-14 07:59:43.069 INFO 9: [emsesp] WiFi connected (IP=192.168.23.95, hostname=ems-esp, TxPower=19.3 dBm)
2024-08-14 07:59:43.096 INFO 10: [emsesp] Starting mDNS Responder service
2024-08-14 07:59:43.096 INFO 11: [emsesp] Starting NTP service
2024-08-14 07:59:47.807 INFO 12: [emsesp] Recognized new boiler with deviceID 0x08
2024-08-14 07:59:47.807 INFO 13: [emsesp] Recognized new thermostat with deviceID 0x10
2024-08-14 07:59:47.807 INFO 14: [emsesp] Recognized new thermostat with deviceID 0x38
2024-08-14 07:59:47.807 INFO 15: [emsesp] Recognized new gateway with deviceID 0x48
2024-08-14 08:00:07.454 INFO 16: [thermostat] ems-esp time set from thermostat
2024-08-14 08:00:46.078 INFO 17: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 13 01 01 BA 
2024-08-14 08:01:15.350 INFO 18: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 13 01 01 BA 
2024-08-14 08:01:27.636 INFO 19: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 13 01 01 BA 
2024-08-14 08:01:38.697 INFO 20: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 12 01 6A 
2024-08-14 08:01:46.074 INFO 21: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 12 01 6A 
2024-08-14 08:02:04.503 INFO 22: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 12 01 6A 
2024-08-14 08:02:16.789 INFO 23: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 12 01 6A 
2024-08-14 08:02:27.848 INFO 24: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 12 01 6C 
2024-08-14 08:02:35.227 INFO 25: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 12 01 6C 
2024-08-14 08:02:53.658 INFO 26: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 12 01 6C 
2024-08-14 08:03:05.945 INFO 27: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 12 01 6C 
2024-08-14 08:03:17.001 INFO 28: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 FF 01 C4 
2024-08-14 08:03:24.376 INFO 29: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 FF 01 C4 
2024-08-14 08:03:42.604 INFO 30: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 FF 01 C4 
2024-08-14 08:03:54.889 INFO 31: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 FF 01 C4 
2024-08-14 08:04:05.960 INFO 32: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 FF 03 4A 
```

This is somethin I cannot remember a few weeks ago. Can reproduce (just sitting waitung ...) at least with dev-29 and dev-30.

harvey637, 2024-08-14:
Some more detailed log:

```
2024-08-14 08:21:46.240 TRACE 244: [emsesp] Me(0x0B) -R-> boiler(0x08), HPSilentMode(0x0484), length: 230 (offset 25)
2024-08-14 08:21:46.277 TRACE 245: [emsesp] boiler(0x08) -W-> Me(0x0B), HPSilentMode(0x0484), data: 01 90 00 F6 00 00 64 00 00 E1 00 00 00 00 0A 64 00 64 54 20 01 01 00 00 23 (offset 25)
2024-08-14 08:21:46.526 TRACE 246: [emsesp] Me(0x0B) -R-> boiler(0x08), HPSilentMode(0x0484), length: 205 (offset 50)
2024-08-14 08:21:46.530 TRACE 247: [emsesp] boiler(0x08) -W-> Me(0x0B), HPSilentMode(0x0484), data: 54 14 00 00 00 00 00 (offset 52)
2024-08-14 08:21:46.751 TRACE 248: [emsesp] Me(0x0B) -R-> boiler(0x08), HPSilentMode(0x0484), length: 196 (offset 59)
2024-08-14 08:21:46.755 TRACE 249: [emsesp] boiler(0x08) -W-> Me(0x0B), HPSilentMode(0x0484), data: <empty> (offset 59)
2024-08-14 08:21:46.755 TRACE 250: [emsesp] Me(0x0B) -R-> boiler(0x08), HPPumps(0x048B), length: 255
2024-08-14 08:21:46.825 DEBUG 251: [command] Called command system/response (Hole Antwort)
2024-08-14 08:21:46.825 TRACE 252: [emsesp] boiler(0x08) -W-> Me(0x0B), HPPumps(0x048B), data: 00 0A 1E 46 00 1E 01 2C 00 01 64 55 05 1F 00 38 00 00 00 1E 01 2C 00 (offset 1)
2024-08-14 08:21:46.863 DEBUG 253: [command] Called command system/response (Hole Antwort)
2024-08-14 08:21:46.904 INFO 254: [command] Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 FF 02 09 
2024-08-14 08:21:47.041 TRACE 255: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300WW2mode(0x02F6), length: 1 (offset 11)
2024-08-14 08:21:47.044 TRACE 256: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300WW2mode(0x02F6), data: <empty> (offset 11)
2024-08-14 08:21:47.124 DEBUG 257: [command] Called command system/response (Hole Antwort)
2024-08-14 08:21:47.147 TRACE 258: [emsesp] Me(0x0B) -B-> All(0x00), Version(0x02), data: 09
2024-08-14 08:21:47.569 TRACE 259: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300WW2mode(0x02F6), length: 1 (offset 10)
2024-08-14 08:21:47.572 TRACE 260: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300WW2mode(0x02F6), data: <empty> (offset 10)
2024-08-14 08:21:47.607 TRACE 261: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300WW2mode(0x02F6), length: 1 (offset 5)
2024-08-14 08:21:47.616 TRACE 262: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300WW2mode(0x02F6), data: <empty> (offset 5)
2024-08-14 08:21:47.726 TRACE 263: [emsesp] Me(0x0B) -R-> boiler(0x08), HPPumps(0x048B), length: 231 (offset 24)
2024-08-14 08:21:47.730 TRACE 264: [emsesp] boiler(0x08) -W-> Me(0x0B), HPPumps(0x048B), data: 00 1E 00 96 00 32 (offset 24)
2024-08-14 08:21:47.730 TRACE 265: [emsesp] Me(0x0B) -R-> boiler(0x08), HPAdditionalHeater(0x0491), length: 255
2024-08-14 08:21:47.798 TRACE 266: [emsesp] boiler(0x08) -W-> Me(0x0B), HPAdditionalHeater(0x0491), data: 00 00 00 00 00 0A 64 00 00 04 01 2C 00 0A 00 01 00 1E 00 00 1E 0A
2024-08-14 08:21:47.798 TRACE 267: [emsesp] Me(0x0B) -R-> boiler(0x08), HPDhwSettings(0x0499), length: 255
2024-08-14 08:21:47.825 TRACE 268: [emsesp] boiler(0x08) -W-> Me(0x0B), HPDhwSettings(0x0499), data: 3C 3F 3F 3B 00 37
2024-08-14 08:21:47.825 TRACE 269: [emsesp] Me(0x0B) -R-> boiler(0x08), HPSettings2(0x049C), length: 255
2024-08-14 08:21:47.869 TRACE 270: [emsesp] boiler(0x08) -W-> Me(0x0B), HPSettings2(0x049C), data: 00 00 00 00 00
2024-08-14 08:21:47.869 TRACE 271: [emsesp] Me(0x0B) -R-> boiler(0x08), HPSettings3(0x049D), length: 255
2024-08-14 08:21:47.911 DEBUG 272: [command] Called command system/response (Hole Antwort)
2024-08-14 08:21:47.911 TRACE 273: [emsesp] boiler(0x08) -W-> Me(0x0B), HPSettings3(0x049D), data: 00 00 00 00 00 00 00 00 00 00 00 00
2024-08-14 08:21:47.911 TRACE 274: [emsesp] Me(0x0B) -R-> boiler(0x08), HPPressure(0x02CC), length: 255
2024-08-14 08:21:47.961 DEBUG 275: [command] Called command system/response (Hole Antwort)
2024-08-14 08:21:47.961 TRACE 276: [emsesp] boiler(0x08) -W-> Me(0x0B), HPPressure(0x02CC), data: 00 (offset 3)
2024-08-14 08:21:48.121 TRACE 277: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300WW2mode(0x02F6), length: 1 (offset 7)
2024-08-14 08:21:48.124 TRACE 278: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300WW2mode(0x02F6), data: <empty> (offset 7)
2024-08-14 08:21:48.151 DEBUG 279: [command] Called command system/response (Hole Antwort)
2024-08-14 08:21:48.289 TRACE 280: [emsesp] Me(0x0B) -R-> boiler(0x08), HPPressure(0x02CC), length: 251 (offset 4)
2024-08-14 08:21:48.295 TRACE 281: [emsesp] boiler(0x08) -W-> Me(0x0B), HPPressure(0x02CC), data: <empty> (offset 4)
2024-08-14 08:21:48.295 TRACE 282: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 255
2024-08-14 08:21:48.394 TRACE 283: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 ED 61 28 00 00 28 2A 05 A0 03 03 03 05 A0 05 A0 00 00 11 02 03 09 42 00
2024-08-14 08:21:48.553 TRACE 284: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300WW2mode(0x02F6), length: 1 (offset 6)
2024-08-14 08:21:48.556 TRACE 285: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300WW2mode(0x02F6), data: <empty> (offset 6)
2024-08-14 08:21:48.639 TRACE 286: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 230 (offset 25)
2024-08-14 08:21:48.685 TRACE 287: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 06 08 00 00 00 00 11 00 30 00 3C 01 3C (offset 25)
2024-08-14 08:21:48.685 TRACE 288: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Set(0x02B9), length: 255
[... cut ...]
2024-08-14 08:21:51.512 DEBUG 347: [command] Called command system/response (Hole Antwort)
2024-08-14 08:21:52.342 DEBUG 348: [command] Called command system/response (Hole Antwort)
2024-08-14 08:21:52.377 DEBUG 349: [command] Called command system/response (Hole Antwort)
2024-08-14 08:21:52.743 DEBUG 350: [command] Called command system/response (Hole Antwort)
2024-08-14 08:21:53.090 TRACE 351: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 00 00 CD 00 00 00 00 02 B2 00 00 00 D0 00 00 00 00 00 CF 00 CD
```

proddy, 2024-08-14:
Max Alloc looks very low (51). Also any idea why all those calls to the system/response command?

harvey637, 2024-08-14:
Hi Proddy,
great to hear from you!

I have now idea of the origin of the system/response commands. Can you easy read what this command is?

With the iobroker/ems-esp interface I have 300 secs for actualisation, no energy statiscics, no km200, no heatdemand.
From raspberrymatic I do a single call to api/boiler and api/thermostat every 5 mins ...

thank you!

MichaelDvP, 2024-08-14:
@proddy This is a ESP32 4M. I think i set the scheduler to sync for versions without PSRAM, this gives ~5-8kB free ram and on these chips we need every kB. For psram versions we can leave scheduler async and also set mqtt to async.

@tp1de Can you look at this, seems the system/send system/response is from ioBroker-adapter. I do not see this on RC35, maybe you automatic activate it for RC300? Maybe for the timer programs.

MichaelDvP, 2024-08-14:
@harvey637 Have you enabled in ioBroker adapter "enable search for additional EMS+ entities (switchTimes & holidayModes) - write access needed"?

harvey637, 2024-08-14:
Hi, 
when speaking of timer programs ... is this the reading of the prog_1 and prog_2 from the thermostat?
I remember, that at some time in history i checked this and found an array ith my switching timestamps.
But now I dont see this anymore .... gone? not filled by this NAN request?

harvey637, 2024-08-14:
YES

tp1de, 2024-08-14:
> @tp1de Can you look at this, seems the system/send system/response is from ioBroker-adapter. I do not see this on RC35, maybe you automatic activate it for RC300? Maybe for the timer programs.

@harvey637 have you selected the search for additional EMS+ entities within ioBroker adapter?
If yes, please deactivate since you have 2 thermostats

harvey637, 2024-08-14:
Hmmmmmm,
I have physically one thermostat (device), the RC100H in my living room.
The other "thermostat" is/looks like an rc310 which in my opinion is the integrated "console" in my bosch 7000 ... heatpump.
Flapping down the cover, reveals a smal LCD, wher I see/modify setting of my heatpump, actually this whole devide is aclled AMWB (looks like a big refrigirator) as I remember, not quite shure

tp1de, 2024-08-14:
I am not happy with the actual solution of sending raw telegrams from ioBroker adapter and trying to read response for holidayModes and switchPrograms.
We discuss other solutions in #1594

harvey637, 2024-08-14:
ok, I can switch OFF the additional search (for now), but is WAS working!

harvey637, 2024-08-14:
ok, the log entries  like
```
2024-08-14 10:40:22.093    INFO 2014:   [command]    Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 FF 01 C4
2024-08-14 10:40:22.120    INFO 2015:   [command]    Called command system/send (Sende EMS-Telegramm) with value 0B NAN FF 00 FF 02 09 
```
are gone.

tp1de, 2024-08-14:
You have an gateway as well. Is km200 search not working within ioBroker?

harvey637, 2024-08-14:
I started long time ago with no km200 structure. So now I am afraid of loosing my historical data, which is sometimes quite interesting eg. in relation to the transition from heating to cooling.
So the gateway never had any entities, so i ignored it. I thougt it as an internal bosch gateway pushing the data into their easycon cloud.

tp1de, 2024-08-14:
You can try to read by a 2nd instance and just activate km200 there (no ems-esp).
It depends, what kind of gateway is installed. You need to set a private password with the MyDevice App first.
If this is working you will get next to holidayModes and switchPrograms  recordings as well (12 months energy statistics)

harvey637, 2024-08-14:
I will give it a try, thank you

harvey637, 2024-08-14:
Hmmmmm, i did not use the Bosch HomeCom for some years (no need for it), now it has no connection to my  devices. I have to solve this first, then I will check out the second instance.
thank you 
Harvey

tp1de, 2024-08-14:
You do not need HomeCom which is cloud based. Bosch has change the login recently ...
You need the MyDevice App which connects locally (LAN) and for this you need to set a private password.
This pw together with the gateway ps is used to decrypt api messages.

MichaelDvP, 2024-08-14:
Just a last question before you disconnect the ems-esp: Are the WDT resets still happen after disable the raw reads from adapter?

tp1de, 2024-08-14:
@harvey637 do not disconnect EMS-ESP since the gateway supports much more entities compared to original gateway and can be polled much faster (up to 15 secs). But do not use the extended search.

Big advantage of original gateway are recordings, switchPrograms (until in ems-esp) and holidayPrograms.
For me the combination of both is the best solution.

tp1de, 2024-08-14:
> I already switched instance.0 to km200, all my historical data is still there - wow!

If you do not enable the parameter to delete states (or do it manually) then you might get double entries.
The delete states parameter will not delete states with history / db entries but the object names are now changed.

harvey637, 2024-08-14:
Thank you for the answer, i guess you you wanted to write: "do not disconnect EMS-ESP since the ems-esp-gateway supports much more entities compared to original build-in-gateway"
I will keep that in mind, but for now I am happy with 5 minutes intervals, my heating is floor heating, so 15 secs poll will only bloat data, may be not, as I only save different values, but I will see.

harvey637, 2024-08-14:
thank you for the warning, but i had the "delete" enabled, so no double entries.
With no heating in summer there will be little change in values, only outdoortemp and curtemp and like this.

If I run into (visable) trouble you will be the first to know! (by creating issue)
It is a pleasure to talk to you, gives me more insight and good ideas, what to try.
ciao
Harvey

tp1de, 2024-08-14:
I added an issue within the ioBroker adapter repo.
@harvey637 have a look in recordings (see pdf docu) - this might be interesting to see energy consumption

harvey637, 2024-08-14:
yep, just scrolling through the iobroker objects ..... nice infos.

closed 2024-09-01 by proddy