emsesp/EMS-ESP32 #1936: BC400 - ems-esp/boiler_data/hc1.hpactivity  is removed    3.7.0-dev.30 [BC400]

Bingo2023, 2024-08-14:
On My BC400 HeatPump I have an automation on the signal
MQTT: ems-esp/boiler_data/hc1.hpactivity             (hpactivity = compressor state)
in Version 3.7.0-dev.30 this signal does not exist. With dev.23 this signal is present.

I've checked if we reflect this in:
    ems-esp/boiler_data/hc1.hpoperatingmod": "Kühlen",
    ems-esp/boiler_data/hc1.hpoperatingstate: "Kühlen",

but these signal dont change if cooling is on/off. Buderus PK9 Cooling station is anyway a passive station, therefore is not reflected to compressorstatus, but warmwater or heating will be in hpactivity (= compressor state) visible.

Cooling is visible here:
ems-esp/boiler_data/hc1.coolingon: "an" or "off"

see also related ticked:
https://github.com/emsesp/EMS-ESP32/issues/1904

Should we just add "hpactivity" again or is it removed for some reason? Can we clean up signals for BC400 (e.g. combine hpactivity with hpoperatingstate)?
@proddy @MichaelDvP what is your recommendation, we have lots of dead Signals and it is difficult to find correct ones. But maybe this "cleaning" is difficult and there are good reasons just to just keep adding new signal....

Bingo2023, 2024-08-14:
edit: 17:50 (corrected message topic, just misstyped), thanky Michael :)

MichaelDvP, 2024-08-14:
> MQTT: ems-esp/boiler_data/hpactivity (hpactivity = compressor state)
in Version 3.7.0-dev.30 this signal does not exist. With dev.23 this signal is present.

There is really no change in boiler entities except removing setburnpow.
dev 23 was released 6.jul.24, dev24 on 7.jul.24, changes to boiler were only 29.jul..24 and 2.aug.24 with no effect on hpactivity:
https://github.com/emsesp/EMS-ESP32/commits/dev/src/devices/boiler.cpp

Bingo2023, 2024-08-14:
investigated, we had 2 Version of dev30

- 3.7.0-dev.30: disappeared but ~2 weeks before
- 3.7.0-dev.30: from 10.08. (replace & uploaded by @proddy)

first I used the old original dev30 and there, the message **ems-esp/boiler_data/hpactivity** was definitely missing.
I also experienced the MQQT disconnected issue, see [https://github.com/emsesp/EMS-ESP32/issues/1930].
I use standard 4mb uC.
file: EMS-ESP-3_7_0-dev_30-ESP32.bin
I had to reenter the MQTT connection parameters user & password.
Maybe there was, because of this some more corrupted in the flashing process. Seems like after some 10-20 software updates the file system gets messed up... we had some discussions about this topic some month ago.

proddy, 2024-08-15:
>  I also experienced the MQQT disconnected issue, see [https://github.com/[/issues/1930](https://github.com/emsesp/EMS-ESP32/issues/1930)]. I use standard 4mb uC. file: EMS-ESP-3_7_0-dev_30-ESP32.bin I had to reenter the MQTT connection parameters user & password. Maybe there was, because of this some more corrupted in the flashing process. Seems like after some 10-20 software updates the file system gets messed up... we had some discussions about this topic some month ago.

Arrghhh...the MQTT missing settings issue. It's so hard to reproduce. Can you remember if this happened after the EMS-ESP was restarted after a new Firmware update? Also, apart from the MQTT IP address, username and password were there any other settings defaulted back to their original values?

Bingo2023, 2024-08-15:
tough question, I needed some time to figure out the root cause of EMS-ESP didn't work correctly...
As I remember, MQTT IP/user/PW was missing exactly what you stated already.

proddy, 2024-09-01:
I think you just picked up an unlucky version of the old dev-30 for the hpactivity issue. The missing MQTT settings issue has a separate GitHub issue to track now.

closed 2024-09-01 by proddy

Bingo2023, 2024-09-01:
the file system of my ESP32 was damaged/not compatible. OTA updates worked and everything looks OK, but the SW was doing unexpected things like not sending some messages randomly.

I fixed it by a fresh format with EMS-ESP-Flasher v1.0.3. tool.
It is running stable since some days now. Sorry for that confusion.