emsesp/EMS-ESP32 #1937: Error A11 816 after updating to 3.7.0 dev

AzonInc, 2024-08-14:
### PROBLEM DESCRIPTION

_A clear and concise description of what the problem is._
After updating to the latest 3.7.0 builds I see Error A11 816 on the thermostat and the Condens display.
Bosch Support said that there is a pump or smth missing on the bus.

The error shows up after a while.

Reverting back to 3.6.5 solved the problem.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

{"system":{"version":"3.7.0-dev.30","uptime":"000+00:43:00.059","uptimeSec":2580,"platform":"ESP32-S3","arduino":"ESP32 Arduino v2.0.17","sdk":"v4.4.7-dirty","freeMem":194,"maxAlloc":179,"freeCaps":8336,"usedApp":2266,"freeApp":5862,"partition":"app0","resetReason":"Software reset CPU / Software reset CPU"},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-50,"BSSID":"set","TxPowerSetting":78,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"10.10.0.21","tzLabel":"Europe/Berlin"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":704,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTConnects":1,"enabled":true,"clientID":"ems-esp","keepAlive":60,"cleanSession":false,"entityFormat":3,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":true},"syslog":{"enabled":true,"syslogStarted":true,"syslogLevel":"error","syslogIP":"10.10.0.21","syslogQueue":0},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":24,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"HT3","busTelegramsReceived":3625,"busReads":777,"busWrites":0,"busIncompleteTelegrams":0,"busReadsFailed":1,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"S32S3","locale":"en","txMode":1,"emsBusID":11,"showerTimer":true,"showerMinDuration":180,"showerAlert":false,"hideLed":true,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":50,"webLogBuffer":50,"modbusEnabled":false},"devices":[{"type":"boiler","name":"GBx72/Trendline/Cerapur/Greenstar Si","deviceID":"0x08","productID":123,"brand":"","version":"07.00","entities":79,"handlersReceived":"0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04","handlersFetched":"0x14 0x16 0x33 0x26 0x28","handlersPending":"0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x2E 0x3B","handlersIgnored":"0x23"},{"type":"thermostat","name":"RC200/CW100/CR120","deviceID":"0x18","productID":157,"brand":"","version":"41.08","entities":46,"handlersReceived":"0x06 0xA2 0x031D","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x02F5 0x023A","handlersPending":"0xA3 0x12 0x13 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x0292 0x02D0 0x0469 0x0293 0x02D2 0x046A 0x0294 0x02F6 0x031B 0x031E 0x0267 0x0240 0xBB 0x023E 0x0269","handlersIgnored":"0xBF"},{"type":"controller","name":"BC25","deviceID":"0x09","productID":125,"brand":"","version":"03.05","entities":0}]}
```

### TO REPRODUCE

_Steps to reproduce the behavior:_
Updating from 3.6.5 to latest 3.7.0 dev

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

MichaelDvP, 2024-08-15:
I can not find A11/816 in the documentation, A11 is almost configuration errors with codes 80x or 82x.
816 are communication errors where Axx specifies the device, A2x for remotes, A3x solar, etc

Please do in terminal a `read 18 BF` and pos the result here to see the full eror message.

AzonInc, 2024-08-15:
I couldn't find that code anywhere either but called Bosh Support and they told me it's a bus communication error related to a pump or similar.

Here is the reading:
```
ems-esp:# read 18 BF
000+17:27:15.827 N 4: [emsesp] thermostat(0x18) -W-> Me(0x0B), SystemError(0xBF), data: 18 9D 00 0E 11 41 31 31 03 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00

```

MichaelDvP, 2024-08-16:
The BF only shows the error and source of error is thermostat. That fits to A11, which are thermostat internal errors. But 816 is a communication error. 
> The error shows up after a while.

How long is a while? Can you specify more exactly? Is it possible to start a log-all a minute before the error is shown and download the log it a minute later. Could also be more that a minute.

AzonInc, 2024-08-18:
I wish I could be more precise about the time it takes but this time it didn't happen for a while. Seems to be a bit random when it happens.

What I usually do when the error occurs is to shutdown the device (not ems-esp) and turn it on again. I hope i can capture logs when it happens the next time.

How can I log all btw?

MichaelDvP, 2024-08-18:
On the web page go to logger, select level "all", confirm. Leave the page open to log inside the browser memory (and not be limited by ems.esp memory). After the error happen, press download log to save it file.

AzonInc, 2024-08-27:
It occured again but unfortunately I only have logs after the error occured:
```
2024-08-27 17:33:16.846 T 820946: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:17.107 T 820947: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:17.406 T 820948: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:17.705 T 820949: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:17.975 T 820950: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:18.242 T 820951: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300Summer(0x02AF), length: 255
2024-08-27 17:33:18.309 T 820952: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300Summer(0x02AF), data: 03 28 00 00 46 2D 11 01 19 1E 00
2024-08-27 17:33:18.657 T 820953: [telegram] Incomplete Rx: 8B 98 FF 00 FF 01 9B A7 60
2024-08-27 17:33:19.058 T 820954: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:19.299 T 820955: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:19.561 T 820956: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:19.823 T 820957: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300Curves(0x029B), length: 255
2024-08-27 17:33:19.883 T 820958: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300Curves(0x029B), data: 04 01 00 00 00 02 05 30 4B
2024-08-27 17:33:19.883 T 820959: [emsesp] Me(0x0B) -R-> thermostat(0x18), HPMode(0x0291), length: 255
2024-08-27 17:33:19.957 T 820960: [emsesp] thermostat(0x18) -W-> Me(0x0B), HPMode(0x0291), data: 01 00 00 00
2024-08-27 17:33:19.977 T 820961: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300WWmode(0x02F5), length: 255
2024-08-27 17:33:20.035 T 820962: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300WWmode(0x02F5), data: 01 00 03 01 00 FF 08 01 00 08 04 00
2024-08-27 17:33:20.035 T 820963: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300OutdoorTemp(0x023A), length: 255
2024-08-27 17:33:20.090 T 820964: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300OutdoorTemp(0x023A), data: 01 0D
2024-08-27 17:33:20.156 T 820965: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:20.456 T 820966: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:20.755 T 820967: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:21.026 T 820968: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:21.288 T 820969: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:21.649 T 820970: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:21.938 T 820971: [emsesp] thermostat(0x18) -B-> All(0x00), RCError(0xA2), data: 41 31 31 03 30
2024-08-27 17:33:22.147 T 820972: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:22.418 T 820973: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:22.681 T 820974: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:23.009 T 820975: [emsesp] boiler(0x08) -B-> All(0x00), MC110Status(0x2A), data: 00 00 00 00 02 00 00 00 B4 00 00 80 00 00 80 00 80 00 80 00 00
2024-08-27 17:33:23.250 T 820976: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:23.589 T 820977: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 28 00 FC 80 00 A1 00 00 01 00 00 37 B5 00 1E 55 00 80 00
2024-08-27 17:33:23.797 T 820978: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 05 01 06 00 00 00 02 40 C0 00 FC 80 00 80 00 00 00 FF 00 00 00 CB 00 00 00
2024-08-27 17:33:24.040 T 820979: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:24.339 T 820980: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:24.674 T 820981: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:24.935 T 820982: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:25.196 T 820983: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:25.466 T 820984: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:25.771 T 820985: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:26.091 T 820986: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:26.427 T 820987: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:26.725 T 820988: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:26.986 T 820989: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:27.284 T 820990: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:27.583 T 820991: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:27.852 T 820992: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:28.158 T 820993: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:28.456 T 820994: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:33:28.714 T 820995: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
[... cut ...]
2024-08-27 17:34:08.476 T 821140: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameters(0x16), data: 00 4B 64 00 06 FA 0A 01 03 64 0A 02 00 00 00 00 00 00 00 00 00 00 00 00 28 00 3C
2024-08-27 17:34:08.527 T 821141: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:34:08.867 T 821142: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
2024-08-27 17:34:09.165 T 821143: [emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
```

Is it normal that `[emsesp] controller(0x09) -R-> 11(0x11), Version(0x02), length: 1` keeps repeating?

MichaelDvP, 2024-08-27:
0x11 is the device id of WM10 switch module. Seems to be configuered , but not connected. Also `UBADevices(0x07), data: 0B 00 01 00 00 00 00 00 00 00 00 00 00` shows that there is no device 0x11 connected.
Do you have changed the configuration to multiple heating circuits without having control elements for this?

AzonInc, 2024-08-27:
I'm not aware of any change in the last few months. I noticed the error first after updating.
Also the two last Categories appeared recently:
[image]

> Do you have changed the configuration to multiple heating circuits without having control elements for this?
Where can I check that?

MichaelDvP, 2024-08-28:
These extra devices recognized by the busmaster and ems-esp could be from a lot of garbage on the bus, maybe bad conncetion or EM-disturbance. But the system info you've send does not show any extra device and good bus quality.
If ems-esp detects a device it will stay. Do you have unplaned restarts with ems-esp? 
Check all cable connections and power supply.

AzonInc, 2024-08-28:
Everything is running for days now without any restarts. Connections are good as well, everything tight.

After the error occurred yesterday I restarted the Condens but not ems-esp. The error is gone for now, let's see if I can capture logs before the error and with the error.

But I'm kinda confused that in the logs I see RC300 but I'm using a RC200.

proddy, 2024-09-01:
I'll close this for now. Do re-open if the disconnect are coming back.

AzonInc, 2024-09-01:
I could finally get some logs before and after the error appears. It starts at 19:34:25.

```
[2024-09-01T19:33:25.732] [INFO] file - boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 00 01 00 00 00 00 00 00 00 00 00 00
[2024-09-01T19:33:25.834] [INFO] file - Me(0x0B) -R-> thermostat(0x18), RC300Monitor(0x02A5), length: 230 (offset 25)
[2024-09-01T19:33:25.942] [INFO] file - thermostat(0x18) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 07 00 00 00 00 00 00 4B 00 3C 01 FF 00 00 (offset 25)
[2024-09-01T19:33:26.041] [INFO] file - Me(0x0B) -R-> thermostat(0x18), RC300Set(0x02B9), length: 255
[2024-09-01T19:33:26.142] [INFO] file - thermostat(0x18) -W-> Me(0x0B), RC300Set(0x02B9), data: 00 2E 2C 26 00 03 00 FF FF 05 28 01 E1 20 01 0F 05
[2024-09-01T19:33:26.244] [INFO] file - Me(0x0B) -R-> thermostat(0x18), RC300Summer(0x02AF), length: 255
[2024-09-01T19:33:26.348] [INFO] file - thermostat(0x18) -W-> Me(0x0B), RC300Summer(0x02AF), data: 03 28 00 00 46 2D 11 01 19 1E 00
[2024-09-01T19:33:26.441] [INFO] file - Incomplete Rx: 8B 98 FF 00 FF 01 9B A7 09
[2024-09-01T19:33:29.251] [INFO] file - Me(0x0B) -R-> thermostat(0x18), RC300Curves(0x029B), length: 255
[2024-09-01T19:33:29.349] [INFO] file - thermostat(0x18) -W-> Me(0x0B), RC300Curves(0x029B), data: 04 01 00 00 00 02 05 30 4B
[2024-09-01T19:33:29.635] [INFO] file - Incomplete Rx: 8B 98 FF 00 FF 01 91 C9 88 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 FD
[2024-09-01T19:33:33.173] [INFO] file - boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 05 01 9F 00 00 00 02 40 C0 01 2B 80 00 80 00 00 00 FF 00 00 00 CB 00 00 00
[2024-09-01T19:33:33.396] [INFO] file - boiler(0x08) -B-> All(0x00), MC110Status(0x2A), data: 00 00 00 00 02 00 00 00 B4 00 00 80 00 00 80 00 80 00 80 00 00
[2024-09-01T19:33:33.646] [INFO] file - boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 28 01 2B 80 00 A1 00 00 01 00 00 37 91 00 1E 5D 00 80 00
[2024-09-01T19:33:33.938] [INFO] file - Me(0x0B) -R-> thermostat(0x18), HPMode(0x0291), length: 255
[2024-09-01T19:33:34.037] [INFO] file - thermostat(0x18) -W-> Me(0x0B), HPMode(0x0291), data: 01 00 00 00
[2024-09-01T19:33:34.141] [INFO] file - Me(0x0B) -R-> thermostat(0x18), RC300WWmode(0x02F5), length: 255
[2024-09-01T19:33:34.236] [INFO] file - thermostat(0x18) -W-> Me(0x0B), RC300WWmode(0x02F5), data: 01 00 03 01 00 FF 08 01 00 08 04 00
[2024-09-01T19:33:34.336] [INFO] file - Me(0x0B) -R-> thermostat(0x18), RC300OutdoorTemp(0x023A), length: 255
[2024-09-01T19:33:34.438] [INFO] file - thermostat(0x18) -W-> Me(0x0B), RC300OutdoorTemp(0x023A), data: 01 39
[2024-09-01T19:33:34.540] [INFO] file - boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 00 01 00 00 00 00 00 00 00 00 00 00
[2024-09-01T19:33:43.119] [INFO] file - boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 05 01 9F 00 00 00 02 40 C0 01 2B 80 00 80 00 00 00 FF 00 00 00 CB 00 00 00
[2024-09-01T19:33:43.346] [INFO] file - boiler(0x08) -B-> All(0x00), MC110Status(0x2A), data: 00 00 00 00 02 00 00 00 B4 00 00 80 00 00 80 00 80 00 80 00 00
[2024-09-01T19:33:43.592] [INFO] file - boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 28 01 2B 80 00 A1 00 00 01 00 00 37 91 00 1E 5D 00 80 00
[2024-09-01T19:33:45.382] [INFO] file - thermostat(0x18) -W-> boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
[2024-09-01T19:33:45.496] [INFO] file - thermostat(0x18) -B-> All(0x00), RC300Monitor(0x02A5), data: 01 29 21 28 00 00 28 28 05 A0 02 03 01 00 CF 01 E3 00 00 11 01 02
[2024-09-01T19:33:45.758] [INFO] file - thermostat(0x18) -B-> All(0x00), RC300Monitor(0x02A5), data: 0B 96 00 00 07 00 00 00 00 00 00 4B 00 3C 01 FF 00 00 (offset 22)
[2024-09-01T19:33:46.286] [INFO] file - thermostat(0x18) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 0A 07
[2024-09-01T19:33:52.505] [INFO] file - thermostat(0x18) -W-> boiler(0x08), ?(0x23), data: 00 00 00
[2024-09-01T19:33:53.040] [INFO] file - boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 28 01 2B 80 00 A1 00 00 01 00 00 37 91 00 1E 5D 00 80 00
[2024-09-01T19:33:53.324] [INFO] file - boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 05 01 9F 00 00 00 02 40 C0 01 2B 80 00 80 00 00 00 FF 00 00 00 CB 00 00 00
[2024-09-01T19:33:53.639] [INFO] file - boiler(0x08) -B-> All(0x00), MC110Status(0x2A), data: 00 00 00 00 02 00 00 00 B4 00 00 80 00 00 80 00 80 00 80 00 00
[2024-09-01T19:34:03.046] [INFO] file - boiler(0x08) -B-> All(0x00), MC110Status(0x2A), data: 00 00 00 00 02 00 00 00 B4 00 00 80 00 00 80 00 80 00 80 00 00
[2024-09-01T19:34:03.589] [INFO] file - boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 28 01 2B 80 00 A1 00 00 01 00 00 37 91 00 1E 5D 00 80 00
[2024-09-01T19:34:03.661] [INFO] file - boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 05 01 9F 00 00 00 02 40 C0 01 2B 80 00 80 00 00 00 FF 00 00 00 CB 00 00 00
[2024-09-01T19:34:07.371] [INFO] file - thermostat(0x18) -B-> All(0x00), RC300Monitor(0x02A5), data: 00 CE 01 E4 (offset 13)
[2024-09-01T19:34:11.426] [INFO] file - thermostat(0x18) -B-> All(0x00), RCTime(0x06), data: 18 09 13 01 22 06 06 01 18 FF 00
[2024-09-01T19:34:12.978] [INFO] file - boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 05 01 9F 00 00 00 02 40 C0 01 2B 80 00 80 00 00 00 FF 00 00 00 CB 00 00 00
[2024-09-01T19:34:13.231] [INFO] file - boiler(0x08) -B-> All(0x00), MC110Status(0x2A), data: 00 00 00 00 02 00 00 00 B4 00 00 80 00 00 80 00 80 00 80 00 00
[... cut ...]
[2024-09-01T19:35:46.538] [INFO] file - controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
[2024-09-01T19:35:46.801] [INFO] file - controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
[2024-09-01T19:35:47.099] [INFO] file - controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
[2024-09-01T19:35:47.436] [INFO] file - controller(0x09) -R-> 11(0x11), Version(0x02), length: 1
[2024-09-01T19:35:47.719] [INFO] file - thermostat(0x18) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 0A 07
```

MichaelDvP, 2024-09-02:
There is nothing to see from a 0x11 device. Shortly before the controller starts to poll the 0x11 there is an incomplete from ems-esp. Try to change the tx-mode, and/or try a different address (e.g. 0x49) for ems-esp, maybe this helps.

AzonInc, 2024-09-03:
I will try changing the address and check if it solves the issue

closed 2025-03-23 by proddy