emsesp/EMS-ESP32 #1938: nrgtotal, nrgheat, dhw.nrg are not reported correctly in 3.7.0-dev.30 compared to 3.6.5

paludi, 2024-08-16:
### PROBLEM DESCRIPTION

After updating to 3.7.0-dev.30, the reported values of nrgtotal, nrgheat, dhw.nrg from the boiler (Buderus GB172 in my case) change compared to 3.6.5:
reported values in 3.7.0-dev.30:
- "nrgtotal": 793.71,
-  "nrgheat": 793.71,
- "dhw.nrg": 0 

3.6.5:
-  "total energy (nrgtotal)": 9647.08,
-   "energy heating (nrgheat)": 793.71,
-   "dhw energy (nrgww)": 8853.37

It looks like 
- nrgheat is still reported correctly
- nrgtotal now reports the value of nrgheat instead of the total energy consumed
- energy for hot water (dhw.nrg) is not reported any longer. 

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ x] Provide the output of http://ems-esp.local/api/system : see attached files at bottom for both releases

```lua
  System information output here:
```

```
logs of dev.30

2024-08-16 07:20:52.547 W 671: [telegram] Incomplete Rx: 10 0B FF 00 01 40 41 00 03 00 00 00 00 00 FF 01 F4 06 FF FF 00 00 00 00 00 00 AA
2024-08-16 07:21:48.799 W 672: [telegram] Incomplete Rx: 10 0B FF 00 01 69 11 01 01 11 89
2024-08-16 07:22:54.945 W 673: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03 90
2024-08-16 07:23:47.909 W 674: [telegram] Incomplete Rx: 10 0B FF 00 01 94 00 00 00 00 FF 00 E4 FE 90
2024-08-16 07:25:46.753 W 675: [telegram] Incomplete Rx: 30 0B FF 00 02 58 FF 00 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 90
2024-08-16 07:25:56.350 W 676: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 00 04 89
2024-08-16 07:26:55.258 W 677: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 FF 90
2024-08-16 07:31:42.293 W 678: [telegram] Incomplete Rx: 10 0B FF 00 02 1B 3C FF AA
2024-08-16 07:32:41.230 W 679: [telegram] Incomplete Rx: 10 0B FF 00 01 94 00 00 00 00 FF 89
2024-08-16 07:33:52.228 W 680: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 00 00 00 03 00 00 00 00 00 E0 93
2024-08-16 07:35:52.601 W 681: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 00 04 05 06 05 00 00 00 00 00 03 03 FF 96
2024-08-16 07:36:47.426 W 682: [telegram] Incomplete Rx: 30 0B FF 00 02 58 FF 00 FF 00 00 00 00 00 00 00 00 00 00 FF 89
2024-08-16 07:38:46.142 E 683: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 33 00 FF
2024-08-16 07:39:48.764 W 684: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 FF 0C 0B 07 05 00 00 90
2024-08-16 07:41:41.620 E 685: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 28 00 FF
2024-08-16 07:41:45.949 W 686: [telegram] Incomplete Rx: 10 0B FF 00 90
2024-08-16 07:41:53.684 W 687: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03 FF 90
2024-08-16 07:45:03.591 W 688: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 FF 90
2024-08-16 07:45:38.957 W 689: [telegram] Incomplete Rx: 10 0B FF 00 01 F5 03 00 04 03 90
2024-08-16 07:47:42.185 W 690: [telegram] Incomplete Rx: 10 0B FF 00 01 3A B4
2024-08-16 07:48:39.826 E 691: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 33 00 FF
2024-08-16 07:50:51.001 W 692: [telegram] Incomplete Rx: 10 0B FF 00 89
2024-08-16 07:51:38.195 W 693: [telegram] Incomplete Rx: 10 0B FF 00 01 F5 03 00 04 03 00 00 08 01 00 08 03 00 00 80 89
2024-08-16 07:51:47.876 W 694: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 02 10 12 06 12 00 00 00 00 00 89
2024-08-16 07:53:43.532 W 695: [telegram] Incomplete Rx: 10 0B FF 00 01 F5 03 00 04 03 C0 90
2024-08-16 07:56:43.511 W 696: [telegram] Incomplete Rx: 10 0B FF 00 01 F5 03 00 04 03 93
2024-08-16 07:57:26.445 W 697: [telegram] Incomplete Rx: 20 00 FF 00 01 D7 00 02 00 FF 89
2024-08-16 07:57:44.077 W 698: [telegram] Incomplete Rx: 30 0B FF 00 02 58 FF 00 FF 00 00 00 00 00 00 00 00 00 00 00 97
2024-08-16 08:00:37.945 W 699: [telegram] Incomplete Rx: 10 0B FF 00 92
2024-08-16 08:02:42.932 W 700: [telegram] Incomplete Rx: 10 0B FF 00 01 F5 03 00 04 E3 90
2024-08-16 08:02:52.367 W 701: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 FF 09 08 07 01 00 00 00 00 00 03 03 03 90
2024-08-16 08:06:49.786 W 702: [telegram] Incomplete Rx: 10 0B FF 00 01 94 00 00 00 00 FF 89
2024-08-16 08:08:43.816 W 703: [telegram] Incomplete Rx: 10 0B FF 00 01 F5 03 00 04 03 00 00 08 01 00 08 03 FF 90
2024-08-16 08:08:54.376 W 704: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 91
2024-08-16 08:09:50.560 W 705: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 00 00 90
2024-08-16 08:12:57.842 W 706: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 90
2024-08-16 08:13:43.324 E 707: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 33 00 FF
2024-08-16 08:13:48.085 W 708: [telegram] Incomplete Rx: 10 0B FF 00 01 3A 89
2024-08-16 08:14:41.280 W 709: [telegram] Incomplete Rx: 10 0B FF 00 89
2024-08-16 08:16:44.275 W 710: [telegram] Incomplete Rx: 10 0B FF 00 02 1B FE 89
2024-08-16 08:16:49.853 W 711: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 90
2024-08-16 08:16:50.111 W 712: [telegram] Incomplete Rx: F8 08 00 07 00 0B 01 00 01 04 01 00 00 00 00 00 00 00 20
2024-08-16 08:17:40.540 W 713: [telegram] Incomplete Rx: 10 0B FF 00 01 F5 03 00 04 03 00 90
2024-08-16 08:18:44.415 W 714: [telegram] Incomplete Rx: 10 0B FF 00 01 69 11 FF 89
2024-08-16 08:21:26.927 W 715: [telegram] Incomplete Rx: 20 00 FF 00 01 D7 00 02 00 89
2024-08-16 08:22:45.392 W 716: [telegram] Incomplete Rx: 10 0B FF 00 01 40 41 00 03 00 00 00 00 00 FF 01 90
2024-08-16 08:24:32.620 W 717: [mqtt] Publish failed: low memory
2024-08-16 08:24:32.637 W 718: [telegram] Incomplete Rx: 89 00 09 00 B9
2024-08-16 08:24:32.773 W 719: [telegram] Incomplete Rx: 89 00 09 00 89 00 09
2024-08-16 08:24:52.572 W 720: [telegram] Incomplete Rx: 2A 0B FF 00 06 E0 00 00 00 00 00 00 00 00 03 8D

```

```
logs of 3.6.5 

2024-08-16 08:31:38.729 E 19: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 26 00 1B
2024-08-16 08:32:21.554 E 20: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 00 19 01 BE
2024-08-16 08:32:29.500 W 21: [telegram] Incomplete Rx: 89 00 09 00
2024-08-16 08:32:29.766 W 22: [telegram] Incomplete Rx: A2 00 89 00
2024-08-16 08:32:29.865 W 23: [telegram] Incomplete Rx: B0 00 30 00 89 00
2024-08-16 08:32:30.034 W 24: [telegram] Incomplete Rx: B2 00 89 00
```

### TO REPRODUCE

update from rel. 3.6.5 to 3.7.0-dev.30 and check the above mentioned values

### EXPECTED BEHAVIOUR

reported energy values are correct (total energy value is increasing, hot water energy reports a value)

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

Thank you for maintaining this great piece of software to keep my heating connected! 
Let me know if I can help with any further information. 

System info and all values for 3.7.0-dev.30 and 3.6.5:

[emsesp_system_allvalues-3.7.0-dev.30.txt](https://github.com/user-attachments/files/16633964/emsesp_system_allvalues-3.7.0-dev.30.txt)
[emsesp_system_info-3.7.0-dev.30.txt](https://github.com/user-attachments/files/16633965/emsesp_system_info-3.7.0-dev.30.txt)
[emsesp_system_allvalues3.6.5.txt](https://github.com/user-attachments/files/16633966/emsesp_system_allvalues3.6.5.txt)
[emsesp_system_info-3.6.5.txt](https://github.com/user-attachments/files/16633967/emsesp_system_info-3.6.5.txt)

MichaelDvP, 2024-08-16:
The dhw circuits are handled different now. This affects also the stored dhw energie value, reset to zero. (i will check if we can add a upgrade conversion).
Just enter the value in boiler dhw energy once manually. The total energy is the sum of heat/dhw energies, with dhw-nrg zero it is identical with nrgheat.

paludi, 2024-08-16:
Thank you @MichaelDvP for your fast response.

I can confirm that your manual workaround works. I created a small PR to add your solution to the breaking changes section of the changelog - feel free to add it if you think it is helpful.

closed 2024-08-16 by paludi

MichaelDvP, 2024-08-16:
The dev is changes to use the old names (fixed).