emsesp/EMS-ESP32 #1940: BC400: Cooling Energy and Consumption values without digit after comma

UlfHRO, 2024-08-16:
### PROBLEM DESCRIPTION

Following values are without any digit after the comma (only full numbers reported even available in the heat pump and Buderus app):
- total energy supplied cooling
- energy consumption compressor cooling

In IObroker EMS-ESP Adapter this related values also only as full numbers without digit:
- ems: total energy supplied cooling (ems-esp.0.boiler.nrgsuppcooling)
- ems: energy consumption compressor cooling (ems-esp.0.boiler.nrgconscompcooling)

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

One digit after comma for that values...

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

MichaelDvP, 2024-08-16:
These energies are published from boiler without decimals. For newer heatpumps there are additional telegrams
0x4AE (energy) and 0x4AF (meters) with resolution 1/100. But we don't know f it contains cooling.
Try in terminal `read 8 4AE` and check the values (always 4 bytes for a value).
Or create custom entities for the unknown values in 4AE like this with offsets 8, 16, 20, 24, 28: 
[image]
Maybe the cooling energy is there

UlfHRO, 2024-08-16:
Thanks... Found both:
- cooling produced: 04AE offset 20
- cooling consumed: 04AF offset 40
Thanks for the tip...

MichaelDvP, 2024-08-16:
Ok ,thanks, i'll add to dev.

proddy, 2024-08-16:
Added to dev 3.7.0-dev-30