emsesp/EMS-ESP32 #1947: HA MQTT: climate entity id's are language dependend

tp1de, 2024-08-17:
The climate entity id's seem to be language dependend.
I do have climate.thermostat_hk1 and climate.thermostat_hk1. (in German)

The name is either HK1 or HK2 - this is ok.

I believe that the entity ids should be hc1 ... hc4 - like all other sensors / switches / numbers.
Just names should be translated not id's.

proddy, 2024-08-17:
Can you post screenshot? It's hardcoded to hc in EMS-ESP, and tags are not translated in MQTT (DeviceValue::DeviceValueTAG_mqtt[] ).  I just set the language to German and the climate is still climate.hc. Perhaps its localized within HA?

tp1de, 2024-08-17:
MQTT explorer shows:
[image]

proddy, 2024-08-17:
are you using a special build? Code shows its hc_<n>

https://github.com/emsesp/EMS-ESP32/blob/c7d721bb10edf146481e0c290e02d64ef01456aa/src/mqtt.cpp#L1264

tp1de, 2024-08-17:
Yes I am using @MichaelDvP 's version (dev.30) since a couple of days.
Yes you are right with the original dev.30 it is right. 

I wanted to test with your latest dev.31 version, but I am confused:
There are two 16 MB versions now. Which one to use?  The new + one.
And download from webui gives a 404 error

MichaelDvP, 2024-08-17:
MY build is only a few commits behnd, it was merged yesterday.
For the version see https://github.com/emsesp/EMS-ESP32/issues/1933#issuecomment-2293958637
16M+ for E32v2, 16M for older S32 gateways.The wrong file will upload and start, but the settings are cleared and default to wrong board setting.

proddy, 2024-08-17:
firmware binary filename has changed (https://github.com/emsesp/EMS-ESP32/issues/1933#issuecomment-2293958637) . the + at the end means PSRAM. 

I looked at Michael's branch and its the same code, so I confused how the HC is translated in the climate HA config. Very strange.

tp1de, 2024-08-17:
I am confused in the moment as well. When I restart the EMS-ESP gateway the hc climate entities are created in mqtt. 
90 seconds later the hk ones are there additionally.

When I delete all climate entities within mqtt explorer, then only the hk ones are recreated.

tp1de, 2024-08-17:
I switched off MQTT in HA and the hk climate entities are not created anymore. I have to look within HA to find the cause.

Sorry for this issue ... I will close.

closed 2024-08-17 by tp1de