emsesp/EMS-ESP32 #1955: [3.7.0-dev.32] virtual remote w/ RC220

Bingo2023, 2024-08-18:
EMS-ESP Version
3.7.0-dev.32

hello,
any idea why the virtual remote with RC220 is not accepted, see log ?

It is running somehow but not so stable...

[image]
[image]

Bingo2023, 2024-08-19:
here is a screenshot from MQTT Explorer from that message:
[image]

Bingo2023, 2024-08-19:
how to extend ? Logging-level "all" does not increase...

proddy, 2024-08-19:
I think we would need to give you a DEBUG compiled build which has more info

MichaelDvP, 2024-08-20:
- control-command does not have the option RC220, check /api/thermostat/hc/control for valid options. [1]
- RC200 only provides temperaature, it does not support humidity
- i don't know how Bosch manages writes to flash/eeprom stored elements, but it's always a very bad idea to write such values in a high frequency.

footnotes: 
[1] some versions ago there was the option control:RC220, but no thermostat emulation for this. We renamed when developing the RT800 emulation. So this could have never worked.

closed 2024-09-01 by proddy