emsesp/EMS-ESP32 #1957: Heatpump with BC400/RC220 DHW mode and Energiemanager

UlfHRO, 2024-08-19:
On my Buderus WLW186 heatpump with ems-esp 3.7.0-dev.30 I use successful the dhw mode to select between the different dhw modes (eco, eco+, comfort, off and auto). But, when I select the auto mode (time program in the hp) I do not see any feedback which mode is actually active during that auto mode...?! Is there a custom entity where the actual mode during auto is indicated...?

I also use the Buderus Energiemanager and it is increasing the setpoint temperature based on PV power available. Also this works fine and I see the increased temp setpoint, but is there any chance to catch that this happened from the Energiemanager and what the forced settings (power, temp , etc.) from the Energiemanager are...?

proddy, 2024-08-19:
For the Energiemanager, see https://emsesp.org/Troubleshooting/#im-missing-certain-data-from-an-ems-device . Basically turn on logging to ALL or DEBUG (either in the telnet console or the webUI), play around with the energiemanager and see if you can capture which telegrams it is sending and the BUS ID, if any, it's using.

MichaelDvP, 2024-08-20:
see also #1717

Seems we have 0x04AA, offset 0 for current power consumption and 0x04A7, offset 0  for the power limit.
Working on some pumps, but not for all. 
Still only as custom entity, not implemented yet

Maybe there are more values or settings.

UlfHRO, 2024-08-20:
Hi Michael,
Thanks...

Confirm for my WLW186:

  *   actual power consumption:

power
0x08
0x4AA
0
UINT16

  *   cooling energy consumption

cool_cons
0x08
0x4AF
40
UINT32

  *   cooling energy produced

cool_prod
0x08
0x4AE
20
UINT32

  *   compressor rpm...? what is the scaling...?

comp_rpm
0x08
0x48D
17
INT16

  *   Absent mode

absent
0x10
0x16E
0
INT8

  *   dhw actual active mode (0-off, 1-eco, 2-comfort, 3-eco+)

ww
0x10
0x31D
8
UINT8

I will monitore the power limit at 0x04A7, offset 0, but actually is 0 even the hp already started dhw increase...?

btw: Do you know the adress for the system pressure? Is it also available with more then 1 digit...?

Thanks, Ulf

Message ID: ***@***.***>

UlfHRO, 2024-08-20:
Hi Proddy,
There is nothing to play with at the Buderus MyEnergyMaster, it works very much as a "blackbox"...
Anyhow please find attached my config and a log file during a time where the dhw was increased by the EnergyMaster and cooling was running, both based on PV surplus...
Hope that helps somehow to get more inside to the EnergyMaster...

Please note that 04A7 offset 0 has only shown 0 during the entire day. That was suspected as powerlimit, but not changed at all on my HP WLW186...

Thanks and regards,
Ulf

------------------
Ulf Hirsekorn
------------------

  "system": {
    "version": "3.7.0-dev.30",
    "uptime": "003+08:18:17.110",
    "uptimeSec": 289097,
    "platform": "ESP32",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "freeMem": 133,
    "maxAlloc": 51,
    "freeCaps": 84,
    "usedApp": 1690,
    "freeApp": 870,
    "partition": "boot",
    "resetReason": "Software reset CPU / Software reset CPU"
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -61,
    "TxPowerSetting": 0,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tzLabel": "Europe/Amsterdam"
  },
  "mqtt": {
    "MQTTStatus": "disconnected",
    "enabled": false,
    "clientID": "esp32-5ae2dec4",
    "keepAlive": 60,
    "cleanSession": true,
    "entityFormat": 0,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 2,
    "haEnabled": false,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 10,
    "publishTimeOther": 60,
[... 561 lines cut ...]
2024-08-20 16:53:45.222 TRACE 3259: [emsesp] thermostat(0x10) -W-> boiler(0x08), PVSettings(0x023E), data: 00 (offset 5)
2024-08-20 16:53:45.241 TRACE 3260: [emsesp] boiler(0x08) -R-> thermostat(0x10), RC300Settings(0x0240), length: 1 (offset 10)
2024-08-20 16:53:45.272 TRACE 3261: [emsesp] thermostat(0x10) -W-> boiler(0x08), RC300Settings(0x0240), data: F4 (offset 10)
2024-08-20 16:53:45.276 TRACE 3262: [emsesp] boiler(0x08) -R-> thermostat(0x10), RC300Settings(0x0240), length: 1 (offset 12)
2024-08-20 16:53:45.290 TRACE 3263: [emsesp] thermostat(0x10) -W-> boiler(0x08), RC300Settings(0x0240), data: FF (offset 12)
2024-08-20 16:53:45.311 TRACE 3264: [emsesp] boiler(0x08) -R-> thermostat(0x10), RC300Settings(0x0240), length: 1 (offset 24)
2024-08-20 16:53:45.325 TRACE 3265: [emsesp] thermostat(0x10) -W-> boiler(0x08), RC300Settings(0x0240), data: FF (offset 24)
2024-08-20 16:53:45.346 TRACE 3266: [emsesp] boiler(0x08) -R-> thermostat(0x10), HPMode(0x0291), length: 1
2024-08-20 16:53:45.372 TRACE 3267: [emsesp] thermostat(0x10) -W-> boiler(0x08), HPMode(0x0291), data: 01
2024-08-20 16:53:45.382 TRACE 3268: [emsesp] boiler(0x08) -R-> thermostat(0x10), HPMode(0x0291), length: 1 (offset 5)
2024-08-20 16:53:45.404 TRACE 3269: [emsesp] thermostat(0x10) -W-> boiler(0x08), HPMode(0x0291), data: 02 (offset 5)
2024-08-20 16:53:45.407 TRACE 3270: [emsesp] boiler(0x08) -R-> thermostat(0x10), HPMode(0x0292), length: 1
2024-08-20 16:53:45.438 TRACE 3271: [emsesp] thermostat(0x10) -W-> boiler(0x08), HPMode(0x0292), data: 00
2024-08-20 16:53:45.448 TRACE 3272: [emsesp] boiler(0x08) -R-> thermostat(0x10), HPMode(0x0292), length: 1 (offset 5)
2024-08-20 16:53:45.472 TRACE 3273: [emsesp] thermostat(0x10) -W-> boiler(0x08), HPMode(0x0292), data: 00 (offset 5)
2024-08-20 16:53:45.476 TRACE 3274: [emsesp] boiler(0x08) -R-> thermostat(0x10), HPMode(0x0293), length: 1
2024-08-20 16:53:45.511 TRACE 3275: [emsesp] thermostat(0x10) -W-> boiler(0x08), HPMode(0x0293), data: 00
2024-08-20 16:53:45.522 TRACE 3276: [emsesp] boiler(0x08) -R-> thermostat(0x10), HPMode(0x0293), length: 1 (offset 5)
2024-08-20 16:53:45.545 TRACE 3277: [emsesp] thermostat(0x10) -W-> boiler(0x08), HPMode(0x0293), data: 00 (offset 5)
2024-08-20 16:53:45.547 TRACE 3278: [emsesp] boiler(0x08) -R-> thermostat(0x10), HPMode(0x0294), length: 1
2024-08-20 16:53:45.578 TRACE 3279: [emsesp] thermostat(0x10) -W-> boiler(0x08), HPMode(0x0294), data: 00
2024-08-20 16:53:45.588 TRACE 3280: [emsesp] boiler(0x08) -R-> thermostat(0x10), HPMode(0x0294), length: 1 (offset 5)
2024-08-20 16:53:45.611 TRACE 3281: [emsesp] thermostat(0x10) -W-> boiler(0x08), HPMode(0x0294), data: 00 (offset 5)
2024-08-20 16:53:45.614 TRACE 3282: [emsesp] boiler(0x08) -R-> thermostat(0x10), RC300Curves(0x029B), length: 1 (offset 1)
2024-08-20 16:53:45.641 TRACE 3283: [emsesp] thermostat(0x10) -W-> boiler(0x08), RC300Curves(0x029B)

closed 2025-03-23 by proddy