emsesp/EMS-ESP32 #1960: Can't upgrade from 3.6.4 to 3.6.5

eelcohn, 2024-08-24:
### PROBLEM DESCRIPTION

I'm on EMS-ESP v3.6.4 and I can't update the firmware to v3.6.5. When I upload the file I get an error message saying "Invalid file extension or incompatible bin file". I've tried re-downloading the v3.6.5 firmware several times, and I've also tried upgrading the firmware on a different laptop.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the output of http://ems-esp.local/api/system :

```lua
{
  "System Info": {
    "version": "3.6.4",
    "platform": "ESP32",
    "uptime": "000+00:02:16.322",
    "uptime (seconds)": 136,
    "free mem": 168,
    "max alloc": 107,
    "free app": 419,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -53,
    "IPv4 address": "192.168.1.224/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "192.168.1.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": true,
    "server": "pool.ntp.org",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 145,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 123,
    "bus reads (tx)": 39,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "MH-ET",
    "locale": "en",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": true,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 15
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "BK13/BK15/Smartline/GB1x2",
      "device id": "0x08",
      "product id": 64,
      "version": "03.05",
      "entities": 75,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x2A 0xD1 0xE3 0xE4 0xE5 0xE9"
    },
    {
      "type": "thermostat",
      "name": "RC10/Moduline 100",
      "device id": "0x17",
      "product id": 79,
      "version": "01.02",
      "entities": 12,
      "handlers received": "0xA2",
      "handlers fetched": "0xB1 0xB0",
      "handlers pending": "0xA3 0x06 0x12 0x13",
      "handlers ignored": "0x1A 0x35"
    },
    {
      "type": "controller",
      "name": "BC10/RFM20",
      "device id": "0x09",
      "product id": 68,
      "version": "02.03",
      "entities": 0
    }
  ]
}
```

### TO REPRODUCE

### EXPECTED BEHAVIOUR

Be able to upgrade from v3.6.4 to the latest stable firmware.

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

mattreim, 2024-08-24:
Check out these discussions #1905.

MichaelDvP, 2024-08-26:
This message could have a lot of reasons.
- check you've downloaded the right bin file, maybe redownload it. 
 You're using board profile MH-Et, for this module it should be this: https://github.com/emsesp/EMS-ESP32/releases/download/v3.6.5/EMS-ESP-3_6_5-ESP32.bin The 16M file is too large and the S3 is wrong processor architecture.
- Try reboot ems-esp and wait until HA-discovery has finished (some minutes), or disable HA discovery, reboot and then try the update.

proddy, 2024-08-28:
You could also download the settings and go straight to https://github.com/emsesp/EMS-ESP32/releases/download/latest/EMS-ESP-3_7_0-dev_32-ESP32-4MB.bin

proddy, 2024-09-01:
Suggest make a copy of the settings, erase ESP32, re-flash with the latest dev. It has better support for detecting the different firmware versions.

closed 2024-09-01 by proddy

eelcohn, 2024-09-18:
For reference for others who experience the same problem: this solved my issue.

> Suggest make a copy of the settings, erase ESP32, re-flash with the latest dev. It has better support for detecting the different firmware versions.