emsesp/EMS-ESP32 #1964: Low memory error

alexdan-77, 2024-08-28:
### PROBLEM DESCRIPTION
huge list of such errors:
2024-08-28 16:07:37.271 W 79707: [mqtt] Publish failed: low memory

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
[log.txt](https://github.com/user-attachments/files/16783850/log.txt)

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2024-08-28:
the system info is missing. Please include

alexdan-77, 2024-08-28:
The latest official release is 3.6.5
[image]

proddy, 2024-08-29:
What we really needed is the system info as mention in the boilerplate GitHub Issue text when you create a new one, and also in https://emsesp.org/Support/. Then we can see which services you are running as your memory is very low (hence the warning errors). 

The ESP32 looks like a 4MB ESP32 (not a S2, S3) with 4MB PSRAM. Is this a custom built board?

alexdan-77, 2024-08-29:
I have bought it at: https://bbqkees-electronics.nl/product/gateway-e32-v2-ethernet-wifi-edition-v2/

Jaano, 2024-08-29:
I have same device as alexdan and same problem.
FW 3.7.0dev12 runs without out of memory messages.

bbqkees, 2024-08-30:
It's a bit misleading with the line spacing.

```
ESP32/ESP32-D0WD-V3 (rev.3, dual-core) @ 240 Mhz

Heap (Free / Max Alloc)
67 KB / 47 KB

PSRAM (Size / Free)
4 093 KB / 3 632 KB

Flash Chip (Size / Speed)
16 384 KB / 40 MHz

Application (Partition: Used / Free)
app1: 1 591 KB / 6 537 KB

File System (Used / Free)
28 KB / 36 KB
```

Now it's more clear it's indeed a E32 V2.

alexdan-77, 2024-08-30:
But i have it. What can i say else...does anyone know how to achieve well working state?

proddy, 2024-08-31:
still waiting for your system info so we can see what is using up all the memory

alexdan-77, 2024-08-31:
but i gave system info! where to find other system info ?

proddy, 2024-08-31:
Ths System Info you can get either from the Web Help page or via http://ems-esp.local/api/system/info

alexdan-77, 2024-08-31:
{"System Info":{"version":"3.6.5","uptime":"000+06:19:58.768","uptime (seconds)":22798,"platform":"ESP32","arduino":"Tasmota Arduino v2.0.14","sdk":"4.4.6.240105","free mem":67,"max alloc":47,"used app":1591,"free app":6537,"partition":"app1","reset reason":"Power on reset / APP CPU reset by PRO CPU"},"Network Info":{"network":"Ethernet","hostname":"ems-esp","TxPower setting":78,"static ip config":false,"enable IPv6":true,"low bandwidth":false,"disable sleep":false,"enable MDNS":true,"enable CORS":false,"AP provision mode":"disconnected","AP security":"wpa2","AP ssid":"ems-esp"},"NTP Info":{"NTP status":"connected","enabled":true,"server":"time.google.com","tz label":"Europe/Moscow"},"OTA Info":{"enabled":false,"port":8266},"MQTT Info":{"MQTT status":"connected","MQTT publishes":28279,"MQTT queued":0,"MQTT publish fails":0,"MQTT connects":1,"enabled":true,"client id":"ems-esp","keep alive":60,"clean session":false,"entity format":1,"base":"home/ems-esp-module","discovery prefix":"homeassistant","discovery type":0,"nested format":1,"ha enabled":false,"mqtt qos":0,"mqtt retain":false,"publish time heartbeat":60,"publish time boiler":45,"publish time thermostat":45,"publish time solar":60,"publish time mixer":30,"publish time other":60,"publish time sensor":30,"publish single":true,"publish2command":false,"send response":false},"Syslog Info":{"enabled":false},"Sensor Info":{"temperature sensors":11,"temperature sensor reads":49970,"temperature sensor fails":3,"analog sensors":0,"analog sensor reads":0,"analog sensor fails":0},"API Info":{"API calls":559,"API fails":0},"Bus Info":{"bus status":"connected","bus protocol":"Buderus","bus telegrams received (rx)":43465,"bus reads (tx)":11637,"bus writes (tx)":562,"bus incomplete telegrams":0,"bus reads failed":77,"bus writes failed":2,"bus rx line quality":100,"bus tx line quality":100},"Settings":{"board profile":"E32V2","locale":"en","tx mode":1,"ems bus id":11,"shower timer":false,"shower alert":false,"hide led":false,"notoken api":true,"readonly mode":false,"fahrenheit":false,"dallas parasite":false,"bool format":6,"bool dashboard":1,"enum format":1,"analog enabled":true,"telnet enabled":true,"max web log buffer":50,"web log buffer":50},"Devices":[{"type":"boiler","name":"GBx72/Trendline/Cerapur/Greenstar Si","device id":"0x08","product id":123,"version":"06.08","entities":84,"handlers received":"0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04","handlers fetched":"0x14 0x16 0x33 0x26 0x28","handlers pending":"0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA","handlers ignored":"0x23"},{"type":"thermostat","name":"RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410","device id":"0x10","product id":158,"version":"18.03","entities":119,"handlers received":"0x06 0xA2 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267","handlers fetched":"0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x02A7 0x02BB 0x02B1 0x029D 0x0291 0x02CE 0x0292 0x02D0 0x0293 0x0294 0x02F5 0x031B 0x023A 0x0240","handlers pending":"0xA3 0x12 0x13 0x0471 0x0472 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x0468 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E","handlers ignored":"0xBF 0x0684"},{"type":"mixer","name":"MM100","device id":"0x21","product id":160,"version":"24.05","entities":7,"handlers received":"0x02D8","handlers fetched":"0x02CE","handlers ignored":"0x0255 0x02E2 0x02EC 0x16 0x33 0x0292 0xBF"},{"type":"mixer","name":"MM100","device id":"0x22","product id":160,"version":"24.05","entities":3,"handlers received":"0x02D9","handlers pending":"0x02D0","handlers ignored":"0x16 0x33 0x0255 0x0293 0x02E3 0x02ED 0xBF"},{"type":"controller","name":"BC25","device id":"0x09","product id":125,"version":"03.03","entities":0}]}

proddy, 2024-08-31:
Can't see anything strange at a first look. I still recommend you upgrade to 3.7 and try again.

alexdan-77, 2024-08-31:
when i did a reboot it dissappears of course...but sometimes stucks...i beleive that is because of very often mqtt sends data. I extended timeouts and it's been sticking less...

The latest development release is 3.7.0-dev.32, i will try. thanks.

closed 2024-09-01 by alexdan-77