emsesp/EMS-ESP32 #1966: Regression: "exit" command does not work in standalone builds any more

mheyse, 2024-08-28:
### PROBLEM DESCRIPTION

The "exit" command does nothing in a standalone build. Other than only being able to exit the standalone build with Ctrl-C this leads to support scripts like `dump_entities.sh` and `update_modbus_registers.sh` to hang forever and not performing its intended tasks any more.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

Built from the latest dev branch, commit

```
commit 1cd547049e0431d6e5d1f9823f6e84d2b303da7f
Author: mattreim <80219712+mattreim@users.noreply.github.com>
Date:   Mon Aug 19 13:44:53 2024 +0200

    Fix for official release button

    Sorry, I overlooked it.
```

### TO REPRODUCE

Make a standalone build as produced by `dump_entities.sh` on Ubuntu, i.e.

```
make clean
make ARGS=-DEMSESP_STANDALONE
```

run `./emsesp`

Try to exit the program by typing `exit`. Instead of exiting, `ems-esp:# exit` is echoed and the ems-esp prompt is displayed again. This can be repeated indefinitely.

### EXPECTED BEHAVIOUR

The application quits and returns to the shell prompt.

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

(none)

MichaelDvP, 2024-08-29:
Oh, bad side effect from https://github.com/emsesp/EMS-ESP32/commit/0105338e29d4e67a48cf444e569ce35872c18c18

Change in lib/uuid-console/src/shell.cpp to
```
#if defined(EMSESP_STANDALONE)
        if (running()) {
#else
        if (running() && !has_flags(CommandFlags::LOCAL)) { // do not close local shell
#endif
```

mheyse, 2024-08-29:
Thanks, that fixed it.

proddy, 2024-09-09:
re-opening this. exit doesn't work on Serial console. I'll find a better solution

MichaelDvP, 2024-09-09:
> exit doesn't work on Serial console.

Yes, this was intension for commit https://github.com/emsesp/EMS-ESP32/commit/0105338e29d4e67a48cf444e569ce35872c18c18
Before exit closes the console (but not the serial line/putty), but you have to reset the esp to connect again.

proddy, 2024-09-09:
Yes I remember. I need to find nicer way to re-enable it after the session is closed. Now with minicom/ptty you cannot exit with killing the host

MichaelDvP, 2024-09-09:
> Now with minicom/ptty you cannot exit with killing the host

I don't understand this. It's a serial connection, putty/terminal stays open until you can kill the serial port or close it explicide. If you stop the console, on esp the serial connection stay, but you can not connect again.

proddy, 2024-09-09:
I wanted to end the session after an exit. I'll add to my PR so you see what I mean

proddy, 2024-09-09:
its in dev-35

closed 2024-09-15 by proddy