emsesp/EMS-ESP32 #1978: Sporadic restarts since dev.33

mattreim, 2024-09-05:
### PROBLEM DESCRIPTION

It can take hours or minutes. Unfortunately I don't have any log data because my putty always switches off.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{
  "system": {
    "version": "3.7.0-dev.33",
    "uptime": "000+00:04:27.276",
    "uptimeSec": 267,
    "platform": "ESP32",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "freeMem": 155,
    "maxAlloc": 87,
    "freeCaps": 105,
    "usedApp": 1690,
    "freeApp": 294,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU"
  },
  "network": {
    "network": "WiFi",
    "hostname": "EMS-ESP",
    "RSSI": -57,
    "BSSID": "set",
    "TxPowerSetting": 0,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": false,
    "enableMDNS": false,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "connected",
    "enabled": true,
    "server": "192.168.178.1",
    "tzLabel": "Europe/Berlin"
  },
  "mqtt": {
    "MQTTStatus": "disconnected",
    "enabled": false,
    "clientID": "esp32-422a13d8",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 0,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 2,
    "haEnabled": false,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 0,
    "publishTimeThermostat": 0,
    "publishTimeSolar": 0,
    "publishTimeMixer": 0,
    "publishTimeWater": 0,
    "publishTimeOther": 0,
    "publishTimeSensor": 0,
    "publishSingle": true,
    "publish2command": true,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {},
  "api": {
    "APICalls": 165,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "Buderus",
    "busTelegramsReceived": 748,
    "busReads": 209,
    "busWrites": 0,
    "busIncompleteTelegrams": 0,
    "busReadsFailed": 0,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "CUSTOM",
    "locale": "de",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 120,
    "showerAlert": false,
    "phyType": 0,
    "rxGPIO": 4,
    "txGPIO": 5,
    "dallasGPIO": 0,
    "pbuttonGPIO": 0,
    "ledGPIO": 0,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 6,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": false,
    "telnetEnabled": true,
    "maxWebLogBuffer": 25,
    "webLogBuffer": 11,
    "modbusEnabled": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "Logamax plus GB172i.2-15",
      "deviceID": "0x08",
      "productID": 195,
      "brand": "Buderus",
      "version": "05.06",
      "entities": 60,
      "handlersReceived": "0xBF 0xC2 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x2E 0x3B 0x04",
      "handlersFetched": "0x14 0xE6 0xEA 0x28",
      "handlersPending": "0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A",
      "handlersIgnored": "0xD7 0xD8 0x02D6 0x029B 0xE7 0x02E0 0x02EA"
    },
    {
      "type": "thermostat",
      "name": "Logamatic BC400",
      "deviceID": "0x10",
      "productID": 4,
      "brand": "",
      "version": "49.04",
      "entities": 43,
      "handlersReceived": "0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x023A 0x0240",
      "handlersPending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E",
      "handlersIgnored": "0xC4 0x047B 0xBF"
    },
    {
      "type": "thermostat",
      "name": "RC100H",
      "deviceID": "0x38",
      "productID": 200,
      "brand": "",
      "version": "40.04",
      "entities": 3,
      "handlersReceived": "0x042B 0x047B",
      "handlersPending": "0x0273 0x0A6A",
      "handlersIgnored": "0xF7"
    }
  ]
}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2024-09-05:
settings look ok. Is this a custom built board? It could be power related. If you could try and capture the log (like using syslog) so we can see what is happening before it restarts. Also, do you have any custom entities or scheduler entities?

proddy, 2024-09-06:
Maybe double check with @bbqkees to see the minimal Amps needed for the board. Newer versions of EMS-ESP are more power hungry.

MichaelDvP, 2024-09-06:
Watchdog reset looks like somthing is blocking. Does the schedule use http-request? 
Can you check the rssi/connection rate in the router and maybe increase the tx-power to get better wifi connection (rssi -57 for router->emsesp is good, the other direction is shown in router).

mattreim, 2024-09-06:
For these devices I use a repeater that is located one room away.

[image]

The watchdog message was also unknown to me.

bbqkees, 2024-09-06:
> Maybe double check with @bbqkees to see the minimal Amps needed for the board. Newer versions of EMS-ESP are more power hungry.

Ever since we moved away from the ESP8266 platform (Wemos D1 Mini) and bus powering the Gateway, power issues are basically never occurring again.
At the time the ESP8266 Gateway Premium II usually worked fine on bus power, but in some cases there was not enough power available, which gave all kinds of strange errors.
And the cheap ESP8266 Wemos D1 Mini clones often had a LDO voltage regulator which could not even deliver 150mA@3V3 so that could also give issues when f.i. the access point was on.
Most of the ESP32 development boards have a more sufficient LDO, so power issues are very unlikely nowadays.

MichaelDvP, 2024-09-06:
The 121 Mbit/s are direction from router(repeater) to emsesp, check other direction. With auto-tx-power i get sometimes only 27Mbit/s:
[image]

The dev34 raises the tx-power by 5dBm, this should give full connection speed.

mattreim, 2024-09-06:
The dev34 has been running for 14 hours now.

[image]

mattreim, 2024-09-07:
Version dev34 seems to have fixed the reboots. Now with over 100 thousand telegrams without any problems. Thank you very much for your time, help and information.

closed 2024-09-07 by mattreim