emsesp/EMS-ESP32 #1979: many MQTT failures after update

thombec, 2024-09-05:
Hi,

after update to the 3.7.0-dev.29 version I recognized that some values from EMS-ESP are not so "acurate" like before. the graph is not smooth - it looks like stairs:
[image]

purple: external measured temperature
blue: tempearture from EMS-ESP

After updating to the current dev version 33 the values looks simmilar - no change.

now I recognized that the MQTT failures are rising rapidly - After some days I have to reset the EMS-ESP module due to the fact that no data are transferred anymore.

[image]

At my last installed version 3.5.1 the MQTT transfer runs perfect - no errors occurred.

proddy, 2024-09-05:
could you try and capture any MQTT errors that are reported in the log? Also send your system info - want to check if you're using PSRAM as some of the newer MQTT code switches between async and sync modes. Also if you're using Discovery. The more details the easier to troubleshoot and replicate in our labs.

thombec, 2024-09-05:
Here the the screenshot from the log:
[image]
and from the sys info:
[image]
[image]

proddy, 2024-09-06:
I see the max alloc is < 50KB so MQTT Publishing is temporary stopped, which accounts for the missing data.  You said it started with dev-29. Can you remember in which version was it working perfectly so we can trace back the changes?

@MichaelDvP this is maybe the change we made to switch from async to sync for Mqtt with boards without PSRAM (like the older S32's).

MichaelDvP, 2024-09-06:
> @MichaelDvP this is maybe the change we made to switch from async to sync for Mqtt with boards without PSRAM (like the older S32's).

No, mqtt was sync all the time, now it's async only for systems with psram, so no change for this system.

But since 3.5.1 we have added a lot of entities, maybe to much for this system. Without support-info we can't say.
@thombec Go to help page, download support info and attach it here in the `add files` field.

thombec, 2024-09-06:
attached you will find the support info.

[emsesp_system_info.txt](https://github.com/user-attachments/files/16907144/emsesp_system_info.txt)

The last version (before I canged to 3.7.0-dev29) was 3.5.1 - that works fine. I left out 3.6.x.

MichaelDvP, 2024-09-06:
We have added a lot of new entities consuming memory. Your S32 have limited ram, so it can not handle it. 
You have 142 entities for boiler and  68 for thermostat.
Update to latest dev34.
- Go to customization select boiler and click `remove from memory` for all entities without values
- Same for thermostat
- save, restart.
If you still get mqtt errors, remove also entities with values you don't need.

proddy, 2024-09-06:
@MichaelDvP shall we make a new option in the Customization web page to automatically select all entities without values and do the 'remove' function?

thombec, 2024-09-06:
ok - done

without deleting entities I get now this picture:
[image]

and after deleting it looks very simmilar:
[image]

in sum it looks than the ammount of errors is now rising in comparison to the last version dev33. I deleted in sum 27 entities. The "quality" of the MQTT messages is much lower in the current version dev34 in comarison to dev33:

[image]

one point I recognized is, that the "deleted" entities are at the bottom of the list and in the list again. Here the both entities at the top I deleted -> they are at the bottom and again in the list above (see hc1/remotetemp and hc1/remotehum).

[image]

In the overview the ammount of entities is not reduced:
[image]

Before the change from version 3.5.1 to 3.7.x the amount of entities was equal (Boiler~134 now 146).

Here is my current protocol:  
[emsesp_system_info(1).txt](https://github.com/user-attachments/files/16910980/emsesp_system_info.1.txt)

proddy, 2024-09-07:
> Before the change from version 3.5.1 to 3.7.x the amount of entities was equal (Boiler~134 now 146).

Would you happen to have an old "system info" dump from 3.6.5 so we can compare? I will also grab an older S32 and see I can simulate your environment.

If you're a heavy user and plan later to use other features like the scheduler, HA, modbus then maybe see if you can upgrade to a newer S3 board?

thombec, 2024-09-07:
Hi, I flashed now version 3.6.5:

the ammount of entities was at the old level (my own entities for cooling mode are not implemented):
[image]

and the MQTT-Bus works fine:
[image]

Here is the system protocoll with 3.6.5:
[emsesp_system_info(3).txt](https://github.com/user-attachments/files/16918496/emsesp_system_info.3.txt)

Back to 3.7.0. dev 34 :-(
[image]

here is the protocol after reflash to 3.7.0.dev34:

[emsesp_system_info(4).txt](https://github.com/user-attachments/files/16918505/emsesp_system_info.4.txt)

MichaelDvP, 2024-09-07:
Interresting, the memory with 3.7. is a bit better. The memory check has changed due to PSRAM support, we check now for freeCaps, but maybe a bit to conservative.
Also for nearly same uptime of ~400 sec there are 379 publisheds for 3.6.5 and 1900 tries for 3.7.0. Seems to be the HA-discovery retrys on every publish. Can you please test with 3.7: 
- disable HA-discovery
- set the publish times to a larger value, e.g. 60 sec instead of 10. (most values changes on the bus  only once a minute, you'll miss very little). (check with enabled HA-dscovery)
- test [this build](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test) with memory check set to 50k (instead of 60)

thombec, 2024-09-07:
ok,
after disabeling HA-discovery no MQTT errors came up - but the entities in HA starts to disappear. I enabled HA-discovering an the  error:
[image]
came up again.

next test: publish time 10-60 sec:
[image]

now the errors occure only every minute - but the ammount of erros seems to be the same.
Here is the corrent log after this test:
[emsesp_system_info(5).txt](https://github.com/user-attachments/files/16918972/emsesp_system_info.5.txt)
I set the value back to 10 sec and implement your new build.
After installing and rebooting I get again many errors:
[image]
The ammount of entities is now lower than in the version before - the deleted entities are gone in the list.

here is the latest log: 
[emsesp_system_info(6).txt](https://github.com/user-attachments/files/16919067/emsesp_system_info.6.txt)

Delta1977, 2024-09-07:
One quick question : 
What hardware and Mqtt broker is on ha side? 

If I read the configuration right you set publish as individual topics. 

As in dev version there are more Matt messages to publish. Maybe your mqtt broker can't get up with the amount of messages

MichaelDvP, 2024-09-08:
> The ammount of entities is now lower than in the version before - the deleted entities are gone in the list.

They will come back, but with these fails it takes a while.

I've change the testbuild to 40k limit (please re-download and test), if this also gives mqtt fails, i'll make a debug build with advanced logging. For now i can't see any other reason than the memory check, but the large count of messages makes me wonder.

thombec, 2024-09-08:
On HA side im working with a HP Pro Desk (8 GB Ram, 128 GB SSD, Intel 4500T). That should be enough powerful to get up with the MQTT messages.

proddy, 2024-09-08:
Building on what @Delta1977 said, one thing you could try is downloading and running mosquito locally on your PC/laptop, in verbose mode to see if there are errors or retries. It's described here: https://emsesp.org/Troubleshooting/#messages-are-not-always-coming-in-via-mqtt

thombec, 2024-09-08:
Hi - I installed your new version - again the failures appear.
[image]

Here is my last protocoll after the the installation of the new version:
[emsesp_system_undefined.txt](https://github.com/user-attachments/files/16921329/emsesp_system_undefined.txt)

thombec, 2024-09-08:
@proddy : done - now the EMS-ESP is sending the MQTT data to my pc:

[image]
but in the protocoll the errors still appear...
[image]

make it sense to make a complete reset of the ems-esp board - if yes how can I do it?

MichaelDvP, 2024-09-08:
Hmm, strange, seems the memory allocation is different as expected. I've changed now the testbuild to the old check from 3.6.5. Also added the topic of faild message to the log. If there are still fails there must be another change in the mqtt publish.

MichaelDvP, 2024-09-08:
> make it sense to make a complete reset of the ems-esp board - if yes how can I do it?

You can do a factory reset in settings, but i don't think this will change  anything. 
Can you reduce the weblog buffer to minimum, that takes always a lot of memory.

thombec, 2024-09-08:
Hi - I changed theVersion to your newest beta version:
no change -> the MQTT errors occures again 
log:  
[emsesp_system_undefined(1).txt](https://github.com/user-attachments/files/16921987/emsesp_system_undefined.1.txt)

then I changed the buffer size to 25 (from 100) and:
[image]

the problem was gone!!! :-) That was it - A high buffer size at the protocoll seems to be the problem of the MQTT errors.

Heres the latest protocoll:
[emsesp_system_info(7).txt](https://github.com/user-attachments/files/16921989/emsesp_system_info.7.txt)

Thank you for help 👍

proddy, 2024-09-08:
glad you got it working. I'll think about how to optimize the web log. For starters I'll add a section in the Wiki FAQ for others that run into the same problem (MQTT publish error)

MichaelDvP, 2024-09-08:
The weblog buffer should automatic reduce on low memory, needs to be checked. 
But maybe we also make it fix to <25 on systems without psram and hide the selection on webpage. 

Now i also understand why my changes to memory check does not have an effect. I pointed to my test-build, but all system info shows "version": "3.7.0-dev.34", (my test-build shows 3.7.0.-test.34).  @thombec Seems you have always tested with the same dev-version.

proddy, 2024-09-08:
I think fix to 25 for ESP32s with no PSRAM and disable the selection in the WebUI. I can do that.

Sometimes I've experienced firmware updates going to the wrong partition by default, and had to do a `restart boot` or `restart app0` to get the right one. Still need to figure out why. We have 3 different methods of updating now, 1) PIO 2) HTTP update.py and 3) EMS-ESP's own WebUI

MichaelDvP, 2024-09-08:
boot can only be flashed via USB, but the bootloader sticks to app partitons if they contain a valid image. So flash to boot with flash-erase boots to boot-partion, without erase it boots to latest app-partiton.

proddy, 2024-09-08:
do you know how the bootloader knows whether to load ota_0/app_0 or ota_1/app_1. They are both data partitions.

MichaelDvP, 2024-09-08:
This is stored in the 8k ota_data partition.

thombec, 2024-09-08:
Hi @MichaelDvP : I used the file you provided here: [this build](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test)

I recognized that there are updates made from you due to the timings behind:
[image]

my last version I downloaded was from today ~12 o´clock:
[image]

For the installation I used the EMS-ESP internal web interface:
[image]

After uploading the EMS-ESP device was rebooting.

Do you think that this procedure don´t updated the installed software?

MichaelDvP, 2024-09-08:
Looks like UI from 3.6.5 is still active, maybe thats why the update does not work. Force a reload page Ctrl-R (or delete browser data for this page), the dev34 should look like this:
[image]

Then use the lower field to drag&drop or select the new firmware.

thombec, 2024-09-08:
ok - you are right. After deleting the browser data I see the the same version like you:
[image]

proddy, 2024-09-09:
closing this, I've updated the doc and in dev-35 will automatically reset web buffer log size to 25.