emsesp/EMS-ESP32 #1982: hc1 temporary set temperature from remote not writeable anymore.

rickroetenberg, 2024-09-06:
In the past I was able to control the "hc1 temporary set temperature from remote". This allowed me to control the cooling temperature. Since a few version this parameter isn't writeable anymore. Is there an option to allow this again?

RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410 (Thermostat)

Version EMS-ESP: 3.7.0-dev.33

MichaelDvP, 2024-09-06:
Oh, was this working for you? It was reported that it does not work if sendet from emsesp (only if sendet from remote), so the command was removed. I can add it back (and remove the comment that it does not work)

rickroetenberg, 2024-09-06:
Well it was working for me in the past at least. I can try with a custom entry, but not sure which parameters I need.

MichaelDvP, 2024-09-06:
I made a PR, just update after it is merged.

rickroetenberg, 2024-09-09:
Thanks. I can control the thermostat remote again. So I think this will allow me to enable/disable cooling again.

closed 2024-09-09 by proddy