emsesp/EMS-ESP32 #1983: Compressor cycles on/off during hot water operation after installation of EMS-ESP

Henrik-Henkel, 2024-09-06:
### PROBLEM DESCRIPTION

Since installing the EMS-ESP the compressor cycles on/off multiple times during hot water operation. Previously it would run continous for about 40 minutes an then shut off once the set temperature was reached. Now it takes almost 2 hours to reach the same water temperature... I've not changed any settings via EMS-ESP.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

```lua
{
  "System Info": {
    "version": "3.6.4",
    "platform": "ESP32-S3",
    "uptime": "000+00:11:25.222",
    "uptime (seconds)": 685,
    "free mem": 210,
    "max alloc": 195,
    "free app": 513,
    "reset reason": "Power on reset / Power on reset"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -62,
    "IPv4 address": "10.107.200.5/255.255.0.0",
    "IPv4 gateway": "10.107.0.1",
    "IPv4 nameserver": "10.0.255.100",
    "BSSID": "set",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": false,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 440,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": true
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 4760,
    "bus reads (tx)": 491,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32S3",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": true,
    "shower alert": true,
    "shower alert coldshot": 10,
    "shower alert trigger": 7,
    "hide led": false,
    "notoken api": false,
    "readonly mode": true,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 15
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "CS6800i/WLW176i",
      "device id": "0x08",
      "product id": 8,
      "version": "00.00",
      "entities": 149,
      "handlers received": "0xBF 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x04AE 0x04AF",
      "handlers pending": "0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A",
      "handlers ignored": "0xEF 0xF7 0xF6 0xF9 0x0497 0x04A0 0x049F 0x04AB 0x17 0x04A9 0x02D6 0x04A8 0x04AA 0x04AC 0x04 0xFF06 0xFE07 0xFF07 0xFE06 0x01 0xEB 0x2D 0x03 0xC6 0xC7 0x02E0 0x04A7 0x04AD 0x0240 0x04B4 0x04CA 0x04CB 0x02CC"
    },
    {
      "type": "thermostat",
      "name": "Rego 3000/UI800/WSW196i/BC400",
      "device id": "0x10",
      "product id": 253,
      "version": "47.09",
      "entities": 58,
      "handlers received": "0x06 0x02BA 0x02B0 0x029C 0x0472 0x02BB 0x02B1 0x029D 0x0473 0x02BC 0x02B2 0x029E 0x0474 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02A7 0x02A8 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E",
      "handlers ignored": "0xF7 0x0165 0x0166 0x0167 0x0168 0x0169 0x016A 0x016B 0x016C 0x01 0x016E 0x1104 0x02FF 0x0AF7 0xF9 0x0AF8 0x0AF9 0x0AFA 0x0B01 0x0B15 0x0B16 0x0B17 0x0B18 0x0B0B 0x0B0C 0x0B0D 0x0B0E 0x0B0F 0x0B10 0x0B11 0x0B12 0x0593 0xE7 0x02E0 0x02EA 0x0507 0x0508 0x0509 0x050A 0x061E 0x02C3 0x059D 0x02C4 0x02C5"
    },
    {
      "type": "gateway",
      "name": "WiFi module",
      "device id": "0x48",
      "product id": 252,
      "version": "08.01",
      "entities": 0,
      "handlers ignored": "0xF7 0x0CBC 0x0CBF 0x01 0x0CC3 0x0CBD 0xF6 0x1259 0x125A 0x0CC2 0x1258 0x2040"
    }
  ]
}
```

### TO REPRODUCE

Simply connect the EMS-ESP to the system via service jack

### EXPECTED BEHAVIOUR

Read out information without disturbing normal operation

### SCREENSHOTS

Power consumption (2 normal days, then installed EMS-ESP)
[image]

MichaelDvP, 2024-09-06:
Normally ems-esp does not change anything on the heating, it just reads settings and writes only if you command it.
There are a few special functions that could write: `forceHeatingOff` disables the heating directt on the boiler and repeats this command automatically. I your case it is the shower alert setting that stops dhw heating after the set time (coldshot). In a buffer system the shower timer does not make sense, but a few people use it to measure the buffer loading time, so it can be set also on buffered systems. 3. special function that can write is the scheduler that can write commands to the bus.

Disable the shower alert and the issue is gone.

@proddy BTW: Im missing the forcedHeatingOff setting in the system info. We should add this info because it could lead to similar issues.

proddy, 2024-09-06:
> @proddy BTW: Im missing the forcedHeatingOff setting in the system info. We should add this info because it could lead to similar issues.

I'll add it. I'm also adding some more partition/boot details too.

Henrik-Henkel, 2024-09-07:
Thank you Michael for the short explanation. The issue is gone now as expected.
Maybe I shouldn't have copied the settings from a tutorial without checking them first... My bad.

closed 2024-09-07 by Henrik-Henkel

proddy, 2024-09-07:
> > @proddy BTW: Im missing the forcedHeatingOff setting in the system info. We should add this info because it could lead to similar issues.
> 
> I'll add it. I'm also adding some more partition/boot details too.

done, in dev-34