emsesp/EMS-ESP32 #1996: HM200 / Missing Enegery Meters in v3.6.5

gruessung, 2024-09-09:
### PROBLEM DESCRIPTION

New energy meters for HM200 (Device 0x53) are missing, see #1463

I have a HM200 with Condens 7400iF heatpump and gas boiler.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ X] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ X] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [X ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{
  "System Info": {
    "version": "3.6.5",
    "uptime": "000+07:23:23.676",
    "uptime (seconds)": 26603,
    "platform": "ESP32",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "free mem": 147,
    "max alloc": 87,
    "used app": 1596,
    "free app": 4100,
    "partition": "app0",
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "Ethernet",
    "hostname": "ems-esp",
    "TxPower setting": 78,
    "static ip config": false,
    "enable IPv6": true,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 10593,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 2,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 413,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 93370,
    "bus reads (tx)": 9049,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 3,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32V2",
    "locale": "de",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": true,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 100,
    "web log buffer": 100
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GC7000F",
      "device id": "0x08",
      "product id": 132,
      "version": "02.09",
      "entities": 77,
      "handlers received": "0xBF 0xC2 0x14 0x15 0x1C 0xD1 0xE3 0xE4 0xE5 0xE9 0x04",
      "handlers fetched": "0xE6 0xEA 0x28",
      "handlers pending": "0x10 0x11 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0xE7 0x02E0 0x02EA 0x08E4 0xF7 0xD8 0x0F52 0x02D6 0x2E 0x3B"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "74.03",
      "entities": 50,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E 0xBB 0x023E",
      "handlers ignored": "0xBF"
    },
    {
      "type": "heatpump",
      "name": "Hybrid Manager HM200",
      "device id": "0x53",
      "product id": 248,
      "version": "05.02",
      "entities": 29,
      "handlers received": "0x09A0 0x099B 0x099C",
      "handlers fetched": "0x0998 0x0999",
      "handlers pending": "0x047B 0x04AE 0x04AF",
      "handlers ignored": "0x099A 0x099D 0x099E 0x099F 0x09A2 0x09A3 0x09A4 0x09A5 0x09A6 0xBF 0x012E 0x013B"
    },
    {
      "type": "gateway",
      "name": "KM200/MB LAN 2",
      "device id": "0x48",
      "product id": 189,
      "version": "04.08",
      "entities": 0,
      "handlers ignored": "0xF9 0x0CBD"
    }
  ]
}

```

proddy, 2024-09-09:
can you check v3.7.0?

gruessung, 2024-09-09:
@proddy 
of course, here is /api/system/info with current dev version:

```lua
{
  "system": {
    "version": "3.7.0-dev.34",
    "uptime": "000+00:01:16.734",
    "uptimeSec": 76,
    "platform": "ESP32",
    "cpuType": "ESP32-D0WD-V3",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "freeMem": 152,
    "maxAlloc": 91,
    "freeCaps": 4158,
    "usedApp": 1691,
    "freeApp": 4005,
    "partitionBootRunning": "app1/app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": false,
    "model": ""
  },
  "network": {
    "network": "Ethernet",
    "hostname": "ems-esp",
    "TxPowerSetting": 78,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tzLabel": "Europe/Berlin"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 202,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTConnects": 1,
    "enabled": true,
    "clientID": "ems-esp",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 3,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": true,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 10,
    "publishTimeOther": 10,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 0,
    "temperatureSensorReads": 0,
    "temperatureSensorFails": 0,
    "analogSensors": 0,
    "analogSensorReads": 0,
    "analogSensorFails": 0
  },
  "api": {
    "APICalls": 2,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "HT3",
    "busTelegramsReceived": 412,
    "busReads": 103,
    "busWrites": 0,
    "busIncompleteTelegrams": 0,
    "busReadsFailed": 0,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "E32V2",
    "locale": "de",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 180,
    "showerAlert": false,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": true,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": true,
    "telnetEnabled": true,
    "maxWebLogBuffer": 100,
    "webLogBuffer": 100,
    "modbusEnabled": false,
    "forceHeatingOff": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "GC7000F",
      "deviceID": "0x08",
      "productID": 132,
      "brand": "Junkers",
      "version": "02.09",
      "entities": 79,
      "handlersReceived": "0xC2 0x15 0x1C 0xD1 0xE3 0xE4 0xE5 0xE9 0x04",
      "handlersFetched": "0x14 0xE6 0xEA 0x28",
      "handlersPending": "0xBF 0x10 0x11 0x18 0x19 0x1A 0x35 0x34 0x2A 0x2E 0x3B",
      "handlersIgnored": "0x08E4 0xE7 0x02E0 0x02EA 0x02D6 0xD8 0x0F52"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "deviceID": "0x10",
      "productID": 158,
      "brand": "",
      "version": "74.03",
      "entities": 55,
      "handlersReceived": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x023A 0x0240 0x0269",
      "handlersPending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E"
    },
    {
      "type": "heatpump",
      "name": "Hybrid Manager HM200",
      "deviceID": "0x53",
      "productID": 248,
      "brand": "",
      "version": "05.02",
      "entities": 29,
      "handlersReceived": "0x09A0 0x099B 0x099C",
      "handlersFetched": "0x0998 0x0999",
      "handlersPending": "0x047B",
      "handlersIgnored": "0x09A6 0x099A 0x099D 0x099E 0x099F 0x09A2 0x09A3 0x09A4 0x09A5"
    },
    {
      "type": "gateway",
      "name": "KM200/MB LAN 2",
      "deviceID": "0x48",
      "productID": 189,
      "brand": "",
      "version": "04.08",
      "entities": 0,
      "handlersIgnored": "0xF9"
    }
  ]
}
```

proddy, 2024-09-09:
i don't see 04AF - need to look into this

gruessung, 2024-09-09:
pls let me know, if you need more information

proddy, 2024-09-10:
does `read 53 4af` or `read 53 4ae` in a telnet console return anything?

gruessung, 2024-09-10:
> does `read 53 4af` or `read 53 4ae` in a telnet console return anything?

```
ems-esp:$ read 53 4af
000+09:36:53.122 N 1: [emsesp] heatpump(0x53) -W-> Me(0x0B), HPMeters(0x04AF), data: <empty>
```

```
ems-esp:$ read 53 4ae
000+09:37:15.174 N 2: [emsesp] heatpump(0x53) -W-> Me(0x0B), HPEnergy(0x04AE), data: <empty>
```

Nope :(

proddy, 2024-09-10:
That's explains it. You'll need to help us find the right information. https://emsesp.org/Troubleshooting/#im-missing-certain-data-from-an-ems-device

gruessung, 2024-09-10:
> That's explains it. You'll need to help us find the right information. https://emsesp.org/Troubleshooting/#im-missing-certain-data-from-an-ems-device

I'm confused. In 3.6.5. 4af and 4ae are added for HM200 :/
I'll take a look into the telegram.

proddy, 2024-09-10:
Yes I can see from your 3.6.5 settings it has `handlers pending: "0x047B 0x04AE 0x04AF"`

maybe @MichaelDvP has an idea

gruessung, 2024-10-16:
@proddy  @MichaelDvP  Is there any cli / gui which will help to find the correct telegram? Honestly i'm a bit confused with the whole output.

proddy, 2024-10-16:
@gruessung can you show examples of what data you're looking for? Where is this being displayed now?

gruessung, 2024-10-16:
@proddy i connect via telnet to ems-esp and watch unknown telegrams and get these:

```003+23:54:39.929 N 157: [emsesp] heatpump(0x53) -B-> all(0x00), ?(0x099F), data: 05 7E 40 00 00 00 00 03 00 00 7B 00 00 78 00 00 79 00 01 A1 00 01 2F 00
003+23:54:40.232 N 158: [emsesp] heatpump(0x53) -B-> all(0x00), ?(0x099F), data: 00 81 00 01 3A 00 00 6A 00 01 AB 00 01 3D 02 80 00 5A 01 86 01 40 00 00 DD (offset 24)
003+23:54:40.865 N 159: [emsesp] heatpump(0x53) -B-> all(0x00), ?(0x099F), data: 00 00 00 00 (offset 49)
003+23:54:41.712 N 160: [emsesp] heatpump(0x53) -B-> all(0x00), ?(0x09A2), data: 00 00 00 00
003+23:54:41.928 N 161: [emsesp] heatpump(0x53) -B-> all(0x00), ?(0x09A3), data: 04 03 01 00 20 01 1E 00 00 00 00 00 00 01 01 00 02 (offset 2)
003+23:54:42.696 N 162: [emsesp] heatpump(0x53) -B-> all(0x00), ?(0x09A4), data: 00 3C 00 00 01 00 00 00 00 00 37 01 00 00 00 00 00 00 00 00 00 00 00 00 00
003+23:54:42.878 N 163: [emsesp] heatpump(0x53) -B-> all(0x00), ?(0x09A4), data: 02 (offset 25)
003+23:54:43.485 N 164: [emsesp] heatpump(0x53) -B-> all(0x00), ?(0x09A5), data: 00 20 01 1E 00 00 00 00 0A 01 00 01 03 03 00 5A 00 20 00 00 00 00 20 20 14
003+23:54:43.672 N 165: [emsesp] heatpump(0x53) -B-> all(0x00), ?(0x09A5), data: 00 00 00 (offset 25)
003+23:54:44.258 N 166: [emsesp] heatpump(0x53) -B-> all(0x00), ?(0x09A6), data: 02 01 00 00 02 1F 01 1F 01 01 03 00 3F 00 00 00 1C 64 00 3C 00 01 00 00
003+23:54:44.449 N 167: [emsesp] heatpump(0x53) -B-> all(0x00), ?(0x09A6), data: 01 FB 00 00 00 (offset 24)
003+23:54:47.206 N 168: [emsesp] heatpump(0x53) -B-> all(0x00), ?(0x09A6), data: 01 FC (offset 24)```

My issue: i'm not able to interpret the data to compare this with my real thermostat display.

proddy, 2024-10-16:
That's fine, we're just interested in what data you expect to see. Can you share photos of the boiler's dashboard? Then we can start going hunting and matching the two, if it exists or not

gruessung, 2024-10-16:
I will provide you the latest pictures tomorrow at the latest when i'm home.

gruessung, 2024-10-16:
[image]

[image]

[image]

I'm looking for this data.

proddy, 2024-10-17:
Could you upgrade to 3.7.0 - then we can compare apples to apples

gruessung, 2024-10-17:
@proddy I'm already on current dev version

proddy, 2024-10-17:
Try this build: [EMS-ESP-3_7_0-dev_46-ESP32-16MB+.zip](https://github.com/user-attachments/files/17415134/EMS-ESP-3_7_0-dev_46-ESP32-16MB%2B.zip)

gruessung, 2024-10-17:
> Try this build: [EMS-ESP-3_7_0-dev_46-ESP32-16MB+.zip](https://github.com/user-attachments/files/17415134/EMS-ESP-3_7_0-dev_46-ESP32-16MB%2B.zip)
> 
> 

Unfortunately no changes.

proddy, 2024-10-17:
no Gesamtenergie, Energie Heizen or Energie Kühlen in the Boiler device entities?

gruessung, 2024-10-17:
Nope. Neither in the boiler nor in the heatpump.

MichaelDvP, 2024-10-17:
Start a full log, go to the thermostat and call the screens shown above while logging, note the values (or make a photo), 
Send the log and the values, maybe we find the values somewhere in log.

tp1de, 2024-10-21:
The energy consumption is calculated by the km200 gateway.
No telegrams available. 
For gas boilers the nrg... entities have similar content.

tp1de, 2024-10-21:
The energy consumption is calculated by the km200 gateway.
The values should come from the hybrid manager hm200.
No telegrams for consumption is available on my gas boiler system. 
For gas boilers the nrg... entities have similar content.

Sbried, 2024-10-22:
Looks like you are looking for the energy consumption for the heat pump only, when I look at the photos.
Maybe you try to read the data types `<dt>` 0x0494 or 0x0495 (energy heat pump) or 0x012B or 0x012E (energy hybrid systems) from all your devices `<dev>` 0x08, 0x10 and 0x53 with
`read <dev> <dt>`
Maybe somewhere is an output.

gruessung, 2024-11-05:
@Sbried 
Here my results:

```
** Device: Wärmepumpe: MH200/Hybrid Manager HM200 (DeviceID:0x53, ProductID:248, Version:05.02) **

ems-esp:$ read 0x53 0x0494
000+00:27:40.694 N 1: [emsesp] heatpump(0x53) -W-> me(0x0B), ?(0x0494), data: <empty>
ems-esp:$ read 0x53 0x0495
000+00:27:46.318 N 2: [emsesp] heatpump(0x53) -W-> me(0x0B), ?(0x0495), data: <empty>
ems-esp:$ read 0x53 0x012B
000+00:27:53.856 N 3: [emsesp] heatpump(0x53) -W-> me(0x0B), ?(0x012B), data: <empty>
ems-esp:$ read 0x53 0x012E
000+00:27:56.300 N 4: [emsesp] heatpump(0x53) -W-> me(0x0B), ?(0x012E), data: 53 F8 00 00 00 00 00 00 00 00 00 00 01 B1 F8 00 00 39 96
```

```
** Device: Thermostat: CW400/RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410 (DeviceID:0x10, ProductID:158, Version:74.03) **

ems-esp:$ read 0x10 0x0494
000+00:30:28.013 N 5: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0494), data: <empty>
ems-esp:$ read 0x10 0x0495
000+00:30:30.555 N 6: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0495), data: <empty>
ems-esp:$ read 0x10 0x012B
000+00:30:34.787 N 7: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x012B), data: <empty>
ems-esp:$ read 0x10 0x012E
000+00:30:36.136 N 8: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x012E), data: 00 00 00 00 00 00 00 00 00 01 B1 F8 00 00 39 96 (offset 3)
000+00:30:36.177 N 9: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x012E), data: <empty> (offset 19)
```

```
** Device: Kessel: Junkers MX2/GC7000F (DeviceID:0x08, ProductID:132, Version:02.09)**

ems-esp:$ read 0x08 0x0494
000+00:31:31.469 N 10: [emsesp] boiler(0x08) -W-> me(0x0B), ?(0x0494), data: <empty>
ems-esp:$ read 0x08 0x0495
000+00:31:33.219 N 11: [emsesp] boiler(0x08) -W-> me(0x0B), ?(0x0495), data: <empty>
ems-esp:$ read 0x08 0x0012B
000+00:31:39.044 N 12: [emsesp] boiler(0x08) -W-> me(0x0B), ?(0x012B), data: <empty>
ems-esp:$ read 0x08 0x0012E
000+00:31:40.267 N 13: [emsesp] boiler(0x08) -W-> me(0x0B), ?(0x012E), data: <empty>
```

Could you convert the data blocks to readable strings?

gruessung, 2024-11-05:
@Sbried  I also look for die COP(JAZ) and energy outcome from my heatpump.
For energy consumption i have an extra power meter into my power installation.

Sbried, 2024-11-09:
The data type 0x012E has some content and it can be as follows:
The hybrid manager answers with his device ID 0x53 and his product-ID 0xF8(=248).
The following "00" are about fuel consumption for CH (index 3-6) and DHW (index 7-10) that do not apply for heat pumps.
Index 11-14 is "00 01 B1 F8" which gives111096 (in 0,1 kWh) for _cumulated_ CH electricity consumption. Did you consume ~11109 kWh for heating?
Index 15-18 is "00 00 39 96" which gives 14742 (in 0,1 kWh) for _cumulated_ DHW electricity consumption. Didi you consume ~1474 kWh for hot water? 
Or ~12583 kWh in total, if you can´t split?
The same data are sent by the thermostat, but without the device-ID and the product-ID.

gruessung, 2024-11-09:
[image]

Yes, that's sound plausible. 

Did you find any data in the following direction?:
[image]

Sbried, 2024-11-11:
You may try `read 0x53 0x13B`.
If it works, the bytes 3-6 of the answer is the CH heat generation and bytes 7-10 are the DHW heat generation

gruessung, 2024-11-11:
> You may try `read 0x53 0x13B`. If it works, the bytes 3-6 of the answer is the CH heat generation and bytes 7-10 are the DHW heat generation

Here the output:

```ems-esp:$ read 0x53 0x13B
005+16:01:15.670 N 1: [emsesp] heatpump(0x53) -W-> me(0x0B), ?(0x013B), data: 53 F8 00 00 04 CC 05 00 00 90 44```

gruessung, 2024-11-11:
@Sbried 
That looks really great :)

```
00 04 CC 05
31437,3 kwh

00 00 90 44
3693,2 kwh
```

Sbried, 2024-11-11:
Yes, maybe they can impelement this for __hybrid systems__. 

For a better overview here again the two data types:

*DT 0x012E (cumulated energy consumption)*
Index 0: Device-ID
Index 1: Product-ID
Index 2: Product-ID
Index 3-6= CH fuel consumption [0,1kWh]
Index 7-10= DHW fuel consumption [0,1kWh]
Index 11-14= CH electricity consumption [0,1kWh]
Index 15-18= DHW electricity consumption [0,1kWh]

*DT 0x013B (cumulated energy generation)*
Index 0: Device-ID
Index 1: Product-ID
Index 2: Product-ID
Index 3-6 = CH electricity generation [0,1kWh]
Index 7-10 = DHW electricity generation [0,1kWh]

gh47110815, 2024-11-23:
In the 2nd post above (https://github.com/emsesp/EMS-ESP32/issues/1996#issuecomment-2468080648) if I reproduce these 4 entities on my ems-esp running 3.7.1-dev.10 then if I click on one of these 4 entries and try to change the factor from 1 (default), to 0,1 it does not work. nothing happens. Is it possible that UINT32 can not have a factor < 1 ?  So how to set the right factor? Do I have to fix this in the automation software (home assistant with helper templates?) Thanks

MichaelDvP, 2024-11-25:
Please test with https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

gh47110815, 2024-11-25:
Works at my installation with 3.7.1-dev.12 ! (factor < 1)
THANK YOU!

closed 2024-11-26 by proddy

gruessung, 2024-12-04:
Thanks for you support! <3