emsesp/EMS-ESP32 #2011: FW120 : report wrong value for room influence factor

xavp06, 2024-09-13:
the FW120 "hc1 room influence factor" is showing 3% in EMS-ESP, but the real FW120 thermostat menu is showing 30%. I guess just a factor to be corrected ?

xavp06, 2024-09-13:
Thermostat.cpp, in Thermostat::set_roominfl_factor : the value is *10. I could imagine that is it is a percentage, it should be *100

MichaelDvP, 2024-09-14:
A bit confusing. For RC300 the factor in emsesp  is DIV10 without uom. For Junkers (FW120) we have uom percent and no scaling factor. To show 30% we have to add MUL10, but a percentage range > 1000% makes no sense.

xavp06, 2024-09-15:
using v3.7.0-dev32
[image]

proddy, 2024-09-15:
thanks. we'll figure this one out...

xavp06, 2024-09-15:
and reporting the same to my mqtt broker and influxdb database
[image]

commit a6ca1313 (2024-09-16): roomifl_factor for FW120 to mul10, fix #2011
closed 2024-09-16 by proddy