emsesp/EMS-ESP32 #2016: Ethernet not always starting after a wam restart

proddy, 2024-09-16:
I found while testing the newer BBQKees E32 V2's, after a firmware upload via the WebUI the EMS-ESP wouldn't always reconnect to the Ethernet. The serial log would show repeatedly:

```
000+00:04:31.581 I 72: [emsesp] Local IPv6 (Eth)=fe80:0000:0000:0000:e665:b8ff:fe2c:62eb
000+00:04:32.274 W 73: [emsesp] Ethernet disconnected
000+00:04:41.581 I 74: [emsesp] Local IPv6 (Eth)=fe80:0000:0000:0000:e665:b8ff:fe2c:62eb
000+00:04:42.274 W 75: [emsesp] Ethernet disconnected
000+00:04:51.581 I 76: [emsesp] Local IPv6 (Eth)=fe80:0000:0000:0000:e665:b8ff:fe2c:62eb
000+00:04:52.274 W 77: [emsesp] Ethernet disconnected
000+00:05:01.581 I 78: [emsesp] Local IPv6 (Eth)=fe80:0000:0000:0000:e665:b8ff:fe2c:62eb
000+00:05:02.274 W 79: [emsesp] Ethernet disconnected
```

I'll debug and see what is happening. I suspect the LAN8720's PHY-clock (crystal oscillator) needs some time to reset, so maybe just adding some delay before the `ETH.begin()` would help.

```
ems-esp:# show system

System:
 Version: 3.7.0-dev.37
 Platform: ESP32 (ESP32-D0WD-V3)
 Model: BBQKees Gateway Model EMS Gateway E32 V2 v2.0/2024071
 Partition: app0
 Language: en
 Board profile: E32V2
 Uptime: 000+00:04:22.163
 Base MAC Address: E4:65:B8:2C:62:E8
 SDK version: v4.4.7-dirty
 CPU frequency: 240 MHz
 Free heap/Max alloc: 168 KB / 107 KB
 App used/free: 2330 KB / 3366 KB
 FS used/free: 24 KB / 2024 KB
 PSRAM size/free: 4093 KB / 4074 KB

Network:
 WiFi MAC address: E4:65:B8:2C:62:E8
 WiFi Network: not connected

Syslog:
 Syslog: disabled
```

MichaelDvP, 2024-09-16:
Check the network settings file from download->settings, Is the board profile still on default? Then the new networksettingsservice calls ETH.begin for testing and system.network init calls it second time. Maybe this is a pingpong (before the networksettings stores the board and reboots to ensure that ETH.begin is only called once).

proddy, 2024-09-16:
> Check the network settings file from download->settings, Is the board profile still on default? Then the new networksettingsservice calls ETH.begin for testing and system.network init calls it [...]

Yes, I checked that. That was my first reason except this has NVS data so the first ETH.begin() is skipped in `WebSettingsService::update()`.

proddy, 2024-09-17:
Added this before the ETH.begin() call in`System::network_init()`. I'll monitor to see if it happens again.

```C
    if (eth_power_ != -1) {
        pinMode(eth_power_, OUTPUT);
        digitalWrite(eth_power_, LOW);
        delay(500);
        digitalWrite(eth_power_, HIGH);
    }
```

closed 2024-09-21 by proddy