emsesp/EMS-ESP32 #2030: Can we add Australian models?

VlastiBroucek, 2024-09-20:
### DESCRIPTION

This is not a bug or issue! This is just potential improvement, extension... I am in Australia and I doubt there are many users here. 

My boiler is identified as:

_boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:12.16)_

It is actually Bosch Condens 5000 W (ZWB 37-2 A23)

My room thermostat is identified as:

_thermostat: RC200/CW100 (DeviceID:0x18, ProductID:157, Version:43.02)_

It is actually Bosch CR50

Controller is correctly identified as:

_controller: HT3 (DeviceID:0x09, ProductID:95, Version:12.16)_

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X ] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [X ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

**(Please remember to close the issue when it has been addressed)**

MichaelDvP, 2024-09-20:
We can add, but it's just a name. We use the product-id to identify devices. 
In v3.7.0 you can edit the name on the customization page.

VlastiBroucek, 2024-09-20:
I agree. I was just thinking that people here in Australia would maybe use it more if they knew that AU branded boilers were supported. When I bought from BBQKees I just took the risk. Being in IT since 1985 I knew I would probably have enough skills to get it working together with you guys, but not everyone can do that.

It's your call, happy whichever way you decide.

BTW, many options in the thermostat are called differently here too, so perhaps not worth of fiddling with it.

MichaelDvP, 2024-09-20:
Yes it's good to know the compatibility and names of the different devices. I've added the names. 
BBQKees already have your devcies on his list https://bbqkees-electronics.nl/ems-device-compatibility/ I think this is  where new users looks first.
> BTW, many options in the thermostat are called differently here too, so perhaps not worth of fiddling with it.

Also in customization you can rename the options long names. Bosch changes the names often with nearly every device. 
Maybe we should ask users to send us the customization files and collect them, named by device, in a folder here. Then other users can upload a `cr50_en.json` and have the right names.

VlastiBroucek, 2024-09-20:
Ha, didn't notice he already put it there... Mine is SWB - includes DHW.

Good idea. I will have a look what I can do for CR50. Is there an example json file somewhere? Or just do something like:
CW100 | CR50
hc1 comfort temperature | hc1 heating temperature
hc1 eco temperature | hc1 setback temperature

TBH, I am not sure if this comes from boiler/HT3 or thermostat?

MichaelDvP, 2024-09-20:
> Is there an example json file somewhere? 

On the customization page edit the entities (name and/or min/max)., save, then go to settings->download and download the customization file. This is the json we need.

VlastiBroucek, 2024-09-20:
Will do. When do you expect 3.7 to come out to prod? Since I use this more for monitoring than anything else right now (so, no damage if something does not work) I might install the dev and do it under it...

closed 2024-09-22 by proddy