emsesp/EMS-ESP32 #2038: Logavent HRV156 - Unrecognized EMS device (deviceID 0x51, productID unknown)

oligh, 2024-09-21:
### DESCRIPTION

Hi, I have a Buderus Logavent HRV156 ventilation system. It is very similar to Logavent HRV176 (case #1172) but seems to be different in details. The behaviour is a little bit strange, right after power-up the device is detected as Logavent HRV176 with some temperature readings but without any option to change anything. Whenever I restart ems-esp, it is unable to detect the ventilation 
[image]
device any more (ems-esp is connected via service plug). 

### REQUESTED INFORMATION

{
  "system": {
    "version": "3.7.0-dev.38",
    "uptime": "000+07:26:07.706",
    "uptimeSec": 26767,
    "platform": "ESP32",
    "cpuType": "ESP32-D0WD-V3",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "freeMem": 166,
    "maxAlloc": 107,
    "freeCaps": 116,
    "usedApp": 1689,
    "freeApp": 295,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": false,
    "model": ""
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-esp-logavent",
    "RSSI": -80,
    "TxPowerSetting": 0,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tzLabel": "Europe/Amsterdam"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 1811,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTConnects": 2,
    "enabled": true,
    "clientID": "ems-esp-logavent",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 3,
    "base": "ems-esp2",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": true,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
[... 123 lines cut ...]

00+00:05:20.696 TRACE 118: [telegram] Incomplete Rx: 0B D1 00 FF E6
000+00:06:09.035 ERROR 119: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B D1 02 00 FF
000+00:06:09.035 TRACE 120: [emsesp] 51(0x51) -W-> me(0x0B), Version(0x02), data: <empty>
000+00:06:09.035 WARNING 121: [emsesp] Unrecognized EMS device (deviceID 0x51, no productID). Please report on GitHub.
000+00:06:09.223 TRACE 122: [emsesp] 51(0x51) -B-> all(0x00), ?(0x0575), data: 01 01 1E 1E 00 00 38 80 00 01
000+00:06:09.428 TRACE 123: [emsesp] 51(0x51) -B-> all(0x00), ?(0x0583), data: 00 F6 01 E0 FF FF FF 00 E9 FF FF FF 00 00 FB 00 F5 04 74 80 00 00 80 00
000+00:06:09.675 TRACE 124: [emsesp] 51(0x51) -B-> all(0x00), ?(0x0583), data: 80 00 (offset 24)
000+00:06:09.880 TRACE 125: [emsesp] 51(0x51) -B-> all(0x00), ?(0x0585), data: 1E 1E 00 00 00 ED 00 03 F3 93 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:06:10.055 TRACE 126: [emsesp] 51(0x51) -B-> all(0x00), ?(0x0585), data: 00 (offset 25)
000+00:06:10.272 TRACE 127: [emsesp] 51(0x51) -B-> all(0x00), ?(0x0586), data: 00 64 64 00 00 00 00 00
000+00:06:10.470 TRACE 128: [emsesp] 51(0x51) -B-> all(0x00), ?(0x0587), data: 00 00 14 00 14 0A 00 01 54 01 00 00 00 00 00 46 00 00 00 02 00 1C 00 1C
000+00:06:10.647 TRACE 129: [emsesp] 51(0x51) -B-> all(0x00), ?(0x0588), data: 11 01
000+00:06:10.889 TRACE 130: [emsesp] 51(0x51) -B-> all(0x00), ?(0x058D), data: 00 0D 4E 2B 00 0C BC 4A 00 00 00 00 00 00 00 00 00 01 F2 E5 00 0C BC 4A
000+00:06:11.066 TRACE 131: [emsesp] 51(0x51) -B-> all(0x00), ?(0x058E), data: 64 00 00
000+00:06:11.257 TRACE 132: [emsesp] 51(0x51) -B-> all(0x00), ?(0x058F), data: 00 08 00 00 00 08 00 00
000+00:06:11.446 TRACE 133: [emsesp] 51(0x51) -B-> all(0x00), ?(0x05CF), data: 01
000+00:06:11.628 TRACE 134: [emsesp] 51(0x51) -B-> all(0x00), ?(0x05D9), data: FF FF 38
000+00:06:11.805 TRACE 135: [emsesp] 51(0x51) -B-> all(0x00), ?(0x05E3), data: 01

### EXPECTED BEHAVIOUR
So basically I see 3 issues:
- Logavent HRV156 unsupported, seems to be slightly different than supported HRV176
- Detection issues after reboot
- Missing options to set levels etc., currently only temperature readings are available (see screenshot)

MichaelDvP, 2024-09-22:
Your custom board has a tx issue:
```
"busStatus": "connected, tx issues - try a different Tx Mode",
"busReads": 0,
"busReadsFailed": 205,
"busTxLineQuality": 0
```
so emsesp can't get the settings telegrams and version info.
What resistors are you using for TX current draw?

Not related, but critical:
`"RSSI": -80,`
Thats much below bad connection, that's very near "no connection".

oligh, 2024-09-22:
Hi Michael,
thanks for your response. What do you mean with resistors? It's the interface board V3.1 from BBQKees (not the complete gateway)...
Regarding WIFI: I know, it's on the roadmap :-)

MichaelDvP, 2024-09-22:
> It's the interface board V3.1 from BBQKees 

Ok, than it's something other.
Do you have connected the interface only to the HRV? I can't see boiler, controller, and thermostat in the log.
I've checked the manual https://buderus-de-de.boschtt-documents.com/download/file/file/6721846107.pdf
There is the standalone mode described, but it says that communication (module MX300) is only possible if the special thermostat  VC310 is connected. Maybe emsesp also can not work without this unit?

The TX-read from emsesp is always scrambled and there is no answer, so emsesp can only see the broadcasted telegrams, but no settings (that needs to be read) and can not change parameters.

Can you make a longer log (attach as file) and, if you have the thermostat, connect it. Also try with different tx-modes.

oligh, 2024-09-22:
yes, I have 2 separate EMS busses: 
One with WLW196i heatpump, VC300, MX300, RC100h and ems-esp-> no problems
One with HRV156, RC100h and ems-esp
I've attached the log, at theend I switched between level 1-4 (on the RC100)
[ems5.log.txt](https://github.com/user-attachments/files/17088828/ems5.log.txt)

MichaelDvP, 2024-09-22:
Why do you use seperate busses? Isn't it better to use one system?

The RC100 is now on address 0x58 and still not answering to emsesp.  The master seems to skip ems addresses, the request 
`0B D8 02 00 FF` becomes `0B D8 00 FF` and is ignored. I think in standalone mode the HRV does not support ems addressing.

But try a different emsesp address, maybe the higher at 0x49,.. and check tx errors on the status->data-traffic page. 
Also try different tx-modes.

oligh, 2024-09-22:
Yeah, you got it! With 0x49 and EMS+ it seems to work. I didn't realize that in prior tests due to the fact that I still needed to power-cycle the whole ventilation system. (I use a smart plug as workaround for this now.)
I now have additional settings for level and the RC100 is detected as well.

Regarding the 2 busses: That's the way my plumber installed it, and he refused to combine it in one bus. Maybe that's because one is EMS and one EMS+

Thank you very much!

closed 2024-09-22 by oligh