emsesp/EMS-ESP32 #2044: Thermostat FW100 wrong reading

4l4R1, 2024-09-22:
### DESCRIPTION

Thermostat FW100 value 'hc1 summer temperature' ('Heating off at outside temperature' in english documentation) in EMS-ESP is 2 times higher. For example, 36 instead of 18.
If the value is to be changed, the correct value must be given (For example, 20, then EMS-ESP shows 40).

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
 {
  "system": {
    "version": "3.7.0-dev.38",
    "uptime": "000+00:13:43.473",
    "uptimeSec": 823,
    "platform": "ESP32",
    "cpuType": "ESP32-D0WD-V3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 144,
    "maxAlloc": 103,
    "freeCaps": 4179,
    "usedApp": 2332,
    "freeApp": 3364,
    "partition": "app0",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 4093,
    "freePsram": 4070,
    "model": ""
  },
  "network": {
    "network": "Ethernet",
    "hostname": "ems-esp",
    "TxPowerSetting": 78,
    "staticIP": true,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "connected",
    "enabled": true,
    "server": "192.168.1.1",
    "tzLabel": "Europe/Tallinn"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 288,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTConnects": 1,
    "enabled": true,
    "clientID": "ems-esp",
    "keepAlive": 60,
    "cleanSession": true,
    "entityFormat": 1,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": true,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 10,
    "publishTimeOther": 10,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 0,
    "temperatureSensorReads": 0,
    "temperatureSensorFails": 0
  },
  "api": {
    "APICalls": 0,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "HT3",
    "busTelegramsReceived": 618,
    "busReads": 141,
    "busWrites": 0,
    "busIncompleteTelegrams": 1,
    "busReadsFailed": 1,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "E32V2",
    "locale": "en",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": true,
    "showerMinDuration": 180,
    "showerAlert": false,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": false,
    "telnetEnabled": true,
    "maxWebLogBuffer": 100,
    "webLogBuffer": 100,
    "modbusEnabled": false,
    "forceHeatingOff": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "deviceID": "0x08",
      "productID": 95,
      "brand": "",
      "version": "23.00",
      "entities": 65,
      "handlersReceived": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04",
      "handlersFetched": "0x14 0x16 0x33",
      "handlersPending": "0xBF 0xC2 0x1A 0x35 0x26 0x2A 0x2E 0x3B",
      "handlersIgnored": "0x23"
    },
    {
      "type": "thermostat",
      "name": "FW100",
      "deviceID": "0x10",
      "productID": 105,
      "brand": "Junkers",
      "version": "13.05",
      "entities": 24,
      "handlersReceived": "0x06",
      "handlersFetched": "0x016F 0x0165 0x01D3",
      "handlersPending": "0xA3 0xA2 0x12 0x13 0x0170 0x0171 0x0172 0x0166 0x0167 0x0168 0xBB 0x23",
      "handlersIgnored": "0xBE"
    },
    {
      "type": "controller",
      "name": "HT3",
      "deviceID": "0x09",
      "productID": 95,
      "brand": "",
      "version": "23.00",
      "entities": 0
    }
  ]
}
```

### TO REPRODUCE

Compare values only

### EXPECTED BEHAVIOUR

Same readings

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2024-09-22:
@MichaelDvP safe to add `DV_NUMOP_DIV2` to https://github.com/emsesp/EMS-ESP32/blob/9787dfc0a01c3e57e16aa539b3f299fb79c2dfd1/src/devices/thermostat.cpp#L4797 

?

MichaelDvP, 2024-09-23:
>safe to add DV_NUMOP_DIV2 to 

Yes, i think this will fix it.

proddy, 2024-09-23:
@4l4R1 try dev-39

4l4R1, 2024-09-23:
Thank you very much!

Now both displaying and changing readings are completely correct, both in the EMS-ESP user interface and in the Home assistant

closed 2024-09-23 by 4l4R1