emsesp/EMS-ESP32 #2052: call of "api/<device>/<name>/value" return JSON of the entity, not the value (for second thermostat)

harvey637, 2024-09-26:
### DESCRIPTION

fetching values via api does not work as expected with two thermostats

### TO REPRODUCE
curl http://ems-esp-s3.fritz.box/api/thermostat/hc1/currtemp/value ; echo
19.2

shows correct VALUE of "currtemp"

curl http://ems-esp-s3.fritz.box/api/thermostat/hc1/airhumidity/value ; echo
{"name":"airhumidity","fullname":"HK1 relative Luftfeuchte","circuit":"hc1","value":58,"type":"number","uom":"%","readable":true,"writeable":false,"visible":true}

shows the json of "airhumidity", but not (only) the value
The same is wrong for "rftemp", "dewtemperature","intoffset". I suspect, that the values of the second thermostat are handled wrong ... just my idea.

### EXPECTED BEHAVIOUR

that fetching the value for airhumidity only returns the value: 58,
and the others values from thermostat RC100H also return the value only.

### ADDITIONAL CONTEXT

My heatpump system was (professionally) build with a RC310 as console and with a roomthermostat RC100H.

These are my two thermostats:
```
    {
      "type": "thermostat",
      "name": "RC310 Konsole",
      "deviceID": "0x10",
      "productID": 158,
      "brand": "",
      "version": "13.07",
      "entities": 24,
      "handlersReceived": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E 0x0269",
      "handlersPending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x02F6 0x031B 0x031E",
      "handlersIgnored": "0xBF"
    },
    {
      "type": "thermostat",
      "name": "RC100H Whz",
      "deviceID": "0x38",
      "productID": 200,
      "brand": "",
      "version": "40.07",
      "entities": 4,
      "handlersReceived": "0x042B 0x047B",
      "handlersFetched": "0x0273",
      "handlersPending": "0x0A6A",
      "handlersIgnored": "0xBF"
    },
```
This is the reading of "thermostat", the values "airhumidity" and "dewtemperature" origin from the RC100H.

```
{
  "datetime": "26.09.2024 12:13",
  "intoffset": 0.0,
  "dampedoutdoortemp": 14.3,
  "hc1": {
    "seltemp": 20.0,
    "currtemp": 19.2,
    "mode": 1,
    "modetype": 1,
    "ecotemp": 14.0,
    "manualtemp": 17.0,
    "comforttemp": 20.0,
    "summertemp": 11,
    "offsettemp": 0,
    "summersetmode": 1,
    "summermode": 1,
    "program": 0,
    "coolingon": 1,
    "dewoffset": 2,
    "roomtempdiff": 1,
    "heatondelay": 4,
    "heatoffdelay": 1,
    "coolstart": 31,
    "coolondelay": 1,
    "cooloffdelay": 2,
    "rftemp": 19.2,
    "dewtemperature": 12.3,
    "airhumidity": 58,
    "intoffset": 0.0
  },
  "dhw": {}
}
```
My guess is, that the last four items (rftemp, dewtemperature, airhumidity, intoffset) are from the RC100H device.

In the iobroker adapter all values are correct, I only see this error in the api call.

Hope this helps to find the error,
cu
Harvey

MichaelDvP, 2024-09-26:
> Hope this helps to find the error,

Yes it helps, thanks. I'll make a PR to fix.

harvey637, 2024-09-26:
Hi, fast response, great!
looks like fixed for me in dev40 !

to be closed (whenever you want, proddy)
cu
Harvey

proddy, 2024-09-26:
thanks for finding this one!

closed 2024-09-26 by proddy