emsesp/EMS-ESP32 #2056: minor issue: search in customizations only searches the english/original names (not the tranlated ones)

harvey637, 2024-09-26:
Hi, this is very minor, but noticable:

if set to a language like "DE" in customizations the german names are displayed  (even in bold), ok.
But searching only searches the smaller "original" names.
It is a bit strange at first sight, that searching for all "WWK" (warm water stuff ...) shows: nothing.
But as I typed the first "W" all WarmWater stuff was shown. With the second "WW" no results.

Than I tried the "dhw" for the english names (or the "original" technical names) everything works as expected.

[image]

Not a big deal, but should be explained to novice users or as a popup ... 
No code change requested, just noticed!
cu
Harvey

proddy, 2024-09-26:
That's actually a really good point, thanks for mentioning it. The search is filtering on the short name only which is the text that is bracketed in light-blue. It should also match the name being displayed. We'll make a change. Thanks!

proddy, 2024-09-26:
it was because of the case conversion. I'll do a PR

commit 243da885 (2024-09-26): Merge pull request #2057 from proddy/dev

fixes #2056
closed 2024-09-26 by proddy