emsesp/EMS-ESP32 #2059: Nefit Moduline 3000 loses connection to external temperature sensor on HRC45 -- ERROR A11/1037

Moving4407, 2024-09-27:
### DESCRIPTION

After 2-3 days of running the outside temperature disappears from the Moduline 3000 thermostat display and the thermostat starts heating (weather-controlled program). This behavior was not present in v3.6.5. Unplugging the EMS gateway resolves the issue (the temperature appears again). This behavior does not occur on a similar setup (HRC 30 with older Moduline 3000)

### REQUESTED INFORMATION

system |  
version | "3.7.0-dev.39"
uptime | "003+00:36:42.569"
uptimeSec | 261402
platform | "ESP32S3"
cpuType | "ESP32-S3"
arduino | "ESP32 Arduino v2.0.17"
sdk | "v4.4.7-dirty"
freeMem | 194
maxAlloc | 179
freeCaps | 8336
usedApp | 2237
freeApp | 5891
partition | "app0"
resetReason | "Power on reset / Power on reset"
psram | true
psramSize | 8189
freePsram | 8143
model | ""
network |  
network | "WiFi"
hostname | "nefit-HRC45"
RSSI | -52
TxPowerSetting | 0
staticIP | false
lowBandwidth | false
disableSleep | true
enableMDNS | true
enableCORS | false
APProvisionMode | "disconnected"
APSecurity | "wpa2"
APSSID | "ems-esp"
ntp |  
NTPStatus | "connected"
enabled | true
server | "ntp.time.nl"
tzLabel | "Europe/Amsterdam"
mqtt |  
MQTTStatus | "connected"
MQTTPublishes | 82975
MQTTQueued | 0
MQTTPublishFails | 0
MQTTConnects | 1
enabled | true
clientID | "ems-esp-nefit-HRC45"
keepAlive | 60
cleanSession | false
entityFormat | 2
base | "ems-esp-nefit-HRC45"
discoveryPrefix | "homeassistant"
discoveryType | 0
nestedFormat | 1
haEnabled | true
mqttQos | 0
mqttRetain | false
publishTimeHeartbeat | 60
publishTimeBoiler | 10
publishTimeThermostat | 10
publishTimeSolar | 10
publishTimeMixer | 10
publishTimeWater | 10
publishTimeOther | 10
publishTimeSensor | 10
publishSingle | false
publish2command | false
sendResponse | false
syslog |  
enabled | false
sensor |  
temperatureSensors | 7
temperatureSensorReads | 365948
temperatureSensorFails | 0
analogSensors | 0
analogSensorReads | 0
analogSensorFails | 0
api |  
APICalls | 2
APIFails | 0
bus |  
busStatus | "connected"
busProtocol | "Buderus"
busTelegramsReceived | 261138
busReads | 78550
busWrites | 0
busIncompleteTelegrams | 12
busReadsFailed | 0
busWritesFailed | 0
busRxLineQuality | 100
busTxLineQuality | 100
settings |  
boardProfile | "S32S3"
locale | "en"
txMode | 1
emsBusID | 11
showerTimer | false
showerMinDuration | 180
showerAlert | false
hideLed | false
noTokenApi | false
readonlyMode | false
fahrenheit | false
dallasParasite | false
boolFormat | 1
boolDashboard | 1
enumFormat | 1
analogEnabled | true
telnetEnabled | true
maxWebLogBuffer | 50
webLogBuffer | 33
modbusEnabled | false
forceHeatingOff | false
devices |  
0 |  
type | "boiler"
name | "Topline/GB162"
deviceID | "0x08"
productID | 115
brand | ""
version | "05.09"
entities | 76
handlersReceived | "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04"
handlersFetched | "0x14 0x16 0x33"
handlersPending | "0xBF 0xC2 0x1A 0x35 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x28 0x2E 0x3B"
1 |  
type | "thermostat"
name | "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410"
deviceID | "0x10"
productID | 158
brand | ""
version | "74.03"
entities | 58
handlersReceived | "0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BE 0x02BF 0x02C0 0x031D 0x0267"
handlersFetched | "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269"
handlersPending | "0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E"
handlersIgnored | "0xBF"
2 |  
type | "controller"
name | "BC10"
deviceID | "0x09"
productID | 114
brand | ""
version | "01.03"
entities | 0
handlersIgnored | "0x04 0x16 0x1C 0x33 0x24 0x10 0x18 0x19"

```

### TO REPRODUCE

Plug in the gateway, wait for 2-3 days and the outside temperate disappears from the thermostat. This happened twice now. When I plug the gateway back in the system continues to operate normally until after a couple of days.

### EXPECTED BEHAVIOUR

Outside temperature should not disappear from Moduline 3000 thermostat display

MichaelDvP, 2024-09-27:
The outdoor sensor is direct connected to the boiler and published in 0x19 telegram. Haven't you got A01(800) error in the log?
I think the sensor has a loose contact, cable is partly broken or an insect or spider has build its home in the sensor housing.
I can't think of a way emsesp can influence the sensor reading.
Can you log the telegrams and check for `boiler(0x08) -B-> all(0x00), UBAMonitorSlow(0x19), data: 00 A0  ...`. The first two data bytes are the sensor value (00A0 = 16.0°C) or `80 00` for broken or `7F 00` for short circuit.

Moving4407, 2024-09-30:
Happened again this morning and I measured the sensor but got a stable 20,58 kOhm value. The thermostat recovered and things resumed to normal. I have logging enabled and will keep an eye out when this happens again. Will also check the sensor housing for spiders

Moving4407, 2024-10-03:
Error just happened again, I can confirm the error code on the thermostat and that the sensor appears to operate normally (I can see it on the boiler itself through the info button)

[image]
[image]

Problem started around 17h05. Below are the logs, there are some gaps as the logs were flooded:

[log1.txt](https://github.com/user-attachments/files/17246465/log1.txt)
[log4.txt](https://github.com/user-attachments/files/17246466/log4.txt)

Due to the size I could not post the logs directly.

Moving4407, 2024-10-09:
Happened again twice, now running dev42 build. I could not find any log entries `boiler(0x08) -B-> all(0x00), UBAMonitorSlow(0x19),` which would mean that the boiler isn't sending them. When I disconnect and reconnect the thermostat it receives one such message and the outside temperature appears:

`2024-10-09 15:15:16.647 T 509586: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorSlow(0x19), data: 00 91 01 22 80 00 00 00 00 4F 02 81 92 04 A5 70 00 00 00 04 04 D0 02 4F DF 00 00
`

But after that initial message it remains quiet and the thermostat loses the outdoor temperature reading again after approx. 4 minutes.

I can see a lot of UBAMonitorSlow messages in the logs but they are between boiler and controller. I'm not sure what the controller is since I only see a boiler and a thermostat in my devices overview.
```

2024-10-09 15:20:00.280 T 510196: [emsesp] boiler(0x08) -W-> controller(0x09), UBAMonitorSlow(0x19), data: 00 91 00 DF 80 00 00 00 00 32
2024-10-09 15:20:00.304 T 510197: [emsesp] controller(0x09) -R-> boiler(0x08), UBAMonitorFast(0x18), length: 23
2024-10-09 15:20:00.343 T 510198: [emsesp] boiler(0x08) -W-> controller(0x09), UBAMonitorFast(0x18), data: 18 00 E0 21 00 01 01 20 60 01 3C 01 C3 00 E0 00 00 13 30 59 00 CC 00
2024-10-09 15:20:00.404 T 510199: [emsesp] controller(0x09) -R-> boiler(0x08), UBAMonitorSlow(0x19), length: 10
2024-10-09 15:20:00.429 T 510200: [emsesp] boiler(0x08) -W-> controller(0x09), UBAMonitorSlow(0x19), data: 00 91 00 DF 80 00 00 00 00 32
2024-10-09 15:20:00.480 T 510201: [emsesp] controller(0x09) -R-> boiler(0x08), UBAMonitorFast(0x18), length: 23
2024-10-09 15:20:00.518 T 510202: [emsesp] boiler(0x08) -W-> controller(0x09), UBAMonitorFast(0x18), data: 18 00 E0 21 00 01 01 20 60 01 3C 01 C3 00 E0 00 00 13 30 59 00 CC 00
2024-10-09 15:20:00.533 T 510203: [emsesp] controller(0x09) -R-> boiler(0x08), UBAMonitorSlow(0x19), length: 10
2024-10-09 15:20:00.555 T 510204: [emsesp] boiler(0x08) -W-> controller(0x09), UBAMonitorSlow(0x19), data: 00 91 00 DF 80 00 00 00 00 32
2024-10-09 15:20:00.580 T 510205: [emsesp] controller(0x09) -R-> boiler(0x08), UBAMonitorFast(0x18), length: 23
2024-10-09 15:20:00.617 T 510206: [emsesp] boiler(0x08) -W-> controller(0x09), UBAMonitorFast(0x18), data: 18 00 E0 21 00 01 01 20 60 01 3C 01 C3 00 E0 00 00 13 30 59 00 CC 00
2024-10-09 15:20:00.633 T 510207: [emsesp] controller(0x09) -R-> boiler(0x08), UBAMonitorSlow(0x19), length: 10
2024-10-09 15:20:00.654 T 510208: [emsesp] boiler(0x08) -W-> controller(0x09), UBAMonitorSlow(0x19), data: 00 91 00 DF 80 00 00 00 00 32
2024-10-09 15:20:00.780 T 510209: [emsesp] controller(0x09) -R-> boiler(0x08), UBAMonitorFast(0x18), length: 23
2024-10-09 15:20:00.818 T 510210: [emsesp] boiler(0x08) -W-> controller(0x09), UBAMonitorFast(0x18), data: 18 00 E0 21 00 01 01 20 60 01 3C 01 C3 00 E0 00 00 13 30 59 00 CC 00
2024-10-09 15:20:00.854 T 510211: [emsesp] controller(0x09) -R-> boiler(0x08), UBAMonitorSlow(0x19), length: 10
2024-10-09 15:20:00.879 T 510212: [emsesp] boiler(0x08) -W-> controller(0x09), UBAMonitorSlow(0x19), data: 00 91 00 DF 80 00 00 00 00 32
2024-10-09 15:20:00.904 T 510213: [emsesp] controller(0x09) -R-> boiler(0x08), UBAMonitorFast(0x18), length: 23
2024-10-09 15:20:00.942 T 510214: [emsesp] boiler(0x08) -W-> controller(0x09), UBAMonitorFast(0x18), data: 18 00 E0 21 00 01 01 20 60 01 3C 01 C3 00 E0 00 00 13 30 59 00 CC 00
2024-10-09 15:20:00.958 T 510215: [emsesp] controller(0x09) -R-> boiler(0x08), UBAMonitorSlow(0x19), length: 10
2024-10-09 15:20:00.979 T 510216: [emsesp] boiler(0x08) -W-> controller(0x09), UBAMonitorSlow(0x19), data: 00 91 00 DF 80 00 00 00 00 32
```

yafbu, 2024-10-14:
Seems like I have the same / a similar problem with my setup, after upgrading to 3.7.x dev end of September on a BBQKees Gateway E32 V1. During Summer with the Boiler in Summer Mode (heating switched off and only hot water active) I did not notice any errors, and no errors were stored in the RC310 error log. 

I have the following devices on my BUS
Logamax GB192i Boiler
RC310 Thermostat
RC200RF + Radio Module as Remote Thermostat
SM100 Solar Module
KM100 Gateway Module

From monitoring in home assistant I can see that every ~48 hours I get the following errors:
A11/1037 Outdoor temperature sensor fault
A11/6004 No communication to solar module
A11/3071 No communication with RF remote

The boiler itself does not loose connection to the temp sensor, but the RC310 does and commands the boiler to heat up as if it freezes outside. After some minutes the system recovers by itself and my heating returns to normal operation.

I just downgraded to 3.6.5 again to verify the errors do not occure with this verison.
Maybe https://github.com/emsesp/EMS-ESP32/issues/1937 is also related to this issue?

yafbu, 2024-10-14:
My System configuration as of  with Firmware v3.6.5:
{"System Info":{
"version":"3.6.5",
"uptime":"001+10:04:23.238",
"uptime (seconds)":122663,"platform":"ESP32",
"arduino":"Tasmota Arduino v2.0.17",
"sdk":"4.4.8.240628",
"free mem":132,"max alloc":67,"used app":1596,"free app":388,"partition":"app1",
"reset reason":"Software reset CPU / Software reset CPU"},
"Network Info":{
"network":"Ethernet",
"hostname":"ems-esp",
"TxPower setting":78,"static ip config":true,"enable IPv6":true,"low bandwidth":false,"disable sleep":true,"enable MDNS":true,"enable CORS":false,"AP provision mode":"disconnected",
"AP security":"wpa2",
"AP ssid":"ems-esp"},
"NTP Info":{
"NTP status":"connected",
"enabled":true,"server":"192.168.11.1",
"tz label":"Europe/Berlin"},
"OTA Info":{
"enabled":false,"port":8266},
"MQTT Info":{
"MQTT status":"connected",
"MQTT publishes":63780,"MQTT queued":0,"MQTT publish fails":0,"MQTT connects":2,"enabled":true,"client id":"ems-esp",
"keep alive":60,"clean session":true,"entity format":0,"base":"ems-esp",
"discovery prefix":"homeassistant",
"discovery type":0,"nested format":1,"ha enabled":true,"mqtt qos":0,"mqtt retain":true,"publish time heartbeat":60,"publish time boiler":10,"publish time thermostat":10,"publish time solar":10,"publish time mixer":10,"publish time other":10,"publish time sensor":10,"publish single":false,"publish2command":false,"send response":false},
"Syslog Info":{
"enabled":false},
"Sensor Info":{
"temperature sensors":1,"temperature sensor reads":24520,"temperature sensor fails":2,"analog sensors":0,"analog sensor reads":0,"analog sensor fails":0},
"API Info":{
"API calls":0,"API fails":0},
"Bus Info":{
"bus status":"connected",
"bus protocol":"Buderus",
"bus telegrams received (rx)":390186,"bus reads (tx)":53373,"bus writes (tx)":0,"bus incomplete telegrams":0,"bus reads failed":0,"bus writes failed":0,"bus rx line quality":100,"bus tx line quality":100},
"Settings":{
"board profile":"E32",
"locale":"en",
"tx mode":2,"ems bus id":73,"shower timer":false,"shower alert":false,"hide led":false,"notoken api":true,"readonly mode":true,"fahrenheit":false,"dallas parasite":false,"bool format":4,"bool dashboard":1,"enum format":2,"analog enabled":true,"telnet enabled":true,"max web log buffer":25,"web log buffer":25},
"Devices":[{"type":"boiler",
"name":"Logamax Plus/GB192/Condens GC9000/Greenstar ErP",
"device id":"0x08",
"product id":208,"version":"01.04",
"entities":85,"handlers received":"0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
"handlers fetched":"0x14 0x16 0x33 0x26 0x28",
"handlers pending":"0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA",
"handlers ignored":"0x23 0x2E 0x17"},
{"type":"thermostat",
"name":"RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
"device id":"0x10",
"product id":158,"version":"18.06",
"entities":52,"handlers received":"0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
"handlers fetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x031B 0x023A 0x0240",
"handlers pending":"0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E",
"handlers ignored":"0xBF"},
{"type":"thermostat",
"name":"RC200/CW100",
"device id":"0x38",
"product id":157,"version":"32.02",
"entities":1,"handlers received":"0x042B",
"handlers pending":"0x047B 0x0273 0x0A6A",
"handlers ignored":"0xBF"},
{"type":"solar",
"name":"SM100/MS100",
"device id":"0x30",
"product id":163,"version":"27.05",
"entities":36,"handlers received":"0x0362 0x0363 0x0366 0x0364 0x036A",
"handlers fetched":"0x0358 0x035A 0x0380 0x038E 0x0391",
"handlers pending":"0x035D 0x035F 0x035C 0x0361",
"handlers ignored":"0x0368 0x03A9 0x03AA 0xBF 0x038A"},
{"type":"gateway",
"name":"KM200/MB LAN 2",
"device id":"0x48",
"product id":189,"version":"04.08",
"entities":0,"handlers ignored":"0xF7 0xF9 0x0CBD"},
{"type":"controller",
"name":"9000i",
"device id":"0x09",
"product id":224,"version":"35.05",
"entities":0,"handlers ignored":"0x2A"},
{"type":"controller",
"name":"M200/RFM200",
"device id":"0x50",
"product id":218,"version":"34.04",
"entities":0,"handlers ignored":"0x0880 0x06A1 0x06A2 0x06A3 0x06A4 0x0881 0x0882 0x0883 0x0884 0x0885 0x0886 0x0887 0x0888 0xBF 0x0889"}]}

If the errors do not occure with the stable version I'll update again to 3.7.x to collect more information if it helps.

Moving4407, 2024-10-14:
> Seems like I have the same / a similar problem with my setup, after upgrading to 3.7.x dev end of September on a BBQKees Gateway E32 V1. During Summer with the Boiler in Summer Mode (heating [...]

I keep having this error popping up, have not downgraded yet. I switched the protocol to EMS+ (as I noticed the newer Moduline 3000 supperted that) but that doesn't make a difference. I posted the logs as @proddy requested but he has not provided any additional guidance so far.

yafbu, 2024-10-14:
@Moving4407
According to your issue I also tried the other protocol modes and changed the bus address, but without success too.
I'll try to get logs when after upgrading again, from HA it seems that the errors occure periodically every 48+ hours:
entity_id,						state,		last_changed
sensor.thermostat_error_code,	A11(3071),	2024-10-04T12:06:13.179Z
sensor.thermostat_error_code,	A11(1037),	2024-10-04T12:10:23.188Z
sensor.thermostat_error_code,	unavailable,	2024-10-04T12:11:13.183Z
sensor.thermostat_error_code,	A11(3071),	2024-10-04T12:26:13.198Z
sensor.thermostat_error_code,	A11(1037),	2024-10-04T12:30:23.204Z
sensor.thermostat_error_code,	unavailable,	2024-10-04T12:37:23.218Z
sensor.thermostat_error_code,	A11(6004),	2024-10-04T12:46:13.214Z
sensor.thermostat_error_code,	A11(3071),	2024-10-04T12:47:03.211Z
sensor.thermostat_error_code,	unavailable,	2024-10-04T12:47:13.215Z
sensor.thermostat_error_code,	A11(3071),	2024-10-04T12:54:13.223Z
sensor.thermostat_error_code,	A11(6004),	2024-10-04T13:00:03.226Z
sensor.thermostat_error_code,	unavailable,	2024-10-04T13:00:13.222Z
sensor.thermostat_error_code,	A11(6004),	2024-10-04T13:26:13.252Z
sensor.thermostat_error_code,	A11(3071),	2024-10-04T13:27:03.253Z
sensor.thermostat_error_code,	unavailable,	2024-10-04T13:27:23.254Z
sensor.thermostat_error_code,	A11(3071),	2024-10-06T13:53:26.038Z
sensor.thermostat_error_code,	A11(6004),	2024-10-06T13:56:16.007Z
sensor.thermostat_error_code,	A11(1037),	2024-10-06T13:56:26.010Z
sensor.thermostat_error_code,	unavailable,	2024-10-06T13:58:26.044Z
sensor.thermostat_error_code,	A11(3071),	2024-10-06T14:12:26.025Z
sensor.thermostat_error_code,	A11(1037),	2024-10-06T14:16:36.030Z
sensor.thermostat_error_code,	unavailable,	2024-10-06T14:18:16.061Z
sensor.thermostat_error_code,	A11(3071),	2024-10-06T14:29:26.042Z
sensor.thermostat_error_code,	unavailable,	2024-10-06T14:50:36.062Z
sensor.thermostat_error_code,	A11(3071),	2024-10-08T15:25:38.815Z
sensor.thermostat_error_code,	A11(6004),	2024-10-08T15:28:28.821Z
sensor.thermostat_error_code,	unavailable,	2024-10-08T15:28:38.824Z
sensor.thermostat_error_code,	A11(1037),	2024-10-08T15:38:08.842Z
sensor.thermostat_error_code,	A11(3071),	2024-10-08T15:38:28.839Z
sensor.thermostat_error_code,	unavailable,	2024-10-08T15:39:48.828Z
sensor.thermostat_error_code,	A11(6004),	2024-10-08T15:58:38.846Z
sensor.thermostat_error_code,	A11(1037),	2024-10-08T16:02:48.854Z
sensor.thermostat_error_code,	unavailable,	2024-10-08T16:03:48.862Z
sensor.thermostat_error_code,	A11(6004),	2024-10-08T16:12:48.860Z
sensor.thermostat_error_code,	A11(3071),	2024-10-08T16:16:28.890Z
sensor.thermostat_error_code,	unavailable,	2024-10-08T16:16:38.878Z
sensor.thermostat_error_code,	A11(6004),	2024-10-08T16:23:28.873Z
sensor.thermostat_error_code,	unavailable,	2024-10-08T16:34:38.897Z
sensor.thermostat_error_code,	A11(3071),	2024-10-08T16:40:38.891Z
sensor.thermostat_error_code,	unavailable,	2024-10-08T16:44:48.897Z
sensor.thermostat_error_code,	A11(6004),	2024-10-08T16:50:48.904Z
sensor.thermostat_error_code,	A11(3071),	2024-10-08T16:54:28.902Z
sensor.thermostat_error_code,	unavailable,	2024-10-08T16:54:38.918Z
sensor.thermostat_error_code,	A11(3071),	2024-10-08T17:23:38.937Z
sensor.thermostat_error_code,	unavailable,	2024-10-08T17:25:38.936Z
sensor.thermostat_error_code,	A11(6004),	2024-10-08T17:36:38.949Z
sensor.thermostat_error_code,	unavailable,	2024-10-08T17:37:48.947Z
sensor.thermostat_error_code,	A11(3071),	2024-10-10T17:10:01.642Z
sensor.thermostat_error_code,	unavailable,	2024-10-10T17:12:01.633Z
sensor.thermostat_error_code,	A11(1037),	2024-10-10T17:48:11.682Z
sensor.thermostat_error_code,	unavailable,	2024-10-10T18:06:41.698Z
sensor.thermostat_error_code,	A11(1037),	2024-10-10T18:18:01.715Z
sensor.thermostat_error_code,	unavailable,	2024-10-10T18:41:51.723Z
sensor.thermostat_error_code,	A11(1037),	2024-10-10T19:16:21.757Z
sensor.thermostat_error_code,	unavailable,	2024-10-10T19:16:51.771Z
sensor.thermostat_error_code,	A11(3071),	2024-10-10T19:25:51.774Z
sensor.thermostat_error_code,	unavailable,	2024-10-10T19:28:01.779Z
sensor.thermostat_error_code,	A11(3071),	2024-10-12T17:40:54.354Z
sensor.thermostat_error_code,	A11(6004),	2024-10-12T17:43:54.385Z
sensor.thermostat_error_code,	A11(1037),	2024-10-12T17:44:04.387Z
sensor.thermostat_error_code,	unavailable,	2024-10-12T17:47:04.389Z
sensor.thermostat_error_code,	A11(3071),	2024-10-12T17:57:04.362Z
sensor.thermostat_error_code,	A11(6004),	2024-10-12T17:57:54.365Z
sensor.thermostat_error_code,	unavailable,	2024-10-12T17:58:04.396Z
sensor.thermostat_error_code,	A11(1037),	2024-10-12T18:00:24.406Z
sensor.thermostat_error_code,	unavailable,	2024-10-12T18:06:14.407Z
sensor.thermostat_error_code,	A11(3071),	2024-10-12T18:14:04.414Z
sensor.thermostat_error_code,	A11(6004),	2024-10-12T18:14:54.409Z
sensor.thermostat_error_code,	unavailable,	2024-10-12T18:15:04.419Z
sensor.thermostat_error_code,	A11(1037),	2024-10-12T18:15:24.415Z
sensor.thermostat_error_code,	unavailable,	2024-10-12T18:21:54.382Z
sensor.thermostat_error_code,	A11(3071),	2024-10-12T18:29:04.427Z
sensor.thermostat_error_code,	A11(6004),	2024-10-12T18:31:54.429Z
sensor.thermostat_error_code,	unavailable,	2024-10-12T18:32:04.426Z
sensor.thermostat_error_code,	A11(3071),	2024-10-12T18:40:14.436Z
sensor.thermostat_error_code,	A11(6004),	2024-10-12T18:43:54.445Z
sensor.thermostat_error_code,	unavailable,	2024-10-12T18:44:04.441Z
sensor.thermostat_error_code,	A11(3071),	2024-10-12T18:54:14.454Z
sensor.thermostat_error_code,	A11(6004),	2024-10-12T19:02:54.462Z
sensor.thermostat_error_code,	unavailable,	2024-10-12T19:03:04.459Z

Moving4407, 2024-11-06:
I tried blocking writes from EMS-ESP but this does not help. Other suggestions (or requests for data) welcome, don't really know what/where to look

yafbu, 2024-11-06:
I can only confirm that the errors do not occure again after downgrading to 3.6.5.
24+ hours ago I updated to 3.7.0, and started collecting log messages.

yafbu, 2024-11-11:
Update: The Errors happen again with v3.7.0, periodically as burst every ~48h, but I did not find time to extract the event time slots from the log files. Though, maybe someone has a hint what i could look for?
[image]

fleischtorte, 2024-11-27:
Hi, got the same error with Junkers ZSB14 and CW400. Before 3.7.0 everything was fine

fleischtorte, 2024-12-10:
Hi, it seems that 3.7.1 fixed this issue? After Updating i dont get the error anymore same for you?

yafbu, 2024-12-11:
Yes, in my System the Errors did not occure again since the update to 3.7.1.

Moving4407, 2025-01-11:
I haven't seen the error in over a month running latest dev (3.7.2. dev 9). Closing.

closed 2025-01-11 by Moving4407