emsesp/EMS-ESP32 #2060: MQTT stopped working

git-WP-git, 2024-09-27:
### DESCRIPTION

After some years of working without problems, suddenly MQTT communication fails:
[image]

Communication with the internal webserver still works without problems:
[image]

Upload of latest firmware always failes:
[image]

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT
[emsesp_info (1).txt](https://github.com/user-attachments/files/17165581/emsesp_info.1.txt)

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2024-09-27:
I recommend you use the Flash tool and upgrade to 3.7.0, which is soon to be released. It has many fixes, including MQTT, and your particular version of EMS-ESP is two years old. By the way, I don't think the MQTT issues are related to EMS-ESP; it's probably an MQTT server change or network related, but we're no longer supporting older versions.

Flash tool: https://github.com/emsesp/EMS-ESP-Flasher/releases/tag/v1.1.0
Firmware: https://github.com/emsesp/EMS-ESP32/releases/download/latest/EMS-ESP-3_7_0-dev_40-ESP32.bin

MichaelDvP, 2024-09-27:
see: https://bbqkees-electronics.nl/wiki/gateway/firmware-update-and-downgrade.html#uploading-the-firmware-via-ems-esp-flasher-flashtool
For the E32 you have to open the housing and connect the USB-TTL adapter.

git-WP-git, 2024-09-29:
OTA-update is not possible (?)

neither this way:
[image]

nor this way:
[image]
[image]

proddy, 2024-09-29:
espota/OTA was removed in later version of EMS-ESP and replaced with a native HTTP upload. 

There are many ways to upgrade EMS-ESP. See https://emsesp.org/Getting-Started/#uploading-the-firmware

In your case, the easiest is to plug into your PC/laptop via the USB and get it on the latest stable 3.6.5. Make sure you maje copies of your settings just in case. To install, either use the Flash tool for this or go to https://install.emsesp.org. Once you're on 3.6.5 you can make the jump to 3.7.0 and when you're on 3.7.x all future updates will go automatically.

git-WP-git, 2024-09-30:
seems not to be so easy :(

I had to install the legacy driver in Windows 11, but it seems to need an individual configuration to enable a connection with the EMS-ESP...

[image]

proddy, 2024-09-30:
Think you need the CH340 driver. @bbqkees has some installation notes on https://bbqkees-electronics.nl/wiki/gateway/firmware-update-and-downgrade.html

git-WP-git, 2024-09-30:
installing the CH340 driver obviously also doesn't work with Windows 11 (?)

[image]

proddy, 2024-09-30:
@bbqkees any ideas?

bbqkees, 2024-09-30:
@git-WP-git Which version of the E32 do you have?
You may need to add a 12V DC power supply in order to have enough power to flash it.

If the driver was installed succesfully, you should see a CH340 com port in the Device Manager.
In Windows 11 the 'old' Device Manager is not visible by default. If you press the Windows key + R, you launch the Windows Run box. If you type devmgmt.msc, you'll open the old Device Manager. There should be a COM port USB-SERIAL CH340.
If not, the driver has not been installed properly. 
[image]
I think if you unpack the CH341.zip file, there are also the driver files in a folder.
If you have an unknown device in the Device Manager list, which would be the Gateway, you can click on it, and then install the driver by instructing the dialog you get to look into that specific folder with the driver files.

bbqkees, 2024-09-30:
If you can't make it work, you can try another laptop. Preferrably a private laptop with Windows 10 for instance.
Company laptops often have restrictions and automatic malware and executable detection and will just delete the exe or the driver.
If that also does not work, as a last resort you may ship the Gateway to me and I'll update it and send it back.

git-WP-git, 2024-09-30:
The problem with my Win 11 PC is that there is no "Ports" section in the device manager.
Trying to install the legacy COM port driver fails (as described before).
Unfortunately I have no older PC available.

Is there absolutely no wireless-option to update the firmware ?

proddy, 2024-09-30:
With 3.6.5 OTA should still work, if it's enabled in EMS-ESP and the ports match. I don't know why it failed for you.

Try again and see what is blocking it. Also use the correct .bin file which is https://github.com/emsesp/EMS-ESP32/releases/download/v3.6.5/EMS-ESP-3_6_5-ESP32-16MB.bin or the -4MB if you have 4MB Flash. It depends on the ESP32 chip you have.

MichaelDvP, 2024-09-30:
@proddy He uses board E32, 4MB, with v3.5.0b2.  For web uploading the esp23-16M will work, but USB-flash with erase the 4MB version is the onlyworking one. 
Sadly 3.5.0b2 does not show the partition scheme in system info, only `"free_app": 74,` 
@git-WP-git What does the Status page shows for  `File System (Used / Free)` and `Apllication (Used / Free)`
Maybe web update does not work with this early 3.5 dev direct to actual 3.6.5 or 3.7.0. Try to update to 3.5.0 or 3.5.1 final first, then to next final step by step if direct update to 3.7.0dev is not working.

git-WP-git, 2024-09-30:
After several attempts i succeeded in installing the CH340 driver.
I also use the correct 4MB firmware file now.

However, i get some strange errormessage indicating the flasher makes any Internet-access (?)

[image]

proddy, 2024-09-30:
That was a bug in the old Flasher that was reported today and fixed. Take the latest v1.1.0 Flash from https://github.com/emsesp/EMS-ESP-Flasher/releases/download/v1.1.0/EMS-ESP-Flasher.exe and try again

almost there!

git-WP-git, 2024-09-30:
Heureka, firmware update worked, device is now running with ver. 3.7.0

MQTT problem, however, still exists.
I have to check again the network-situation...
[image]

Thank you very much for your support!

proddy, 2024-09-30:
Some tips for MQTT at https://emsesp.org/Troubleshooting/#messages-are-not-always-coming-in-via-mqtt
If you're running Mosquitto in home assistant, enabling logging there will tell you why the connection from EMS-ESP is dropping

git-WP-git, 2024-10-01:
could the red lines within the log be any indicator for the MQTT-problem (?)

[image]

[image]

proddy, 2024-10-01:
no, that is just noise on the EMS bus line. It's ok.

Can you give more information on your MQTT broker? What is it, where is it hosted, what security etc? And send over the support info so we can see if there are any strange settings

git-WP-git, 2024-10-01:
I'm so stupid !
MQTT broker was inacticve - that was the problem from the beginning 8:(

Now everything works well - thank's again for the excellent support!!

[image]

closed 2024-10-01 by proddy