emsesp/EMS-ESP32 #2064: entity "Differental Pressure" shows wrong value

theozordako, 2024-09-28:
### DESCRIPTION

see https://github.com/emsesp/EMS-ESP32/issues/1563

"set diffferential pressure" shows wrong value

### REQUESTED INFORMATION

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT) --> no discord
- [X] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{"System Info":{"version":"3.6.5","uptime":"007+01:18:01.983","uptime (seconds)":609481,"platform":"ESP32","arduino":"Tasmota Arduino v2.0.14","sdk":"4.4.6.240105","free mem":196,"max alloc":107,"used app":1591,"free app":4105,"partition":"app0","reset reason":"Software reset CPU / Software reset CPU"},"Network Info":{"network":"Ethernet","hostname":"ems-esp","TxPower setting":78,"static ip config":false,"enable IPv6":true,"low bandwidth":false,"disable sleep":false,"enable MDNS":true,"enable CORS":false,"AP provision mode":"never","AP security":"wpa2","AP ssid":"ems-esp"},"NTP Info":{"NTP status":"connected","enabled":true,"server":"ptbtime2.ptb.de","tz label":"Europe/Berlin"},"OTA Info":{"enabled":false,"port":8266},"MQTT Info":{"MQTT status":"disconnected","enabled":false,"client id":"ems-esp","keep alive":60,"clean session":false,"entity format":1,"base":"ems-esp","discovery prefix":"homeassistant","discovery type":0,"nested format":1,"ha enabled":false,"mqtt qos":0,"mqtt retain":false,"publish time heartbeat":60,"publish time boiler":10,"publish time thermostat":10,"publish time solar":10,"publish time mixer":10,"publish time other":10,"publish time sensor":10,"publish single":false,"publish2command":false,"send response":false},"Syslog Info":{"enabled":false},"Sensor Info":{"temperature sensors":0,"temperature sensor reads":0,"temperature sensor fails":0,"analog sensors":0,"analog sensor reads":0,"analog sensor fails":0},"API Info":{"API calls":2026,"API fails":3},"Bus Info":{"bus status":"connected","bus protocol":"Buderus","bus telegrams received (rx)":2294290,"bus reads (tx)":417619,"bus writes (tx)":14,"bus incomplete telegrams":2,"bus reads failed":0,"bus writes failed":0,"bus rx line quality":100,"bus tx line quality":100},"Settings":{"board profile":"E32V2","locale":"en","tx mode":2,"ems bus id":11,"shower timer":false,"shower alert":false,"hide led":false,"notoken api":false,"readonly mode":true,"fahrenheit":false,"dallas parasite":false,"bool format":1,"bool dashboard":2,"enum format":1,"analog enabled":true,"telnet enabled":true,"max web log buffer":50,"web log buffer":38},"Devices":[{"type":"boiler","name":"CS6800i/WLW176i","device id":"0x08","product id":8,"version":"00.00","entities":146,"handlers received":"0xBF 0x14 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F","handlers fetched":"0xE6 0xEA 0x048D 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x04AE 0x04AF 0x02CC","handlers pending":"0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x048A","handlers ignored":"0x023E 0x0240 0x0291 0x0292 0x0293 0x0294 0x029B 0x02A5 0x02F5 0x02D6 0x04A8 0x04A9 0x04AA 0x04AB 0x04AC 0x04CA 0x04CB 0xE7 0x02E0 0x02EA 0x061E 0x17 0x0497 0xF7 0xF6 0x04A0"},{"type":"thermostat","name":"Rego 3000/UI800/WSW196i/BC400","device id":"0x10","product id":253,"version":"47.09","entities":62,"handlers received":"0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267","handlers fetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E","handlers pending":"0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E","handlers ignored":"0x0507 0x0508 0x0509 0x050A 0x059D 0xBF 0xF7 0xF9 0xF6"},{"type":"gateway","name":"MX300","device id":"0x48","product id":252,"version":"08.02","entities":0,"handlers ignored":"0xF9 0x2040 0x0495 0x0494 0x04AF 0x04AE 0x0DE8 0x0DE9 0x0DEC 0x0DED 0x0DEF 0x0DF0 0xF7 0x0CC2 0x0CBC 0x01 0x0CC3 0x0CBD 0xF6 0x03 0x04 0x17 0x0165 0x0166 0x0167 0x0168 0x0169 0x016A 0x016B 0x016C 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0xEF 0x02CC 0x02E0 0x048B 0x048D 0x048F 0x0499 0x04A7 0x04AB 0x04AC "}]}

```

### TO REPRODUCE

open EMS-ESP dashboard -> Open Boiler Device (Buderus WLW176 in my case) and open value for "Differental Pressure". The shown value does not match the actual configured value. In my case 5 mBar which is impossible =)

### EXPECTED BEHAVIOUR

Show actual currect value for differental pressur of the circulation pump (in my case it should be 250 mBar as configured on heat pump terminal)

### SCREENSHOTS

There recently was a user specific entity that shows the value currecly and works fine:
[image]

EMSESP Product feature seems to forget to upscale the value with factor 50 (--> 5*50 = 250)
[image]

### ADDITIONAL CONTEXT

Nope

proddy, 2024-09-29:
I see in the code it is multiplied by 50. Perhaps its rendered incorrectly. It's also incorrectly stored as a single byte so that needs adjusting as well (`uint8_t  hpSetDiffPress_`).

proddy, 2024-09-29:
fixed in 3.7.0 only

closed 2024-09-29 by proddy