emsesp/EMS-ESP32 #2068: Update firmware

HansRemmerswaal, 2024-09-30:
I tried to download the latest firmware from with EMS-ESP.

It tried to download:
https://github.com/emsesp/EMS-ESP32/releases/download/latest/EMS-ESP-3_7_0-dev_41-ESP32_16M.bin

But that failed with a strange github error.
Then I checked the download folder and noticed that the filenames are slightly different, it's 16MB.bin instead of 16M.bin
so maybe that is the problem?

https://github.com/emsesp/EMS-ESP32/releases/download/latest/EMS-ESP-3_7_0-dev_41-ESP32_16MB.bin

[image]

[image]

bbqkees, 2024-09-30:
The firmware naming scheme has changed.
If you are on f.i. 3.6.5, the update feature in the web interface will look for the old name.
Check here which bin file to use:
https://emsesp.org/Getting-Started/#bbqkees-electronics-ems-gateway
For the E32 V2 you would need https://github.com/emsesp/EMS-ESP32/releases/download/latest/EMS-ESP-3_7_0-dev_41-ESP32-16MB+.bin

HansRemmerswaal, 2024-09-30:
Thanks for your feedback, as I I have the Gateway E32 V2 (Ethernet + WiFi Editie V2) I should have installed the 16MB+ version instead of the 16MB version. I have correct this now.

But the reason for this ticket is that I was using 3.7.0-dev.xx and used the UPDATE button within de EMS-ESP. That button failed to download the firmware as the filename was wrong. It tried to download the 16M instead of the 16MB version (or in my case the 16MB+ version). And that 16M version doesn't exist in that download folder.

proddy, 2024-09-30:
I will look into that - I thought we fixed all the mappings but may have missed one. did you go from 3.6.5 (stable)?

HansRemmerswaal, 2024-09-30:
> I will look into that - I thought we fixed all the mappings but may have missed one. did you go from 3.6.5 (stable)?

I had 3.7.0-dev.30 (nice that we can see that in ems-esp in Home Assistant!)
And from within EMS-ESP I pressed the button to download the latest version.
The link it tried to open was:
https://github.com/emsesp/EMS-ESP32/releases/download/latest/EMS-ESP-3_7_0-dev_41-ESP32_16M.bin

proddy, 2024-10-01:
that's strange. It shouldn't do that and in the web codes it hardcodes to 16MB and 4MB. I expect it's a browser caching issue when coming from 3.6.x

proddy, 2024-10-01:
It's ok. I checked the file history and the firmware naming was changed in dev-32 just a week after dev-30 so I guess you were just unlucky. Thanks for reporting it though. Living on the edge!

closed 2024-10-01 by proddy