emsesp/EMS-ESP32 #2071: Buderus WLW186i device model

suaveolent, 2024-10-04:
### DESCRIPTION

I have a Buderus WLW186i which is not correctly identified:

- For the Boiler/HP it shows:  CS5800i/CS6800i/WLW176i.
- For the Thermostat it shows: Rego 3000/UI800/WSW196i/BC400
- 

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{"system":{"version":"3.7.0-dev.42","uptime":"000+00:05:11.968","uptimeSec":311,"platform":"ESP32S3","cpuType":"ESP32-S3","arduino":"ESP32 Arduino v2.0.17","sdk":"v4.4.7-dirty","freeMem":198,"maxAlloc":183,"freeCaps":8327,"usedApp":2237,"freeApp":3715,"partition":"app1","resetReason":"Software reset CPU / Software reset CPU","psram":true,"psramSize":8189,"freePsram":8130,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-46,"TxPowerSetting":0,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"disconnected","enabled":false,"server":"time.google.com","tzLabel":"Europe/Amsterdam"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":353,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTConnects":1,"enabled":true,"clientID":"esp32-553bdaec","keepAlive":60,"cleanSession":false,"entityFormat":0,"base":"Buderus WLW186i","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"Buderus","busTelegramsReceived":1645,"busReads":389,"busWrites":0,"busIncompleteTelegrams":0,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"CUSTOM","locale":"en","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"phyType":0,"rxGPIO":17,"txGPIO":18,"dallasGPIO":0,"pbuttonGPIO":0,"ledGPIO":38,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":50,"webLogBuffer":19,"modbusEnabled":false,"forceHeatingOff":false},"devices":[{"type":"boiler","name":"CS5800i/CS6800i/WLW176i","deviceID":"0x08","productID":8,"brand":"Buderus","version":"00.00","entities":164,"handlersReceived":"0xBF 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F","handlersFetched":"0x14 0xE6 0xEA 0x048D 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x04AE 0x04AF 0x02CC 0x04AA 0x04A7 0x02D6","handlersPending":"0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A","handlersIgnored":"0x023E 0x0240 0x0291 0x0292 0x0293 0x0294 0x029B 0x02A5 0x02F5 0x02CD 0x04A8 0x04A9 0x04AB 0x04AC 0x04CA 0x04CB 0x061E 0xD2 0xE2"},{"type":"thermostat","name":"Rego 3000/UI800/WSW196i/BC400","deviceID":"0x10","productID":253,"brand":"","version":"47.10","entities":61,"handlersReceived":"0x06 0x02BA 0x02BB 0x02BC 0x0267","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E","handlersPending":"0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031D 0x031E","handlersIgnored":"0x0507 0x0508 0x0509 0x050A 0x059D 0xBF"},{"type":"mixer","name":"MM100","deviceID":"0x20","productID":160,"brand":"","version":"24.05","entities":7,"handlersReceived":"0x02D7","handlersFetched":"0x02CD","handlersIgnored":"0x02E1 0x02EB 0x0255 0xE6 0x0291 0xBF"},{"type":"gateway","name":"WiFi module","deviceID":"0x48","productID":252,"brand":"Buderus","version":"08.02","entities":0,"handlersIgnored":"0x2040"}]}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_
Setup ems-bus with Buderus WLW186i

### EXPECTED BEHAVIOUR

- For the Boiler/HP:  CS5800i/CS6800i/WLW176i/WLW186i
- For the Thermostat: Rego 3000/UI800/WLW186i/WSW196i/BC400

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2024-10-04:
Thanks, we'll update the text descriptions. Did you know in 3.7.0 you can rename the devices using the Customization module in the WebUI.

suaveolent, 2024-10-04:
Thanks, I was already looking for this feature, but did not bother to look again after updating to `v3.7.0`.  🚀

proddy, 2024-10-05:
3.7.0-dev.43

commit 34fc9f30 (2024-10-05): fixes #2071
closed 2024-10-05 by proddy