emsesp/EMS-ESP32 #2073: improve EMS brand detection

proddy, 2024-10-06:
EMS-ESP tries to determine the manufacturer/brand (Buderus, Sieger, Junkers, Bosch, Nefit etc..) from each connected EMS device by reading the last byte from the Version 0x02 telegram (in `emsesp.cpp::process_version()`).

But it does a poor job and only detects a few brands correctly. Let's improve this, ask the community for make/model and the results of a `read <deviceID> 2` to correctly build the mapping.

MichaelDvP, 2024-10-06:
Good idea, i think the 2nd and 3rd identifier WILL gives some more info, also for the thermostats with same 1st ID. I'll start with my system:
Type Brand Name DeviceID data
```
Boiler		Buderus	GB125	(0x08): data: 48 03 00 4B 04 1C 00 00 00 00
Controller	Buderus	BC10 	(0x09): data: 44 02 03
Thermostat	Buderus	RC35 	(0x10): data: 56 01 12
Mixer		Buderus	MM10 	(0x21): data: 45 02 01
Solar		Buderus	SM10 	(0x30): data: 49 02 00
```

proddy, 2024-10-06:
Mine:
```
Boiler       Nefit Trendline (0x08):     data 7B 06 01 00 00 00 00 00 00 04
Controller   BC10 (0x09):                data BE 01 03
Thermostat   Nefit Moduline 300 (0x17):  data 4D 03 03
```
Notes
- the controller is recognised as a BC10 but it's built into the boiler, since its a combi-boiler
- the boiler's brand is identified correctly as Nefit because of the `04` in the last byte

VlastiBroucek, 2024-10-07:
Mine (I am including more data FYI, since the device 0x08 and 0x09 return same string)...
```
ems-esp:$ show devices
These EMS devices are currently active:

boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:12.16)
 Received telegram type IDs: 0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04
 Fetched telegram type IDs: 0x14 0x16 0x33
 Pending telegram type IDs: 0xBF 0xC2 0x1A 0x35 0x26
 Ignored telegram type IDs: 0x23

thermostat: RC200/CW100 (DeviceID:0x18, ProductID:157, Version:43.02)
 Received telegram type IDs: 0x06 0xA2 0x31D
 Fetched telegram type IDs: 0x2A5 0x2B9 0x2AF 0x29B 0x291 0x2F5 0x23A
 Pending telegram type IDs: 0xA3 0x12 0x13 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2A9 0x2BD 0x2B3 0x29F 0x475 0x2AA 0x2BE 0x2B4 0x2A0 0x476 0x2AB 0x2BF 0x2B5 0x2A1 0x477 0x2AC 0x2C0 0x2B6 0x2A2 0x478 0x2CC 0x467 0x2CE 0x468 0x292 0x2D0 0x469 0x293 0x2D2 0x46A 0x294 0x31B 0x31E 0x267 0x240 0xBB 0x23E
 Ignored telegram type IDs: 0xBF

controller: HT3 (DeviceID:0x09, ProductID:95, Version:12.16)

ems-esp:$ read 0x08 2
001+03:26:24.319 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), Version(0x02), data: 5F 0C 10 00 00 00 00 00 00 00
ems-esp:$ read 0x09 2
001+03:29:46.595 N 2: [emsesp] controller(0x09) -W-> Me(0x0B), Version(0x02), data: 5F 0C 10 00 00 00 00 00 00 00
ems-esp:$ read 0x18 2
001+03:31:13.439 N 3: [emsesp] thermostat(0x18) -W-> Me(0x0B), Version(0x02), data: 9D 2B 02 00 FF 00 00 00 00
ems-esp:$
````

These are Australian market.

- Boiler is sold as Bosch Condens 5000W ZWB 37-2 A23 - combi boiler for natural gas
- It is supposed to have built in Heatronic 3 controller with 2-wire BUS
- Room thermostat is Bosch CR50, connected by 2 wires to the BUS

I can send PDFs with installation manuals if they would be of any help.

Delta1977, 2024-10-07:
Boiler: Buderus Logamax GB172i.2-15KW 
Thermostat Buderus RC220
Controller: Internal Buderus BC400

`ems-esp:$ show devices
These EMS devices are currently active:

boiler: Buderus Condens 5000i, Greenstar 8000/GC9800IW, GB192i*2 (DeviceID:0x08, ProductID:195, Version:05.06)
 Received telegram type IDs: 0xBF 0xC2 0x14 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x2E 0x3B 0x04
 Fetched telegram type IDs: 0xE6 0xEA 0x28
 Pending telegram type IDs: 0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A
 Ignored telegram type IDs: 0x2E0 0x2EA 0xD7 0xD8 0x2D6 0x29B 0x17 0x36

thermostat: RT800, RC220 (DeviceID:0x38, ProductID:3, Version:21.04)
 Received telegram type IDs: 0x42B 0x47B
 Fetched telegram type IDs:
 Pending telegram type IDs: 0x273 0xA6A
 Ignored telegram type IDs: 0xBF 0x3B

thermostat: UI800, BC400 (DeviceID:0x10, ProductID:4, Version:49.07)
 Received telegram type IDs: 0x06 0x2BA 0x2BB 0x2BC 0x267
 Fetched telegram type IDs: 0x2A5 0x2B9 0x2AF 0x29B 0x2CC 0x291 0x292 0x293 0x294 0x2F5 0x2F6 0x23A 0x240
 Pending telegram type IDs: 0xA3 0xA2 0x12 0x13 0x471 0x2A6 0x2B0 0x29C 0x472 0x2A7 0x2B1 0x29D 0x473 0x2A8 0x2B2 0x29E 0x474 0x2A9 0x2BD 0x2B3 0x29F 0x475 0x2AA 0x2BE 0x2B4 0x2A0 0x476 0x2AB 0x2BF 0x2B5 0x2A1 0x477 0x2AC 0x2C0 0x2B6 0x2A2 0x478 0x467 0x2CE 0x468 0x2D0 0x469 0x2D2 0x46A 0x31B 0x31D 0x31E 0xBB 0x23E 0x269
 Ignored telegram type IDs: 0xC4 0x31 0xBF 0x3B

ems-esp:$ read 08 2
001+08:01:14.651 N 1: [emsesp] boiler(0x08) -W-> me(0x0B), Version(0x02), data: C3 05 06 EF 02 12 00 00 00 03 00 00
ems-esp:$ read 38 2
001+08:01:20.762 N 2: [emsesp] thermostat(0x38) -W-> me(0x0B), Version(0x02), data: 03 15 04 00 FF
ems-esp:$ read 10 2
001+08:01:29.691 N 3: [emsesp] thermostat(0x10) -W-> me(0x0B), Version(0x02), data: 04 31 07 00 FF 00 00 00 00 00
`

kwertie01, 2024-10-08:
```
Boiler       Nefit Smartline HRC24/CW4 (0x08):  data 40 03 02
Controller   BC10 (built-in)           (0x09):  data 44 02 01
Thermostat   Nefit Moduline 400        (0x10):  data 4E 02 14
```

mattreim, 2024-10-08:
```
Boiler: Buderus Logamax plus GB172i.2-15
(0x08) -W-> me(0x0B), Version(0x02), data: C3 05 06 EF 02 12 00 00 00 03 00 00

Thermostat: Logamatic BC400
(0x10) -W-> me(0x0B), Version(0x02), data: 04 31 04 00 FF 00 00 00 00 00
```

4l4R1, 2024-10-08:
boiler(0x08) -W-> me(0x0B), Version(0x02), data: 5F 17 00 00 00 00 00 00 00 00
Junkers Cerapur Smart ZWB28-3C (detected as Condens 2500/5000W, Logamax/Logomatic, Cerapur Top, Greenstar, Generic HT3)

thermostat(0x10) -W-> me(0x0B), Version(0x02), data: 69 0D 05 00 00 00 00 00 00 02
Junkers FW100

controller(0x09) -W-> me(0x0B), Version(0x02), data: 5F 17 00 00 00 00 00 00 00 00
HT3

suaveolent, 2024-10-10:
Here the data for Buderus Logatherm WLW186i (heatpump) , MM100 (mixer) and MX300 (Wifi-Module)

```ems-esp:$ show devices
These EMS devices are currently active:

boiler: Buderus CS*800i, Logatherm WLW* (DeviceID:0x08, ProductID:8, Version:00.00)
 Received telegram type IDs: 0xBF 0x14 0xD1 0xE3 0xE4 0xE5 0xE9 0x494 0x495 0x48F 
 Fetched telegram type IDs: 0xE6 0xEA 0x48D 0x4A2 0x485 0x486 0x492 0x488 0x484 0x48B 0x491 0x499 0x49C 0x49D 0x4AE 0x4AF 0x2CC 0x4AA 0x4A7 0x2D6 
 Pending telegram type IDs: 0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x48A 0x4A5 
 Ignored telegram type IDs: 0x61E 0xD2 0xE2 0x23E 0x240 0x291 0x292 0x293 0x294 0x29B 0x2A5 0x2F5 0x2CD 0x4A8 0x4A9 0x4AB 0x4AC 0x4CA 0x4CB 0x4A0 0xF7 0xF6 

thermostat: Rego 3000, UI800, Logatherm WLW186i/WSW196i, Logamatic BC400 (DeviceID:0x10, ProductID:253, Version:47.10)
 Received telegram type IDs: 0x06 0x2BA 0x2BB 0x2BC 0x267 
 Fetched telegram type IDs: 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2CC 0x467 0x291 0x292 0x293 0x294 0x2F5 0x23A 0x240 0xBB 0x23E 
 Pending telegram type IDs: 0xA3 0xA2 0x12 0x13 0x2A6 0x2B0 0x29C 0x472 0x2A7 0x2B1 0x29D 0x473 0x2A8 0x2B2 0x29E 0x474 0x2A9 0x2BD 0x2B3 0x29F 0x475 0x2AA 0x2BE 0x2B4 0x2A0 0x476 0x2AB 0x2BF 0x2B5 0x2A1 0x477 0x2AC 0x2C0 0x2B6 0x2A2 0x478 0x2CE 0x468 0x2D0 0x469 0x2D2 0x46A 0x2F6 0x31B 0x31D 0x31E 0x269 
 Ignored telegram type IDs: 0x507 0x508 0x509 0x50A 0x59D 0xBF 0xF7 0xF9 

mixer: MM100 (DeviceID:0x20, ProductID:160, Version:24.05)
 Received telegram type IDs: 0x2D7 
 Fetched telegram type IDs: 0x2CD 
 Pending telegram type IDs: 
 Ignored telegram type IDs: 0x2E1 0x2EB 0xE6 0x255 0x291 0xBF 

gateway: Buderus WiFi module (DeviceID:0x48, ProductID:252, Version:08.02)```

proddy, 2024-10-11:
Analyzing the data (thanks everyone!), maybe there's is something in offset 1+2 which we use for the version.

@VlastiBroucek is there way you can see the version of the boiler and/or controller on the unit yourself? Does it also show 12.16?

@suaveolent could you also report back the version showed on you EMS devices, to see if they are the same what EMS-ESP thinks they are.

suaveolent, 2024-10-11:
In principle, they look good, with some slight deviations and different naming:
Here is the actual versions I found.

**Boiler:** 

- Control panel unit: Logamatic BC400: `NF47.10`
- Inside unit: Logatherm WLW MB-7 AR: SW version: `9.6.0.-caa36357`
- Outside unit: Logatherm WLW186i-12 E: SW-Version: `9.6.1-733fefcc`

**Mixer**

- MM100: SW-Version: `V24.05`

**Gateway:** 

- MX300:  Firmware version `08.02.04`

proddy, 2024-10-11:
Thanks @suaveolent . Seeing that the version telegram from your Boiler gives `08 00 00 00 00 00 00 00 00 03 03 18` the version (which is 47.10) must come from some other telegram.

airhead1234, 2024-10-11:
```
Boiler		Buderus Logamax kompakt WS170		(0x08): data: DB 33 04 00 FF 00 00 00 00 00
Thermostat	Buderus Logamatic RC310				(0x10): data: 9E 4A 03 00 FF 00 00 00 00
```

suaveolent, 2024-10-11:
> Thanks @suaveolent . Seeing that the version telegram from your Boiler gives `08 00 00 00 00 00 00 00 00 03 03 18` the version (which is 47.10) must come from some other telegram.

@proddy the 47.10 comes from the BC400 (identified as thermostat) and is already correctly identified.
But for the boiler itself they must come from somewhere else. You are right.

VlastiBroucek, 2024-10-12:
> @VlastiBroucek is there way you can see the version of the boiler and/or controller on the unit yourself? Does it also show 12.16?
Yes, it shows software version (service function code 8.A) as "CF1216"

vmonkey, 2024-10-12:
For me,  the requested data are as follows: 

ems-esp:$ show devices
These EMS devices are currently active:

boiler: Condens 2500/5000W, Logamax/Logomatic, Cerapur Top, Greenstar, Generic HT3 (DeviceID:0x08, ProductID:95, Version:18.14)
 Received telegram type IDs: 0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04
 Fetched telegram type IDs: 0x14 0x16 0x33 0x26
 Pending telegram type IDs: 0xBF 0xC2 0x1A 0x35 0x2A 0x2E 0x3B
 Ignored telegram type IDs: 0x23

thermostat: RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410 (DeviceID:0x10, ProductID:158, Version:18.05)
 Received telegram type IDs: 0x06 0xA2 0x2BB 0x2BC 0x2BD 0x2BE 0x2BF 0x2C0 0x31D 0x267
 Fetched telegram type IDs: 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2CC 0x467 0x291 0x2CE 0x468 0x292 0x293 0x294 0x2F5 0x2F6 0x31B 0x23A 0x240 0xBB 0x23E 0x269
 Pending telegram type IDs: 0xA3 0x12 0x13 0x2A7 0x2B1 0x29D 0x473 0x2A8 0x2B2 0x29E 0x474 0x2A9 0x2B3 0x29F 0x475 0x2AA 0x2B4 0x2A0 0x476 0x2AB 0x2B5 0x2A1 0x477 0x2AC 0x2B6 0x2A2 0x478 0x2D0 0x469 0x2D2 0x46A 0x31E
 Ignored telegram type IDs:

mixer: MM100 (DeviceID:0x21, ProductID:160, Version:24.05)
 Received telegram type IDs: 0x2D8
 Fetched telegram type IDs: 0x2CE
 Pending telegram type IDs:
 Ignored telegram type IDs: 0x255 0x16 0x33 0x292 0x2E2 0x2EC

controller: HT3 (DeviceID:0x09, ProductID:95, Version:18.14)

The actual products are
Boiler: Junkers Cerapur Comfort ZSBR 16-3E
Thermostat: Bosch CW400
Mixer: MM100

Versions match (except for MM100 thermostat reports NF24.05)

Sbried, 2024-10-17:
For my installation the data is as follows:

## Heat pump
*Detected by ESM-ESP:*
**boiler: Enviline, Compress 6000AW, Hybrid 3000-7000iAW, SupraEco/Geo 5xx, WLW196i/WSW196i (DeviceID:0x08, ProductID:172, Version:02.02)**
*Installed:*
**Buderus Logatherm WLW166i12 T190 SP AR (Indoor unit WLW166i, Outdoor unit WLW-12 SP AR P3)**
*Read output data type 2*
**001+06:06:32.241 N 1: [emsesp] boiler(0x08) -W-> me(0x0B), Version(0x02), data: AC 02 02 AD 01 01 00 00 00 00**

*Data shown in the control unit HMC310*
- Steuereinheit (control unit): CUHP
- Kältekreistyp (refrigant circuit?): 3p12-16
- Inneneinheit (indoor unit): Tower2
- SW-Vers. Kältekreis (SW refrigant? circuit): 00.00
- SW-Vers. Inneneinheit (SW indoor unit): 02.02
- SW-Vers. Erweit.modul (extension module): 00.00

## Thermostat
*Detected by ESM-ESP:*
**thermostat: RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410 (DeviceID:0x10, ProductID:158, Version:73.04)**
*Installed:*
**HMC310**
*Read output data type 2*
**001+06:06:53.889 N 2: [emsesp] thermostat(0x10) -W-> me(0x0B), Version(0x02), data: 9E 49 04 00 FF 00 00 00 00**

*Data shown in the control unit HMC310*
- SW 73.04

## Gateway
*Detected by ESM-ESP:*
**Buderus WiFi module (DeviceID:0x48, ProductID:252, Version:08.02)**
*Installed:*
**MX300**
*Read output data type 2*
**001+06:22:39.134 N 3: [emsesp] gateway(0x48) -W-> me(0x0B), Version(0x02), data: FC 08 02 00 04 00 00 00 00 03 00 FF**

*Data shown in the control unit HMC310*
- Type Internetmodule: MX30
- SW 08.02

proddy, 2024-10-18:
thanks @Sbried - I never thought about telegram 0x01. It would make sense to store the device information there (as 2 is only the version). On my Nefit Trendline combi-boiler offset 27 of 0x01 is the letters CBS (Combi-Boiler System?). 

Could we ask everyone to do a `read <device-id> 1` to compare. Maybe we can extract the brand from that data.

MichaelDvP, 2024-10-18:
I know when connecting a device the 0x01 is requested/sent, in normal operation is not repeated any more.
My system have only 26 bytes in each 01 telegram, seems no ascii:
```
boiler(0x08) -W-> me(0x4A),     ?(0x01), data: AA 25 30 20 62 05 29 68 71 85 72 92 1A AA 06 00 00 00 00 00 00 00 00 00 00 00 00
thermostat(0x10) -W-> me(0x4A), ?(0x01), data: 25 30 20 62 14 02 58 71 85 81 71 2A AA AA 0E 35 00 00 00 00 00 00 00 00 00 00 00
mixer(0x21) -W-> me(0x4A),      ?(0x01), data: AA 25 30 20 63 66 65 01 02 19 15 AA AA AA 06 00 00 00 00 00 00 00 00 00 00 00 00
solar(0x30) -W-> me(0x4A),      ?(0x01), data: 70 00 72 79 01 32 32 00 93 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

suaveolent, 2024-10-18:
here for my devices:

```
boiler(0x08) -W-> me(0x0B), ?(0x01), data: 30 35 37 30 33 37 37 31 30 31 32 39 30 38 37 35 30 37 36 37 38 30 31 00 FF FF FF
boiler(0x08) -W-> me(0x0B), ?(0x01), data: 58 43 55 5F 54 48 48 00 00 00 00 00 00 00 00 00 (offset 27)

thermostat(0x10) -W-> me(0x0B), ?(0x01), data: 33 31 36 33 30 38 35 35 30 32 33 30 38 33 30 39 39 30 30 31 31 30 31 38 37 33 37
thermostat(0x10) -W-> me(0x0B), ?(0x01), data: 48 4D 49 38 30 30 2E 32 00 00 00 00 00 00 00 00 (offset 27)

mixer(0x20) -W-> me(0x0B), ?(0x01), data: 32 35 33 30 31 37 38 30 30 33 34 32 34 37 37 33 38 31 31 30 31 31 34 00 FF FF FF
mixer(0x20) -W-> me(0x0B), ?(0x01), data: 4D 4D 31 30 30 00 00 00 00 00 00 00 00 00 00 00 (offset 27)

gateway(0x48) -W-> me(0x0B), ?(0x01), data: 32 35 33 30 33 33 37 33 30 36 33 30 34 38 37 33 38 38 30 36 31 32 34 00 00 00 00
gateway(0x48) -W-> me(0x0B), ?(0x01), data: 4D 58 33 30 30 00 00 00 00 00 00 00 00 00 00 00 05 00 00 00 00 00 00 00 00 00 00 (offset 27)
gateway(0x48) -W-> me(0x0B), ?(0x01), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 54)
gateway(0x48) -W-> me(0x0B), ?(0x01), data: <empty> (offset 81)
```

vmonkey, 2024-10-18:
In my case, the responses are

```
ems-esp:$ read 0x08 1
000+00:01:29.526 N 2: [emsesp] boiler(0x08) -W-> me(0x0B), ?(0x01), data: 00 00 49 07 56 42 00 20 01 21 02 00 02 02 FF FF
ems-esp:$ read 0x10 1
000+00:01:51.995 N 3: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x01), data: 32 35 33 30 39 35 37 39 31 31 32 31 34 37 37 33 38 31 31 31 30 38 33 00 FF FF FF
000+00:01:52.058 N 4: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x01), data: 43 57 34 30 30 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
ems-esp:$ read 0x21 1
000+00:02:05.599 N 5: [emsesp] mixer(0x21) -W-> me(0x0B), ?(0x01), data: 32 35 33 30 39 35 37 30 30 30 34 39 36 37 37 33 38 31 31 30 31 33 38 00 FF FF FF
000+00:02:05.666 N 6: [emsesp] mixer(0x21) -W-> me(0x0B), ?(0x01), data: 4D 4D 31 30 30 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
ems-esp:$ read 0x09 1
000+00:02:33.915 N 7: [emsesp] controller(0x09) -W-> me(0x0B), ?(0x01), data: 00 00 49 07 56 42 00 20 01 21 02 00 02 02 FF FF
```

proddy, 2024-10-18:
> here for my devices

The offset 27 is definitely a name of sorts, but not the brand so probably not useful to us. Your mixer says MM100, the Thermostat says HMI800.2, Gateway MX300 and the Boiler XCU_THH !

Delta1977, 2024-10-18:
Boiler: Buderus Logamax GB172i.2-15KW
Thermostat Buderus RC220
Controller: Internal Buderus BC400

read 0x08 1
000+18:37:54.199 N 10: [emsesp] boiler(0x08) -W-> me(0x0B), ?(0x01), data: 33 36 37 34 39 39 39 39 38 32 34 30 31 32 33 32 33 30 31 34 39 34 34 38 37 35 30
000+18:37:54.239 N 11: [emsesp] boiler(0x08) -W-> me(0x0B), ?(0x01), data: 41 43 55 2D 4D 48 00 00 00 00 00 00 00 00 00 00 (offset 27)

 read 0x10 1
000+18:38:26.028 N 12: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x01), data: 33 31 36 33 30 38 35 35 30 32 33 31 30 33 30 39 39 30 30 35 33 32 30 38 37 33 37
000+18:38:26.123 N 13: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x01), data: 42 43 34 30 30 2D 46 4F 00 00 00 00 00 00 00 00 37 31 35 33 30 33 00 00 00 00 00 (offset 27)
000+18:38:28.702 N 14: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x01), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 54)
000+18:38:29.514 N 15: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x01), data: <empty> (offset 81)

read 0x38 1
000+18:38:48.044 N 16: [emsesp] thermostat(0x38) -W-> me(0x0B), ?(0x01), data: 32 35 33 30 33 37 37 34 30 31 30 37 31 37 37 33 38 31 31 32 39 34 38 00 00 00 00
000+18:38:49.423 N 17: [emsesp] thermostat(0x38) -W-> me(0x0B), ?(0x01), data: 52 43 32 32 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF (offset 27)
000+18:38:50.160 N 18: [emsesp] thermostat(0x38) -W-> me(0x0B), ?(0x01), data: <empty> (offset 54)

MichaelDvP, 2024-10-18:
> The offset 27 is definitely a name of sorts, but not the brand so probably not useful to us. 

We can set this as custom name if not set by user. Maybe more specific that the name list in database.
But offset 27 only exist for ems+ devices. Also the first 26 bytes seems to be ascii numbers (0x3?) for ems+, on ems devices this is binary.

vmonkey, 2024-10-18:
Can it be that there is no info about the brand at all? Checking system info in my system using CW400, I see controller as HT3, thermostat as CW400 and mixer as MM100. No brand mentioned. And the component names match the offset 27 (except for controller that does not publish offset 27 in my case). 

In the worst case, brand can be identified by creating a map of component names (e.g., CW400 is Bosch)

VlastiBroucek, 2024-10-18:
```
eems-esp:$ read 0x08 1
e002+20:19:34.243 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), ?(0x01), data: 00 00 54 28 59 25 00 20 01 28 02 00 02 02 FF FF
eems-esp:$ read 0x09 1
e002+20:19:51.733 N 2: [emsesp] controller(0x09) -W-> Me(0x0B), ?(0x01), data: 00 00 54 28 59 25 00 20 01 28 02 00 02 02 FF FF
eems-esp:$ read 0x18 1
e002+20:20:22.171 N 3: [emsesp] thermostat(0x18) -W-> Me(0x0B), ?(0x01), data: 30 35 37 30 38 35 35 32 30 30 30 30 31 37 37 33 38 31 31 31 30 32 32
30 FF FF FF
e002+20:20:22.246 N 4: [emsesp] thermostat(0x18) -W-> Me(0x0B), ?(0x01), data: 43 52 35 30 00 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
ems-esp:$
```

Sbried, 2024-10-19:
Next "research" result: data type 0x02 offset 9: is 0=undefined, 1=Bosch, 2= Junkers, 3=Buderus, 4=Nefit?
On 2 of my modules I get back 0, which maybe indicates that it is used in some/all brands. But the MX300 returns 3 and it is Buderus.
Maybe others try `read <module-ID> 2 9 1` and compare the output with what they have installed

proddy, 2024-10-19:
> Next "research" result: data type 0x02 offset 9: is 0=undefined, 1=Bosch, 2= Junkers, 3=Buderus, 4=Nefit? On 2 of my modules I get back 0, which maybe indicates that it is used in some/all brands. [...]

This is what EMS-ESP does now, reads the 9th data byte from 0x02 (Version). It should be shown in EMS-ESP when you click on the info button. Problem with this is that it isn't always accurate and missing on some devices.

Sbried, 2024-10-19:
ok, I see

VlastiBroucek, 2024-10-20:
```
ems-esp:$ read 0x08 2 9 1
003+23:58:36.570 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), Version(0x02), data: 00 (offset 9)
ems-esp:$ read 0x09 2 9 1
003+23:59:27.425 N 2: [emsesp] controller(0x09) -W-> Me(0x0B), Version(0x02), data: 00 (offset 9)
ems-esp:$ read 0x18 2 9 1
003+23:59:32.724 N 3: [emsesp] thermostat(0x18) -W-> Me(0x0B), Version(0x02), data: <empty> (offset 9)
```

Boiler is branded Bosch, controller is inside of the boiler and Room Thermostat is branded Bosch

[30kw__37kw_user_manual.pdf](https://github.com/user-attachments/files/17447814/30kw__37kw_user_manual.pdf)
[2019_-_condens_5000w_30kw__37kw.pdf](https://github.com/user-attachments/files/17447815/2019_-_condens_5000w_30kw__37kw.pdf)
[cr50_controller_installation_instructions.pdf](https://github.com/user-attachments/files/17447816/cr50_controller_installation_instructions.pdf)
[cr50_operating_instructions.pdf](https://github.com/user-attachments/files/17447817/cr50_operating_instructions.pdf)

proddy, 2024-10-28:
The name we capture, I called it model and added to 3.7.1. It will be prefixed to the Device Name if no custom one is set.

Ed1954, 2024-12-07:
Detected Buderus GB192i*2
Installed Logamax plus GB182i.2  

"type": "boiler",
      "name": "ACU-MH/Condens 5000i, Greenstar 8000/GC9800IW, GB192i*2",
      "deviceID": "0x08",
      "productID": 195,
      "brand": "Buderus",
      "version": "05.06",
      "entities": 85,
      "handlersReceived": "0xBF 0xC2 0x14 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x2E 0x3B 0x04",
      "handlersFetched": "0xE6 0xEA 0x28",
      "handlersPending": "0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlersIgnored": "0x02E0 0x02EA 0xD7 0xD8 0x02D6 0xF7 0x17 0x36"
    },

ovenystas, 2025-01-06:
Outdoor: IVT AirX 407
Indoor: IVT AirModule E9
Thermostat/Controller: HMC310/Rego2000 (According to user manual)
(IVT is owned by Bosch and has Bosch components)

```
ems-esp:$ show devices
These EMS devices are currently active:

Värmepanna/VP: CUH/Enviline, Compress 6000AW, Hybrid 3000-7000iAW, SupraEco/Geo 5xx, WLW196i/WSW196i (DeviceID:0x08, ProductID:172, Version:01.21) (125)
 Received telegram type IDs: 0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x494 0x495 0x48F
 Fetched telegram type IDs: 0x14 0xE6 0xEA 0x48D 0x48A 0x4A2 0x485 0x486 0x492 0x488 0x484 0x48B 0x491 0x499 0x49C 0x49D 0x2CC 0x4AA 0x4A7 0x2D6
 Pending telegram type IDs: 0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x4AE 0x4AF 0x4A5
 Ignored telegram type IDs: 0xE7 0x2E0 0x2EA 0x61E 0x49F 0x4A0 0x4A3 0x240 0x29B 0x4AB 0x4AC 0xF7 0x497 0x3B

Termostat: HMC310/RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410 (DeviceID:0x10, ProductID:158, Version:73.04) (51)
 Received telegram type IDs: 0x06 0x2BA 0x2BB 0x2BC 0x2BD 0x2BE 0x2BF 0x2C0 0x31D 0x267
 Fetched telegram type IDs: 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2CC 0x467 0x291 0x292 0x293 0x294 0x2F5 0x23A 0x240 0xBB 0x23E 0x269
 Pending telegram type IDs: 0xA3 0xA2 0x12 0x13 0x2A6 0x2B0 0x29C 0x472 0x2A7 0x2B1 0x29D 0x473 0x2A8 0x2B2 0x29E 0x474 0x2A9 0x2B3 0x29F 0x475 0x2AA 0x2B4 0x2A0 0x476 0x2AB 0x2B5 0x2A1 0x477 0x2AC 0x2B6 0x2A2 0x478 0x2CE 0x468 0x2D0 0x469 0x2D2 0x46A 0x2F6 0x31B 0x31E 0x16E
 Ignored telegram type IDs: 0x508 0x509 0x50A 0x507 0xBF 0x239

ems-esp:$ read 0x08 2
000+00:51:13.230 N 3: [emsesp] boiler(0x08) W me(0x0B), Version(0x02), data: AC 01 15 AD 04 13 00 00 00 00

ems-esp:$ read 0x10 2
000+00:51:43.256 N 4: [emsesp] thermostat(0x10) W me(0x0B), Version(0x02), data: 9E 49 04 00 FF 00 00 00 00

ems-esp:$ read 0x08 1
000+01:05:43.933 N 5: [emsesp] boiler(0x08) W me(0x0B), DeviceName(0x01), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+01:05:43.960 N 6: [emsesp] boiler(0x08) W me(0x0B), DeviceName(0x01), data: 43 55 48 50 (offset 27)
ASCII-decoded: CUHP

ems-esp:$ read 0x10 1
000+01:06:55.575 N 7: [emsesp] thermostat(0x10) W me(0x0B), DeviceName(0x01), data: 32 35 33 30 33 37 35 33 31 31 30 30 34 38 37 33 37 37 31 32 37 36 32 FF FF FF FF
000+01:06:55.643 N 8: [emsesp] thermostat(0x10) W me(0x0B), DeviceName(0x01), data: 48 4D 43 33 31 30 00 00 00 00 00 00 00 00 00 00 (offset 27)
ASCII-decoded: HMC310
```

Negrod, 2025-01-29:
IVT GEO606C:
boiler(0x08) data: AC 03 04 AD 03 04 00 00 00 0D

REGO 3000:
thermostat(0x10) data: FD 2F 07 00 FF 00 00 00 00

RC100H:
thermostat(0x38) data: C8 28 04 C8 00

K 30 RF:
gateway(0x48) data: FC 0B 01 00 01 00 00 00 00 00 00 00

```
Boiler/HP: IVT CUH/Enviline, Compress 6000AW, Hybrid 3000-7000iAW, SupraEco/Geo 5xx, WLW196i/WSW196i (DeviceID:0x08, ProductID:172, Version:03.04) (157)
 Received telegram type IDs: 0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x494 0x495 0x48F
 Fetched telegram type IDs: 0x14 0xE6 0xEA 0x48D 0x48A 0x4A2 0x485 0x486 0x492 0x488 0x484 0x48B 0x491 0x499 0x49C 0x49D 0x2CC 0x4A5 0x4AA 0x4A7 0x2D6
 Pending telegram type IDs: 0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x4AE 0x4AF
 Ignored telegram type IDs: 0xE7 0x2E0 0x2EA 0x240 0x291 0x29B 0x489 0x61E 0x4AC 0xF7 0x497

Thermostat: HMI800.2/Rego 3000, UI800, Logamatic BC400 (DeviceID:0x10, ProductID:253, Version:47.07) (77)
 Received telegram type IDs: 0x06 0x2BA 0x2BB 0x2BC 0x31D 0x267
 Fetched telegram type IDs: 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2CC 0x467 0x291 0x292 0x293 0x294 0x2F5 0x23A 0x240 0xBB 0x23E 0x16E
 Pending telegram type IDs: 0xA3 0xA2 0x12 0x13 0x2A6 0x2B0 0x29C 0x472 0x2A7 0x2B1 0x29D 0x473 0x2A8 0x2B2 0x29E 0x474 0x2A9 0x2BD 0x2B3 0x29F 0x475 0x2AA 0x2BE 0x2B4 0x2A0 0x476 0x2AB 0x2BF 0x2B5 0x2A1 0x477 0x2AC 0x2C0 0x2B6 0x2A2 0x478 0x2CE 0x468 0x2D0 0x469 0x2D2 0x46A 0x2F6 0x31B 0x31E 0x269
 Ignored telegram type IDs: 0x507 0x508 0x509 0x50A 0x59D 0xBF

Thermostat: RC100H/RC100H, CR10H (DeviceID:0x38, ProductID:200, Version:40.04) (4)
 Received telegram type IDs: 0x42B 0x47B
 Fetched telegram type IDs: 0x273
 Pending telegram type IDs: 0xA6A
 Ignored telegram type IDs: 0xBF

Gateway Module: K30RF/WiFi module (DeviceID:0x48, ProductID:252, Version:11.01) (0)
```

VlastiBroucek, 2025-02-18:
Just found that name of my thermostat is at offset 27 of the telegram 0x01
```
ems-esp:$ read 18 1
001+09:51:55.981 N 74: [emsesp] thermostat(0x18) W me(0x0B), DeviceName(0x01), data: 30 35 37 30 38 35 35 32 30 30 30 30 31 37 37 33 38 31 31 31 30 32 32 30 FF FF FF
001+09:51:56.056 N 75: [emsesp] thermostat(0x18) W me(0x0B), DeviceName(0x01), data: 43 52 35 30 00 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
```
43 52 35 30 is CR50

Can other people try the same?

VlastiBroucek, 2025-02-18:
> > Just found that name of my thermostat is at offset 27 of the telegram 0x01
> > 
> > ```
> > ems-esp:$ read 18 1
> > 001+09:51:55.981 N 74: [emsesp] thermostat(0x18) W me(0x0B), DeviceName(0x01), data: 30 35 37 30 38 35 35 32 30 30 30 30 31 37 37 33 38 31 31 31 30 32 32 30 FF FF FF
> > 001+09:51:56.056 N 75: [emsesp] thermostat(0x18) W me(0x0B), DeviceName(0x01), data: 43 52 35 30 00 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
> > ```
> > 
> > 43 52 35 30 is CR50
> > 
> > Can other people try the same?
> 
> pls scroll back to Oct 17th.....

Haha, missed that... Too much going on right now...

ChristianBuechel, 2025-02-19:
Hi everybody, what a great project. Here is the fingerprint of my system, if you need additional infos, I am glad to help.

Boiler is Bosch heatpump CS7001iAW170R-T
ems-esp:$ read 0x08 2
007+10:27:17.499 N 2: [emsesp] boiler(0x08) -> me(0x0B), W, Version(0x02), data: AC 01 14 AD 04 13 00 00 00 01

Thermostat is HPC410
ems-esp:$ read 0x10 2
007+10:27:32.770 N 3: [emsesp] thermostat(0x10) -> me(0x0B), W, Version(0x02), data: 9E 49 04 00 FF 00 00 00 00

Gateway is Bosch Connect-Key K 30 RF, Internet-Gateway
ems-esp:$ read 0x48 2
007+10:27:58.342 N 4: [emsesp] gateway(0x48) -> me(0x0B), W, Version(0x02), data: FC 0B 01 00 01 00 00 00 00 01 00 00