emsesp/EMS-ESP32 #2075: Cant set several setting Junkers ZSBE

anno171986, 2024-10-07:
### DESCRIPTION

_A clear and concise description of what the problem is or the change requested._

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```
system |  
version | "3.7.0-dev.43"
uptime | "000+00:25:34.900"
uptimeSec | 1534
platform | "ESP32"
cpuType | "ESP32-D0WD-V3"
arduino | "Tasmota Arduino v2.0.17"
sdk | "4.4.8.240628"
freeMem | 165
maxAlloc | 103
freeCaps | 115
usedApp | 1690
freeApp | 294
partition | "app0"
resetReason | "Software reset CPU / Software reset CPU"
psram | false
model | ""
network |  
network | "WiFi"
hostname | "Heizung"
RSSI | -73
BSSID | "set"
TxPowerSetting | 78
staticIP | false
lowBandwidth | false
disableSleep | true
enableMDNS | true
enableCORS | false
APProvisionMode | "disconnected"
APSecurity | "wpa2"
APSSID | "ems-esp"
ntp |  
NTPStatus | "connected"
enabled | true
server | "time.google.com"
tzLabel | "Europe/Amsterdam"
mqtt |  
MQTTStatus | "disconnected"
enabled | false
clientID | "ems-esp"
keepAlive | 60
cleanSession | false
entityFormat | 3
base | "ems-esp"
discoveryPrefix | "homeassistant"
discoveryType | 0
nestedFormat | 1
haEnabled | false
mqttQos | 0
mqttRetain | false
publishTimeHeartbeat | 60
publishTimeBoiler | 10
publishTimeThermostat | 10
publishTimeSolar | 10
publishTimeMixer | 10
publishTimeWater | 10
publishTimeOther | 10
publishTimeSensor | 10
publishSingle | false
publish2command | false
sendResponse | false
syslog |  
enabled | false
sensor |  
temperatureSensors | 0
temperatureSensorReads | 0
temperatureSensorFails | 0
analogSensors | 0
analogSensorReads | 0
analogSensorFails | 0
api |  
APICalls | 76
APIFails | 0
bus |  
busStatus | "connected"
busProtocol | "HT3"
busTelegramsReceived | 1217
busReads | 219
busWrites | 6
busIncompleteTelegrams | 0
busReadsFailed | 0
busWritesFailed | 0
busRxLineQuality | 100
busTxLineQuality | 100
settings |  
boardProfile | "CUSTOM"
locale | "de"
txMode | 1
emsBusID | 11
showerTimer | false
showerMinDuration | 180
showerAlert | false
phyType | 0
rxGPIO | 4
txGPIO | 5
dallasGPIO | 14
pbuttonGPIO | 0
ledGPIO | 2
hideLed | false
noTokenApi | false
readonlyMode | false
fahrenheit | false
dallasParasite | false
boolFormat | 6
boolDashboard | 1
enumFormat | 1
analogEnabled | true
telnetEnabled | true
maxWebLogBuffer | 25
webLogBuffer | 25
modbusEnabled | false
forceHeatingOff | false
devices |  
0 |  
type | "boiler"
name | "Condens 2500/5000W, Logamax/Logomatic, Cerapur Top, Greenstar, Generic HT3"
deviceID | "0x08"
productID | 95
brand | ""
version | "18.08"
entities | 68
handlersReceived | "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04"
handlersFetched | "0x14 0x16 0x33"
handlersPending | "0xBF 0xC2 0x1A 0x35 0x26 0x2A 0x2E 0x3B"
handlersIgnored | "0x23"
1 |  
type | "thermostat"
name | "FW100"
deviceID | "0x10"
productID | 105
brand | "Junkers"
version | "13.07"
entities | 23
handlersReceived | "0x06"
handlersFetched | "0x016F 0x0165 0x01D3"
handlersPending | "0xA3 0xA2 0x12 0x13 0x0170 0x0171 0x0172 0x0166 0x0167 0x0168 0xBB 0x23"
handlersIgnored | "0xBE"
2 |  
type | "controller"
name | "HT3"
deviceID | "0x09"
productID | 95
brand | ""
version | "18.08"
entities | 0
```

### TO REPRODUCE

Always there

### EXPECTED BEHAVIOUR

I can't set several Entities of my Junkers ZSBE Boiler (e.g. boiler/boilhystoff , boiler/hyston, boiler/boilhystoff). I expected that this works like it do for boiler/boilhyston.

### SCREENSHOTS

none

### ADDITIONAL CONTEXT

Log output:

```
2024-10-07 20:43:59.196 I 663: [command] Called command boiler/hyston (Einschalttemperaturdifferenz) with value 5 and id 9 on device 0x08
2024-10-07 20:43:59.343 T 664: [emsesp] me(0x0B) -W-> boiler(0x08), UBAParameterWW(0x33), data: 05 (offset 3)
2024-10-07 20:43:59.410 T 665: [emsesp] boiler(0x08) -B-> all(0x00), UBAParameterWW(0x33), data: 08 FF 2D 00 00 46 FF 01 4B 00 FF FF
2024-10-07 20:43:59.749 T 666: [emsesp] me(0x0B) -R-> boiler(0x08), UBAParameterWW(0x33), length: 1 (offset 3)
2024-10-07 20:43:59.771 T 667: [emsesp] boiler(0x08) -W-> me(0x0B), UBAParameterWW(0x33), data: 00 (offset 3)

2024-10-07 20:45:02.083 I 711: [command] Called command boiler/hystoff (Ausschalttemperaturdifferenz) with value 5 and id 9 on device 0x08
2024-10-07 20:45:02.218 T 712: [emsesp] me(0x0B) -W-> boiler(0x08), UBAParameterWW(0x33), data: 05 (offset 4)
2024-10-07 20:45:02.377 T 713: [emsesp] boiler(0x08) -B-> all(0x00), UBAParameterWW(0x33), data: 08 FF 2D 00 00 46 FF 01 4B 00 FF FF
2024-10-07 20:45:02.655 T 714: [emsesp] me(0x0B) -R-> boiler(0x08), UBAParameterWW(0x33), length: 1 (offset 4)
2024-10-07 20:45:02.676 T 715: [emsesp] boiler(0x08) -W-> me(0x0B), UBAParameterWW(0x33), data: 00 (offset 4)

2024-10-07 20:45:33.454 I 730: [command] Called command boiler/boilhystoff (Ausschaltdifferenz) with value 5
2024-10-07 20:45:33.577 T 731: [emsesp] me(0x0B) -W-> boiler(0x08), UBAParameters(0x16), data: 05 (offset 4)
2024-10-07 20:45:33.674 T 732: [emsesp] boiler(0x08) -B-> all(0x00), UBAParameters(0x16), data: FF 3A 64 00 00 FA 05 01 03 64 0A 05
2024-10-07 20:45:33.907 T 733: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 22 01 56 64 25 09 03 25 C0 80 00 80 00 01 31 FF FF 0A 00 00 00 00 00 02 18
2024-10-07 20:45:34.129 T 734: [emsesp] boiler(0x08) -W-> 13(0x13), ?(0x05), data: 00 (offset 34)
2024-10-07 20:45:34.460 T 735: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorWW(0x34), data: 0F 01 F1 01 F1 A1 00 00 03 00 01 0D 5A 00 09 91 00
2024-10-07 20:45:34.700 T 736: [emsesp] me(0x0B) -R-> boiler(0x08), UBAParameters(0x16), length: 1 (offset 4)
2024-10-07 20:45:34.728 T 737: [emsesp] boiler(0x08) -W-> me(0x0B), UBAParameters(0x16), data: 00 (offset 4)

2024-10-07 20:47:34.597 I 826: [command] Called command boiler/boilhyston (Einschaltdifferenz) with value -5
2024-10-07 20:47:34.738 T 827: [emsesp] me(0x0B) -W-> boiler(0x08), UBAParameters(0x16), data: FB (offset 5)
2024-10-07 20:47:34.810 T 828: [emsesp] boiler(0x08) -B-> all(0x00), UBAParameters(0x16), data: FF 3A 64 00 00 FB 05 01 03 64 0A 05
2024-10-07 20:47:35.144 T 829: [emsesp] me(0x0B) -R-> boiler(0x08), UBAParameters(0x16), length: 1 (offset 5)
2024-10-07 20:47:35.164 T 830: [emsesp] boiler(0x08) -W-> me(0x0B), UBAParameters(0x16), data: FB (offset 5)
```

proddy, 2024-10-10:
is there a way you can manually change these settings on your system and use EMS-ESP to monitor the telegram traffic? That way we can verify the mappings.

anno171986, 2024-10-11:
Until now I didn't found any of this settings on the control panel, also not the one which I can set. I will have a deeper look into the settings. I searched also the manual but can't find it until now.

anno171986, 2024-10-21:
I can't find it. Maybe someone have the same heater and can help