emsesp/EMS-ESP32 #2077: Support for ESP32(-S3) temperature sensor

proddy, 2024-10-09:
As @bbqkees made me aware, the ESP32(-S3) has an internal temperature sensor. To do

1 - auto-add as a temperaturesensor for this board, so it's shown in the Sensors
2 - write a FAQ on how to use the Scheduler to send an alert if the EMS-ESP board gets too hot

https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/temp_sensor.html

bbqkees, 2024-10-09:
I would classify it as a system sensor as well, next to the free RAM, used flash etc.

proddy, 2024-10-09:
Yes, adding as a system attribute makes much more sense.

MichaelDvP, 2024-10-10:
I've tested the sensor, works good. But nasty that ESP32 does not have one and IDF 4.x implementation is different from IDF 5.x. A lot of ' #if' 
Added in mqtt-heartbeat and api/system/info, in web i've added it to hardware status-CPU
[image]
Should i add to my dev and make a PR?

proddy, 2024-10-10:
I think we can add it to 3.7.0 while I finish up the Dashboard

proddy, 2024-10-17:
works great. a nice addition.

closed 2024-10-17 by proddy