emsesp/EMS-ESP32 #2078: ESP32 entity values end up in HA in the wrong language depending on ESP32 entity language setting

reiffert2, 2024-10-09:
### DESCRIPTION

ESP32-D0WD on 3.7.0-dev.43 feeding values to MQTT on HomeAssistant. 

ESP32 Language for entities set to <b>German</b> results in:
<pre>
Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:123
integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 9:26:09 AM (17 occurrences)
Last logged: 9:26:09 AM

Invalid option for select.thermostat_dhw_circulation_pump_mode: 'aus' (valid options: ['off', 'on', 'auto', 'own prog'])
Invalid option for select.thermostat_hc1_nofrost_mode: 'Außen' (valid options: ['room', 'outdoor', 'room outdoor'])
Invalid option for select.thermostat_hc1_control_mode: 'Wetter kompensiert' (valid options: ['weather compensated', 'outside basepoint', 'n/a', 'room', 'power', 'constant'])
Invalid option for select.thermostat_hc2_switch_program_mode: 'Absolut' (valid options: ['level', 'absolute'])
Invalid option for select.thermostat_building_type: 'Mittel' (valid options: ['light', 'medium', 'heavy'])
</pre>
ESP32 Language for entities set to <b>English</b> results in:
<pre>
Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:123
integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 9:26:09 AM (63 occurrences)
Last logged: 9:29:59 AM

Invalid option for select.thermostat_dhw_circulation_pump_mode: 'off' (valid options: ['aus', 'an', 'auto', 'Eigenprog.'])
Invalid option for select.thermostat_hc1_nofrost_mode: 'outdoor' (valid options: ['Raum', 'Außen', 'Raum + Außen'])
Invalid option for select.thermostat_hc1_control_mode: 'weather compensated' (valid options: ['Wetter kompensiert', 'Basispunkt Außentemp.', 'n/a', 'Raum', 'Leistung', 'konstant'])
Invalid option for select.thermostat_hc2_switch_program_mode: 'absolute' (valid options: ['Level', 'Absolut'])
Invalid option for select.thermostat_building_type: 'medium' (valid options: ['Leicht', 'Mittel', 'Schwer'])
</pre>

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [X] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{"system":{"version":"3.7.0-dev.43","uptime":"000+00:10:00.729","uptimeSec":600,"platform":"ESP32","cpuType":"ESP32-D0WD","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","freeMem":133,"maxAlloc":59,"freeCaps":83,"usedApp":1690,"freeApp":294,"partition":"app1","resetReason":"Software reset CPU / Software reset CPU","psram":false,"model":""},"network":{"network":"Ethernet","hostname":"ems-esp","TxPowerSetting":78,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"never","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"time.google.com","tzLabel":"Europe/Berlin"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":753,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTConnects":3,"enabled":true,"clientID":"ems-esp","keepAlive":60,"cleanSession":false,"entityFormat":0,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":true,"syslogStarted":true,"syslogLevel":"notice","syslogIP":"10.20.0.2","syslogQueue":0},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"Buderus","busTelegramsReceived":1456,"busReads":429,"busWrites":0,"busIncompleteTelegrams":11,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"E32","locale":"en","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":true,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":25,"webLogBuffer":25,"modbusEnabled":false,"forceHeatingOff":false},"devices":[{"type":"boiler","name":"Condens 5000i, Greenstar 8000/GC9800IW, GB192i*2","deviceID":"0x08","productID":195,"brand":"Buderus","version":"03.07","entities":73,"handlersReceived":"0xBF 0xC2 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x04","handlersFetched":"0x14 0xE6 0xEA","handlersPending":"0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A 0x2E 0x3B","handlersIgnored":"0x0CB8 0x0CB9 0x02D6 0xE7 0x02E0 0x02EA 0xD2 0xE2"},{"type":"thermostat","name":"RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410","deviceID":"0x10","productID":158,"brand":"","version":"74.03","entities":91,"handlersReceived":"0x06 0x02BB 0x02BC 0x02BE 0x02BF 0x02C0 0x031D 0x0267","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x02CC 0x0291 0x02CE 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x023A 0x0240 0x0269","handlersPending":"0xA3 0xA2 0x12 0x13 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02D0 0x0469 0x02D2 0x046A 0x031E","handlersIgnored":"0xBF"},{"type":"mixer","name":"MM100","deviceID":"0x21","productID":160,"brand":"","version":"24.05","entities":7,"handlersReceived":"0x02D8","handlersFetched":"0x02CE","handlersIgnored":"0x02E2 0x02EC 0xE6 0x0255 0x0292 0xBF"},{"type":"controller","name":"9000i","deviceID":"0x09","productID":224,"brand":"","version":"35.06","entities":0}]}

```

### TO REPRODUCE

Setup HomeAssistant, MQTT (core-mosquitto), setup HA with Settings->System->General->Language: English

### EXPECTED BEHAVIOUR

Identify how to fix this, whether this is an ESP32, HA or MQTT topic. If ESP32, fix the issue

### SCREENSHOTS

See Description

### ADDITIONAL CONTEXT

N/A

**(Please remember to close the issue when it has been addressed)**

proddy, 2024-10-09:
It may be just left overs old MQTT topics when switching between languages as the topics are retained, but not sure. I'm away from a week so it's hard to test (no HA). What you could try is using MQTT Explorer and manually deleting everything under homeassistant and restart EMS-ESP?

reiffert2, 2024-10-09:
MQTT explorer found the values in the correct language and according to the entity language setting on ESP32 when targeting HomeAssistant with MQTT explorer.
I then went ahead and was deleting all ESP32 related devices in HomeAssistant, was restarting ESP32 and the error messages are no longer observable. 
Would we want to dig deeper into changing the entity language or is this more a HA/MQTT topic?

MichaelDvP, 2024-10-09:
The HA topics are created on startup and it takes a while (some minutes). If you change the language you have to reboot emsesp and wait until all HA entires are update.

reiffert2, 2024-10-09:
That explains it. Thank you all!

closed 2024-10-09 by reiffert2

MichaelDvP, 2024-10-09:
I've checked, reboot is not needed. Changing thelanguage restarts mqtt andrebuilds the configuration. But still it needs time to send all the configs. So in the first minutes you will get a warning on each publish.