emsesp/EMS-ESP32 #2079: RC310: dhw1 missing

richii93, 2024-10-09:
### DESCRIPTION

I did update from 3.7.0_dev.4 to 3.7.0_dev.43 but now I am missing the warm water entities dhw1 in the RC310. I do see dhw2 instead but it does not change anything.

### REQUESTED INFORMATION

```json
  {"system":{"version":"3.7.0-dev.43","uptime":"000+00:06:55.326","uptimeSec":415,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","freeMem":143,"maxAlloc":67,"freeCaps":93,"usedApp":1690,"freeApp":294,"partition":"app1","resetReason":"Software reset CPU / Software reset CPU","psram":false,"model":""},"network":{"network":"WiFi","hostname":"ems-Scheune","RSSI":-81,"BSSID":"set","TxPowerSetting":78,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"time.google.com","tzLabel":"Europe/Amsterdam"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":415,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTConnects":1,"enabled":true,"clientID":"EMS-Scheune","keepAlive":60,"cleanSession":true,"entityFormat":2,"base":"EMS-Scheune","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":true,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0},"api":{"APICalls":1,"APIFails":1},"bus":{"busStatus":"connected","busProtocol":"Buderus","busTelegramsReceived":1164,"busReads":364,"busWrites":0,"busIncompleteTelegrams":1,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"CUSTOM","locale":"en","txMode":2,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"phyType":0,"rxGPIO":23,"txGPIO":5,"dallasGPIO":19,"pbuttonGPIO":0,"ledGPIO":2,"hideLed":false,"noTokenApi":true,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":false,"telnetEnabled":true,"maxWebLogBuffer":25,"webLogBuffer":24,"modbusEnabled":false,"forceHeatingOff":false},"devices":[{"type":"boiler","name":"Topline, GB162","deviceID":"0x08","productID":115,"brand":"Buderus","version":"05.09","entities":61,"handlersReceived":"0x10 0x11 0x15 0x1C 0x18 0x19 0x04","handlersFetched":"0x14 0x16 0x33","handlersPending":"0xBF 0xC2 0x1A 0x35 0x34 0x2A 0xD1 0xE3 0xE4 0xE5 0xE9 0x2E 0x3B","handlersIgnored":"0x1E"},{"type":"thermostat","name":"RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410","deviceID":"0x10","productID":158,"brand":"","version":"74.04","entities":133,"handlersReceived":"0x06 0xA2 0x02BC 0x02BD 0x02BF 0x02C0 0x031D 0x0267","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x02A7 0x02BB 0x02B1 0x029D 0x0291 0x02CE 0x0292 0x02D0 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269","handlersPending":"0xA3 0x12 0x13 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02D2 0x046A 0x031E","handlersIgnored":"0xBF"},{"type":"thermostat","name":"RC100, CR10, Moduline 1000/1010","deviceID":"0x3A","productID":165,"brand":"","version":"40.04","entities":2,"handlersReceived":"0x042D","handlersFetched":"0x0275","handlersPending":"0x047D","handlersIgnored":"0xBF"},{"type":"mixer","name":"MM100","deviceID":"0x20","productID":160,"brand":"","version":"24.05","entities":7,"handlersReceived":"0x02D7","handlersFetched":"0x02CD","handlersIgnored":"0x16 0x33 0x0255 0x0291 0x02E1 0x02EB 0xBF"},{"type":"mixer","name":"MM100","deviceID":"0x21","productID":160,"brand":"","version":"24.05","entities":7,"handlersReceived":"0x02D8","handlersFetched":"0x02CE","handlersIgnored":"0x16 0x33 0x0255 0x0292 0x02E2 0x02EC"},{"type":"mixer","name":"MM100","deviceID":"0x22","productID":160,"brand":"","version":"24.05","entities":7,"handlersReceived":"0x02D9","handlersFetched":"0x02CF","handlersIgnored":"0x0255 0x02E3 0x02ED 0x16 0x33 0x0293 0xBF"},{"type":"controller","name":"BC10, RFM20","deviceID":"0x09","productID":68,"brand":"","version":"02.03","entities":0},{"type":"water","name":"MM100","deviceID":"0x28","productID":160,"brand":"","version":"24.05","entities":8,"handlersReceived":"0x0331","handlersFetched":"0x0313","handlersIgnored":"0x0255 0x033B 0x16 0x33"}]}

```

### TO REPRODUCE

When I downgrade to version 3.7.0_dev.4 it works again.

### SCREENSHOTS

[image]

version: dev.4

[image]

version: dev.43

MichaelDvP, 2024-10-09:
Please make a full log of ~ 2-3 minutes.

richii93, 2024-10-09:
[log.txt](https://github.com/user-attachments/files/17308228/log.txt)

MichaelDvP, 2024-10-09:
Hmm, we support more than one dhw circuit since #1634. 
I thought the first byte in 0x2F5/0x2F6 shows the dhw circuit. It was 00 for no circuit and 01 for dhw1 02 for dhw2. You have 02, so it's mapped as dhw2. 
Maybe this is a marker for 00- no dhw, 01 - boiler managed, 02 - mixer managed. I don't know.

But actual on your system it only shows as dhw2, but all should work.

richii93, 2024-10-09:
I don’t know much about bytes and markers - that’s why I am glad that you are here. All I know is that it did show as dhw1 in version dev04 and changing “dhw/mode” did work and now in version dev43 it shows as dhw2 (that would not be a problem), but when I try to change “dhw2/mode”, nothing happens (I use this quite often to turn on or off the water circuit). Am I missing something here?

MichaelDvP, 2024-10-09:
> when I try to change “dhw2/mode”, nothing happens (

Can you please make a full log of this.

richii93, 2024-10-09:
[log 2.txt](https://github.com/user-attachments/files/17317782/log.2.txt)

MichaelDvP, 2024-10-10:
Thanks, there is a bug in the mapping of dhw circuits, it writes the mode to 0x2F6 instead of 0x2F5.
Please check if t works with my builds: https://github.com/MichaelDvP/EMS-ESP32/releases
- dev.43 is the same as official dev with a fix for the ' dhw2'  circuit
- test.43 maps to ' dhw'  instead of' dhw2'  for your system (and is the test for thermostat timer programs).

Would be nice if you can check both versions and send a log if something does not work correctly.

richii93, 2024-10-10:
I tried both and I do get an error message in both versions when I try to change dhw/mode

log dev.43:

[log 3.txt](https://github.com/user-attachments/files/17325576/log.3.txt)

log test.43:

[log 4.txt](https://github.com/user-attachments/files/17325582/log.4.txt)

MichaelDvP, 2024-10-10:
After a restart emsesp needs some time to collect the data from the bus. The command only works if the entity is registered and shown. Wait a few minutes before sending a command. (Check in web that the entiy is shown with value).

richii93, 2024-10-10:
I did also wait a few minutes and the entity is shown in the web interface with value. I also use the web interface to change the value.

[image]

And when I try to change it, it says “Request failed”.

MichaelDvP, 2024-10-10:
Oh, the same with my RC35 (only one dhw circuit). i've fixed it in both builds, same version number, but you need to download again and flash.

richii93, 2024-10-10:
Now it does work in both versions.

Thank you very much.

closed 2024-10-10 by richii93

MichaelDvP, 2024-10-10:
@proddy 
what should we do in 3.7.0. I'm not sure what is right. 
In #1634 we have 2 water circuits and two mode telegrams 0x2F5 and 0x2F6,
- first byte is 01 in 0x2F5, 02 in 0x2F6, first dhw circuit is managed by boiler, second by mixer 0x29 (dhw2)
- in systems with only one dhw circuit first byte was always 01 in 2F5 and 00 in 2F6
- When solar (SM100) module is used for dhw it is dhw3, with module address 0x2A,.., but i did not see the thermostat telegrams for this.
- Now we have only one dhw circuit and first byte of 2F5 is 02. 

I thought that the fist byte is the dhw circuit and handled telegram offset and dhw circuit number separate, but here we have mixer at 0x28 (dhw1). 
I now think the first byte is 00- no dhw, 01- handled by boiler, 02- handled by mixer. 
So the right way is to set the dhw circuit only on telegram offset only. 
Remove the extra circuit number and store only offset? Or leave the circuit number in code, but ignore it?

proddy, 2024-10-11:
> @proddy what should we do in 3.7.0. I'm not sure what is right. In #1634 we have 2 water circuits and two mode telegrams 0x2F5 and 0x2F6,
>
> * first byte is 01 in 0x2F5, 02 in 0x2F6, first dhw circuit [...]

I would say remove the circuit number and calculate based on the offset. That way it will match the messaging seen in the built-in interfaces. I'm trying to think if it will break anything....