emsesp/EMS-ESP32 #2082: serial console broken on esp32S3

MichaelDvP, 2024-10-10:
Since a few versions the serial console is dead on the esp32s3, tested on Gateway S3 and Liligo module. 
It works on esp32 for both versions S32 (tasmota platform) and E32V2(psram, espressif platform).
Maybe because the esp32s3 have a native USB and not a usb/uart chip?
I've checked older versions, but it's a while since i last used the S3. I've only found dev32 from 18.8.24 and this is working. 
For dev 33-dev43 i have only esp32 bins. I tried to checkout older builds an compile, but the nasty module import breaks.
I tested current version compiled with platform 6.8.1 or pioarduino 3.0.5, also not working serial console.

I can't find a change that could cause this.

serial output on boot:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4214b432
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
```

proddy, 2024-10-10:
can you explain "not working" ?

MichaelDvP, 2024-10-10:
When connecting a usb cable i have a valid serial port. 
Open putty and connect to serial port works, no output except esp bootloader message `ESP-ROM:esp32s3-20210327`
No reaction on keys ctrl-S. 
restart emsesp via web gives the bootmessage from first post, nothing more, port stays connected.
Looks like this https://github.com/emsesp/EMS-ESP32/blob/60ab5a3e4294af2024c17d62ed2f6184a84c89dd/src/emsesp.cpp#L1543 does not work, a serial_console_.print(...); after begin does not give a message. 

But no issue with esp32.

proddy, 2024-10-10:
That's odd, haven't seen that behaviour and I've been testing on all boards with Windows, Linux and Mac. I know some of the Gateway boards have two USBs and one is use for I/O (the other for flashing) but you already know that. What it could be is the boot partition. I've noticed when using pio to upload a .bin it would reboot and use the old partition. I have to manually 'restart boot' to get it to use the latest. I'm still not sure what is causing this. I'll run some tests here too and see what I can find

MichaelDvP, 2024-10-10:
i have the 3 partitions boot/app0/app1 with different software. i can change from telnet with `restart app0`, etc. and have the board connected to serial the same time. 
Now app0: dev 31, app1: dev44
a `restart app0` shows in serial terminal on com7: 
```

┌───────────────────────────────────────┐
│  EMS-ESP version 3.7.0-dev.31 (D)     │
│                                       │
│  help to show available commands      │
│  su to access admin commands          │
│                                       │
│  https://github.com/emsesp/EMS-ESP32  │
│                                       │
└───────────────────────────────────────┘

ems-esp:#
Applying defaults to /config/networkSettings.json

Writing settings to /config/networkSettings.json {"ssid":"HZTD67Fr45d","bssid":"","password":"kd-3575GH5rjZ","hostname":"ems-esp","static_ip_config":false,"bandwidth20":false,"nosleep":false,"enableMDNS":true,"enableCORS":false,"CORSOrigin":"*","tx_power":0}

Applying defaults to /config/apSettings.json

Writing settings to /config/apSettings.json {"provision_mode":1,"ssid":"ems-esp","password":"ems-esp-neo","channel":1,"ssid_hidden":false,"max_clients":4,"local_ip":"192.168.4.1","gateway_ip":"192.168.4.1","subnet_mask":"255.255.255.0"}

Applying defaults to /config/ntpSettings.json
```
etc: and i can enter commands
a `restart app1` shows in serial terminal on com7: 
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4214b432
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
```
and nothing happens if hit a key. But still i can connect web and telnet and restart from telnet to the other partiton and com7-serial will show up dev 31 welcome message again.

proddy, 2024-10-10:
I just tried it on a WEMOS Lolin S3, `pio run -t erase` to start fresh and then `pio run -e s3_16M_P -t upload` and I could get to the Serial on both my Macbook and Windows. No idea what is wrong with your setup.

bbqkees, 2024-10-11:
Both the S3 Gateways and the Liligo T7 S3 use the native USB port of the ESP32-S3.

If you have one of my earlier S3 prototypes (probable a purple board), there is also a second micro-USB port on the inside connected to the UART via a CH340. So they have both.
Then you could check if serial works on the UART.

MichaelDvP, 2024-10-11:
> Then you could check if serial works on the UART.

Yes it's the purple board and with internal USB the serial console works on this port, but not on the USB-C. 
I think the Lolin S3 also have 2 ports.
On the Liligo there is only USB-C. What is bothering is, that it works on this USB with dev32, but not with actual. I've tried to checkout older versions after a version count-up and use PIO upload&monitor to check it i get a emsesp-prompt.
Now i'm on dev34, not working, getting closer.

proddy, 2024-10-11:
> Yes it's the purple board and with internal USB the serial console works on this port, but not on the USB-C.

That's correct. Same with the Lolin S3, it has 2 USBs (UART & USG). You can't use Serial over USG. https://docs.espressif.com/projects/esp-iot-solution/en/latest/usb/usb_overview/usb_otg.html

bbqkees, 2024-10-11:
The esptool.py also uses a serial port as well.
https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/get-started/establish-serial-connection.html
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/usb-otg-console.html#usb-otg-console

So it should work via the native USB port as well.

MichaelDvP, 2024-10-11:
Found it. 
A bug in pio.ini overwriting board.build_flags for the S3. I'll fix.

closed 2024-10-11 by MichaelDvP

proddy, 2024-10-20:
now I'm getting the same issue, no Serial to S3. I'll dig deeper and try and figure out why

MichaelDvP, 2024-10-20:
Have you checked your pio_local.ini? For me it's working with unmodified lolin_s3 board setting.