emsesp/EMS-ESP32 #2084: Difference between `watch on` and `watch raw` with extended data types

Sbried, 2024-10-11:
### DESCRIPTION

I found a difference in the value of the first byte of the number of extended data types, depending if you are watching the bus traffic with `watch on` or `watch raw`. In formatted output with `watch on` the first byte is always 1 higher than in raw mode.
With normal data types (not extended) the output is the same between raw and formatted.

Some examples comparing the same messages in raw and formatted:

[telegram] Rx:0B                    90 FF 00 FF   **0191** 7A
[emsesp] me(0x0B) -R-> thermostat(0x10), HPMode(0x**0291**), length: 255

[telegram] Rx:       10            0B FF 00      **0191**         01 00 00 00 FF 02 9F
[emsesp] thermostat(0x10) -W-> me(0x0B), HPMode(0x**0291**), data: 01 00 00 00 FF 02

[telegram] Rx: 0B                  90 FF 00 FF   **0192** 79
[emsesp] me(0x0B) -R-> thermostat(0x10), HPMode(0x**0292**), length: 255

[telegram] Rx:       10            0B FF 00      **0192**         00 00 00 00 FF 00 7D
[emsesp] thermostat(0x10) -W-> me(0x0B), HPMode(0x**0292**), data: 00 00 00 00 FF 00

[telegram] Rx: 0B                  90 FF 00 FF   **0193** 78
[emsesp] me(0x0B) -R-> thermostat(0x10), HPMode(0x**0293**), length: 255

[telegram] Rx:       10            0B FF 00      **0193**         00 00 00 00 FF 00 3D
[emsesp] thermostat(0x10) -W-> me(0x0B), HPMode(0x**0293**), data: 00 00 00 00 FF 00

[telegram] Rx: 0B                  90 FF 00 FF   **0194** 7F
[emsesp] me(0x0B) -R-> thermostat(0x10), HPMode(0x**0294**), length: 255

[telegram] Rx:       10            0B FF 00      **0194**         00 00 00 00 FF 00 E4
[emsesp] thermostat(0x10) -W-> me(0x0B), HPMode(0x**0294**), data: 00 00 00 00 FF 00

This drove me crazy yesterday when I wanted to compare the data type specification and bus variable names with what is going on on the bus ;-)

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

{
    "system": {
        "version": "3.7.0-dev.43",
        "uptime": "000+15:38:11.654",
        "uptimeSec": 56291,
        "platform": "ESP32",
        "cpuType": "ESP32-D0WD-V3",
        "arduino": "ESP32 Arduino v2.0.17",
        "sdk": "v4.4.7-dirty",
        "freeMem": 156,
        "maxAlloc": 107,
        "freeCaps": 4173,
        "usedApp": 2333,
        "freeApp": 3363,
        "partition": "app0",
        "resetReason": "Software reset CPU / Software reset CPU",
        "psram": true,
        "psramSize": 4093,
        "freePsram": 4052,
        "model": "BBQKees Gateway Model EMS Gateway E32 V2 v2.0/2024071"
    },
    "network": {
        "network": "Ethernet",
        "hostname": "ems-esp",
        "TxPowerSetting": 0,
        "staticIP": false,
        "lowBandwidth": false,
        "disableSleep": true,
        "enableMDNS": true,
        "enableCORS": false,
        "APProvisionMode": "disconnected",
        "APSecurity": "wpa2",
        "APSSID": "ems-esp"
    },
    "ntp": {
        "NTPStatus": "disconnected",
        "enabled": false,
        "server": "time.google.com",
        "tzLabel": "Europe/Amsterdam"
    },
    "mqtt": {
        "MQTTStatus": "connected",
        "MQTTPublishes": 12442,
        "MQTTQueued": 0,
        "MQTTPublishFails": 0,
        "MQTTConnects": 1,
        "enabled": true,
        "clientID": "esp32-2cb865e4",
        "keepAlive": 60,
        "cleanSession": false,
        "entityFormat": 0,
        "base": "ems-esp",
        "discoveryPrefix": "homeassistant",
        "discoveryType": 0,
        "nestedFormat": 1,
        "haEnabled": true,
        "mqttQos": 0,
        "mqttRetain": false,
        "publishTimeHeartbeat": 60,
        "publishTimeBoiler": 10,
        "publishTimeThermostat": 10,
        "publishTimeSolar": 10,
        "publishTimeMixer": 10,
        "publishTimeWater": 10,
        "publishTimeOther": 10,
        "publishTimeSensor": 10,
        "publishSingle": false,
        "publish2command": false,
        "sendResponse": false
    },
    "syslog": {
        "enabled": false
    },
    "sensor": {
        "temperatureSensors": 0,
        "temperatureSensorReads": 0,
        "temperatureSensorFails": 0,
        "analogSensors": 0,
        "analogSensorReads": 0,
        "analogSensorFails": 0
    },
    "api": {
        "APICalls": 388,
        "APIFails": 6
    },
    "bus": {
        "busStatus": "connected",
        "busProtocol": "Buderus",
        "busTelegramsReceived": 202946,
        "busReads": 48979,
        "busWrites": 4,
        "busIncompleteTelegrams": 2,
        "busReadsFailed": 0,
        "busWritesFailed": 0,
        "busRxLineQuality": 100,
        "busTxLineQuality": 100
    },
    "settings": {
        "boardProfile": "E32V2",
        "locale": "de",
        "txMode": 2,
        "emsBusID": 11,
        "showerTimer": false,
        "showerMinDuration": 180,
        "showerAlert": false,
        "hideLed": false,
        "noTokenApi": false,
        "readonlyMode": false,
        "fahrenheit": false,
        "dallasParasite": false,
        "boolFormat": 5,
        "boolDashboard": 1,
        "enumFormat": 1,
        "analogEnabled": true,
        "telnetEnabled": true,
        "maxWebLogBuffer": 50,
        "webLogBuffer": 18,
        "modbusEnabled": false,
        "forceHeatingOff": false
    },
    "devices": [
        {
            "type": "boiler",
            "name": "Enviline, Compress 6000AW, Hybrid 3000-7000iAW, SupraEco/Geo 5xx, WLW196i/WSW196i",
            "deviceID": "0x08",
            "productID": 172,
            "brand": "",
            "version": "02.02",
            "entities": 155,
            "handlersReceived": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
            "handlersFetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x02CC 0x04AA 0x02D6",
            "handlersPending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x04AE 0x04AF 0x04A5 0x04A7",
            "handlersIgnored": "0xF7 0x02E0 0x02EA 0x061E 0x0240 0x049F 0x04A0 0x04A3 0xE7 0x0497 0x2E 0x3B"
        },
        {
            "type": "thermostat",
            "name": "RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410",
            "deviceID": "0x10",
            "productID": 158,
            "brand": "",
            "version": "73.04",
            "entities": 67,
            "handlersReceived": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BF 0x02C0 0x031D 0x0267",
            "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E 0x0269",
            "handlersPending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x02F6 0x031B 0x031E",
            "handlersIgnored": "0x0508 0x0509 0x050A 0x0507 0x0239 0xBF 0xF7"
        },
        {
            "type": "gateway",
            "name": "WiFi module",
            "deviceID": "0x48",
            "productID": 252,
            "brand": "Buderus",
            "version": "08.02",
            "entities": 0,
            "handlersIgnored": "0xF9 0x2040 0x0495 0x0494 0x3B 0xE5 0xE9 0x0DE8 0x0DE9 0x0DEA 0x0DEC 0x0DED 0x0DEF 0x0DF0 0x0DF1"
        }
    ]
}

### TO REPRODUCE

In the console: trace bus traffic with `watch on` and with `watch raw`

### EXPECTED BEHAVIOUR

Independent of the used watch command, the output data shold be the same.

### SCREENSHOTS

examples are in the description of the problem.

### ADDITIONAL CONTEXT

none

**(Please remember to close the issue when it has been addressed)**

proddy, 2024-10-11:
https://github.com/emsesp/EMS-ESP32/discussions/2025

Sbried, 2024-10-12:
OK, understood that this way of storing the data has the purpose to save memory.
But are there any objections to substract 0x0100 before printing out to the console? Then the information would be correct and match the raw data.
Hopefully the EMS+ spec will not define a data type with an extended address 0xFF.. Then the strategy will fail ;-)

closed 2024-10-16 by Sbried

proddy, 2024-10-16:
or just remove `watch raw` as it confuses people. it's only for power users and people that understand the code

Sbried, 2024-10-16:
Removing may not be good idea as it is essential to see what is really sent over the bus.
But ok, I at understood the reason behind and can deal now with the difference.