emsesp/EMS-ESP32 #2088: Missing entitie for low-temperature boilers pump start temp

SMHRambo, 2024-10-13:
I have a Buderus MC10 with RC310 Controller.
The MC10 is installed in an Buderus S315T low-temperature boiler.
For low-temperature boilers, there is a setting available to define a minimum switch-on temperature for the pump.
But the value is missing in ems-esp.

The telegram is:
thermostat(0x10) -W-> boiler(0x08), UBAParameters(0x16), data: 22 (offset 23)

```json
{
  "system": {
    "version": "3.7.0-dev.43",
    "uptime": "000+01:23:13.128",
    "uptimeSec": 4993,
    "platform": "ESP32",
    "cpuType": "ESP32-D0WDQ6",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "freeMem": 158,
    "maxAlloc": 83,
    "freeCaps": 108,
    "usedApp": 1690,
    "freeApp": 294,
    "partition": "app0",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": false,
    "model": ""
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -69,
    "TxPowerSetting": 78,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "connected",
    "enabled": true,
    "server": "10.13.12.15",
    "tzLabel": "Europe/Berlin"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 1845,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTConnects": 1,
    "enabled": true,
    "clientID": "ems-esp",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 3,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 2,
    "haEnabled": true,
    "mqttQos": 1,
    "mqttRetain": true,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 0,
    "publishTimeThermostat": 0,
    "publishTimeSolar": 0,
    "publishTimeMixer": 0,
    "publishTimeWater": 10,
    "publishTimeOther": 10,
    "publishTimeSensor": 0,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 1,
    "temperatureSensorReads": 998,
    "temperatureSensorFails": 0
  },
  "api": {
    "APICalls": 880,
    "APIFails": 82
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "Buderus",
    "busTelegramsReceived": 6420,
    "busReads": 1681,
    "busWrites": 1,
    "busIncompleteTelegrams": 1,
    "busReadsFailed": 0,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "CUSTOM",
    "locale": "de",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": true,
    "showerMinDuration": 180,
    "showerAlert": false,
    "phyType": 0,
    "rxGPIO": 23,
    "txGPIO": 5,
    "dallasGPIO": 14,
    "pbuttonGPIO": 0,
    "ledGPIO": 25,
    "hideLed": true,
    "noTokenApi": true,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 6,
    "boolDashboard": 2,
    "enumFormat": 1,
    "analogEnabled": false,
    "telnetEnabled": false,
    "maxWebLogBuffer": 25,
    "webLogBuffer": 7,
    "modbusEnabled": false,
    "forceHeatingOff": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "Logano GB1*5, Logamatic MC10",
      "deviceID": "0x08",
      "productID": 72,
      "brand": "",
      "version": "03.05",
      "entities": 72,
      "handlersReceived": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x34 0x04",
      "handlersFetched": "0x16 0x33",
      "handlersPending": "0xBF 0xC2 0x1A 0x35 0x2E 0x3B"
    },
    {
      "type": "thermostat",
      "name": "RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410",
      "deviceID": "0x10",
      "productID": 158,
      "brand": "",
      "version": "74.05",
      "entities": 57,
      "handlersReceived": "0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269",
      "handlersPending": "0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E",
      "handlersIgnored": "0xBF"
    },
    {
      "type": "mixer",
      "name": "MM50",
      "deviceID": "0x20",
      "productID": 159,
      "brand": "",
      "version": "20.03",
      "entities": 7,
      "handlersReceived": "0x02D7",
      "handlersFetched": "0x02CD",
      "handlersIgnored": "0x02E1 0x02EB 0x0255 0x16 0x33 0xBF"
    },
    {
      "type": "controller",
      "name": "BC10, RFM20",
      "deviceID": "0x09",
      "productID": 68,
      "brand": "",
      "version": "02.03",
      "entities": 0
    },
    {
      "type": "temperaturesensor",
      "name": "temperaturesensor",
      "entities": 1
    },
    {
      "type": "custom",
      "name": "custom",
      "entities": 1
    }
  ]
}
```
[image]

proddy, 2024-10-13:
Could you create a Custom Entity to test this, both reading and writing? When it works, we can add it to the C++ code. That will save us hours of troubleshooting. thanks.

SMHRambo, 2024-10-14:
I have already tested it with a custom entity.
Both reading and writing work.
[image]
[image]

proddy, 2024-10-14:
perfect, it's an easy add then. thanks.

proddy, 2024-10-14:
In 3.7.0-dev.45

closed 2024-10-14 by proddy