emsesp/EMS-ESP32 #2090: Buderus BC400 Away mode and holiday mode missing

suaveolent, 2024-10-14:
### DESCRIPTION

The Buderus BC400 has an Away and Holiday Mode. Away is a toggle, holiday can have a start and end date, a target temperature as well as a custom name.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{"system":{"version":"3.7.0-dev.44","uptime":"000+01:44:17.909","uptimeSec":6257,"platform":"ESP32S3","cpuType":"ESP32-S3","arduino":"ESP32 Arduino v2.0.17","sdk":"v4.4.7-dirty","freeMem":196,"maxAlloc":179,"freeCaps":8325,"usedApp":2242,"freeApp":3710,"partition":"app1","resetReason":"Power on reset / Power on reset","psram":true,"psramSize":8189,"freePsram":8130,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-46,"TxPowerSetting":0,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"disconnected","enabled":false,"server":"time.google.com","tzLabel":"Europe/Amsterdam"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":2926,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTConnects":1,"enabled":true,"clientID":"esp32-553bdaec","keepAlive":60,"cleanSession":false,"entityFormat":3,"base":"Buderus-WLW186i","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"analogSensors":2,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"Buderus","busTelegramsReceived":24222,"busReads":5642,"busWrites":0,"busIncompleteTelegrams":2,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"CUSTOM","locale":"en","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"phyType":0,"rxGPIO":17,"txGPIO":18,"dallasGPIO":0,"pbuttonGPIO":0,"ledGPIO":38,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":50,"webLogBuffer":27,"modbusEnabled":false,"forceHeatingOff":false},"devices":[{"type":"boiler","name":"CS*800i, Logatherm WLW*","deviceID":"0x08","productID":8,"brand":"Buderus","version":"00.00","entities":164,"handlersReceived":"0xBF 0x14 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F","handlersFetched":"0xE6 0xEA 0x048D 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x04AE 0x04AF 0x02CC 0x04AA 0x04A7 0x02D6","handlersPending":"0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x048A 0x04A5","handlersIgnored":"0x023E 0x0240 0x0291 0x0292 0x0293 0x0294 0x029B 0x02A5 0x02F5 0x02CD 0x04A8 0x04A9 0x04AB 0x04AC 0x04CA 0x04CB 0x061E 0xD2 0xE2 0xF7 0xF6"},{"type":"thermostat","name":"Rego 3000, UI800, Logatherm WLW186i/WSW196i, Logamatic BC400","deviceID":"0x10","productID":253,"brand":"","version":"47.10","entities":61,"handlersReceived":"0x06 0x02BA 0x02BB 0x02BC 0x0267","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E","handlersPending":"0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x02F6 0x031B 0x031D 0x031E 0x0269","handlersIgnored":"0xBF 0x0507 0x0508 0x0509 0x050A 0x059D"},{"type":"mixer","name":"MM100","deviceID":"0x20","productID":160,"brand":"","version":"24.05","entities":7,"handlersReceived":"0x02D7","handlersFetched":"0x02CD","handlersIgnored":"0x02E1 0x02EB 0x0255 0xE6 0x0291 0xBF"},{"type":"gateway","name":"WiFi module","deviceID":"0x48","productID":252,"brand":"Buderus","version":"08.02","entities":0,"handlersIgnored":"0xF9 0x2040 0x0495 0x0494 0x04AF 0x04AE"},{"type":"analogsensor","name":"analogsensor","entities":2}]}
```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

Devices thermostat --> BC400 should show these entities.

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

MichaelDvP, 2024-10-14:
Does the settings from Rego3000 thermostat (id 172) also work for you? 
See #1735 and the custom entities used there.

suaveolent, 2024-10-14:
No it does not seem to work.

```
ems-esp:$ read 10 269
000+00:05:07.876 N 6: [emsesp] thermostat(0x10) -W-> me(0x0B), RC300Holiday1(0x0269), data: <empty>
```

Here are the devices.
```
These EMS devices are currently active:

boiler: Buderus CS*800i, Logatherm WLW* (DeviceID:0x08, ProductID:8, Version:00.00)
 Received telegram type IDs: 0xBF 0xD1 0xE3 0xE4 0xE5 0xE9 0x494 0x495 0x48F 
 Fetched telegram type IDs: 0x14 0x16 0x33 0x26 0xE6 0xEA 0x28 0x48D 0x48A 0x4A2 0x485 0x486 0x492 0x488 0x484 0x48B 0x491 0x499 0x49C 0x49D 0x4AE 0x4AF 0x2CC 0x4A5 0x4AA 0x4A7 0x2D6 
 Pending telegram type IDs: 0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A 
 Ignored telegram type IDs: 0x23E 0x240 0x291 0x292 0x293 0x294 0x29B 0x2A5 0x2F5 0x2CD 0x4A8 0x4A9 0x4AB 0x4AC 0x4CA 0x4CB 0xD2 0xE2 0x61E 

thermostat: Rego 3000, UI800, Logatherm WLW186i/WSW196i, Logamatic BC400 (DeviceID:0x10, ProductID:253, Version:47.10)
 Received telegram type IDs: 0x06 0x2BA 0x2BB 0x2BC 0x267 
 Fetched telegram type IDs: 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2CC 0x467 0x291 0x292 0x293 0x294 0x2F5 0x2F6 0x31B 0x23A 0x240 0xBB 0x23E 0x269 
 Pending telegram type IDs: 0xA3 0xA2 0x12 0x13 0x2A6 0x2B0 0x29C 0x472 0x2A7 0x2B1 0x29D 0x473 0x2A8 0x2B2 0x29E 0x474 0x2A9 0x2BD 0x2B3 0x29F 0x475 0x2AA 0x2BE 0x2B4 0x2A0 0x476 0x2AB 0x2BF 0x2B5 0x2A1 0x477 0x2AC 0x2C0 0x2B6 0x2A2 0x478 0x2CE 0x468 0x2D0 0x469 0x2D2 0x46A 0x31D 0x31E 
 Ignored telegram type IDs: 0x507 0x508 0x509 0x50A 0x59D 0xBF 

mixer: MM100 (DeviceID:0x20, ProductID:160, Version:24.05)
 Received telegram type IDs: 0x2D7 
 Fetched telegram type IDs: 0x2CD 
 Pending telegram type IDs: 
 Ignored telegram type IDs: 0xE6 0x255 0x291 0x2E1 0x2EB 

gateway: Buderus WiFi module (DeviceID:0x48, ProductID:252, Version:08.02)
```

suaveolent, 2024-10-14:
For holiday mode it seems to be these telegrams:

Enable from 14.10.2024 | 00:30 --> 21.10.2024 | 01:45, 
```
000+00:19:43.634 T 1568: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0165), data: 00 00 00 00 00 00 00 00
000+00:19:43.835 T 1569: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0165), data: 00 00 00 00 00 00 00 00 02 12
000+00:19:44.457 T 1572: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0165), data: FF (offset 17)
000+00:19:44.663 T 1573: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0165), data: 02 12 02 00 01 FF FF FF FF FF 00 FF (offset 8)

Delete:
```
000+00:19:46.427 T 1579: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0165), data: 12 (offset 9)
```
or
```
000+00:19:47.596 T 1585: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x016D), data: 00
```

MichaelDvP, 2024-10-14:
Ok, i think the absent/away mode is 0x16E offset 0. Is this hc specific or global for the thermostat? 
What hc's do you have.
0x16D also looks like on/off switch. 
In 0x165 nothing looks like a date, can you log wihle changing the dates.

suaveolent, 2024-10-14:
The away mode is global for the thermostat / boiler (the BC400 is integrated in the heat pump unit).

I also did some digging regarding the holiday mode. It seems like the BC400 supports 8 different holiday entries.

Here is the log for 

18.10 2024 - 00:45 - 24.10.2024 - 02:30
Name: Holiday 1 (You can add a custom name for the holiday)
Target heating: hc1 (you can select from available options)
Target temperature:  18°C

```
000+02:34:58.251 T 18: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0B0B), data: 00 48 00 6F 00 6C 00 69 00 64 00 61 00 79 00 20 00 31 00 00 00 00 00 00 00
000+02:34:58.485 T 19: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0B0B), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 25)
000+02:34:58.718 T 20: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0B0B), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 50)
000+02:34:59.004 T 21: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0B0B), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 75)
000+02:34:59.706 T 22: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0165), data: 18 0A 12 03 18 0A 18 0A
000+02:35:00.146 T 23: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x016D), data: 01
000+02:35:00.343 T 24: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0165), data: 18 0A 12 03 18 0A 18 0A
```

Here for:

19.10.2024 -- 05:45 -- 28.10.2024 -- 04:00
Target temperature:  18°C
Name: No custom name
Target heating: hc1 (you can select from available options)

and then subsequent messages from changing the date to I think 01.11.2024

```
00+02:37:49.225 T 202: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0166), data: 18 0A 13 17 18 0A 1E 10
000+02:37:49.225 T 202: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0166), data: 18 0A 13 17 18 0A 1E 10
000+02:39:39.795 T 263: [emsesp] thermostat(0x10) -B-> all(0x00), ?(0x0166), data: 18 0A 13 17 18 0B 01 10
```

suaveolent, 2024-10-15:
I think I decoded the date setting.

### Custom Name
The holiday name is just plain ASCII encoding stored via `0x0B0B`. I still need to record the other 8 types.

### Date Encoding
The date is encoded using four bytes in the following order:

- Year: The first byte represents the year. For example, `18` in hexadecimal equals `24`.
- Month: The second byte represents the month. For example, `0A`  in hex equals `10`.
- Day: The third byte represents the day of the month. For example, `13` in hex equals `19`.
- Time Encoding: The time is encoded in one byte since you can only set the times in 15 minute intervals. For example:

`17` (hex) equals `23` in decimal, which splits into 5 hours and 45 minutes (i.e., 05:45).
`10` (hex) equals `16` in decimal, which splits into 4 hours and 0 minutes (i.e., 04:00).

Example:
For the log message `18 0A 13 17 18 0A 1E 10`:

`18 0A 13 17`: Represents 19th October 2024, 05:45.
`18 0A 1E 10`: Represents 30th October 2024, 04:00.

MichaelDvP, 2024-10-15:
Thanks,
> Target heating: hc1 (you can select from available options)

What options are available? Please read the following telegrams `read 10 167`, etc. to see if they contains the hc2, .. dates.
Also try to change the options to see where it writes to.

Sbried, 2024-10-15:
There are 8 holiday function periods that can be defined. They have the extended addresses 0x0065, 0x0066, .... 0x006C (1st byte +1 for EMS-ESP internal storage).
Bytes 1-4 define the start of the period as described above, Bytes 5-8 define the end of the period
Byte 9 is the operation mode (1=auto, 2=fix temp, 3=off, 4=eco)
Byte 10 is the fix temperature (if byte 9 =2)
Byte 11 is DHW operation mode (1=auto, 2 =off, 3=eco, 4=low, 5= high)
Byte 12 is thermal desinfection (0=no, -1=yes)

suaveolent, 2024-10-15:
Great, thanks.

MichaelDvP, 2024-10-15:
The broadcasting of the telegram is very fragmented, is the complete telegram shown in one message if you `read 10 165` in telenet? The data is mostly similar to RC300, but RC300 have only start/stop date, no time of day.
This is a lot of data, i think we should handle it like the timer as json in 3.7.1, see #1594

suaveolent, 2024-10-15:
Here is a full read of `165`:

Date: 01.01.2025 - 01:00 - 02.01.2025 - 02:00
Target Temperature: 19°C 

```
000+08:18:07.208 N 1: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0165), data: 19 01 01 04 19 01 02 08 02 13 02 00 01 FF FF FF FF FF 00 FF
```

suaveolent, 2024-10-15:
Custom names for these entries are stored from `0x0B0B`  to `0x0B12`, e.g.

```
ems-esp:$ read 10 0x0B0B
000+08:22:40.954 N 2: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0B0B), data: 00 48 00 6F 00 6C 00 69 00 64 00 61 00 79 00 20 00 31 00 00 00 00 00 00 00
000+08:22:43.023 N 3: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0B0B), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 25)
000+08:22:44.030 N 4: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0B0B), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 50)
000+08:22:45.895 N 5: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0B0B), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 75)
000+08:22:47.078 N 6: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0B0B), data: <empty> (offset 100)
ems-esp:$ read 10 0x0B0C
000+08:22:50.991 N 7: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0B0C), data: 00 48 00 6F 00 6C 00 69 00 64 00 61 00 79 00 20 00 32 00 00 00 00 00 00 00
000+08:22:52.305 N 8: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0B0C), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 25)
000+08:22:53.351 N 9: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0B0C), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 50)
000+08:22:55.266 N 10: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0B0C), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 75)
000+08:22:56.207 N 11: [emsesp] thermostat(0x10) -W-> me(0x0B), ?(0x0B0C), data: <empty> (offset 100)
```

proddy, 2024-10-17:
shall I add @MichaelDvP  ?

proddy, 2024-10-23:
This needs some work, and some custom entities with new naming, and test code. Saving for 3.8

andusvan, 2025-07-04:
Hello!
I'm not sure about the current status of this feature.
It seems somebody implemented it, it's postponed to Version 3.8 and now it's closed due to inactivity?
I'm waiting for this feature and want to know, if I can help here in any way...
Regards
 Sven

proddy, 2025-07-04:
I don't think it has been implemented. We'll need screenshots/exports of the custom entities that work, along with names to use and then we can code it.