emsesp/EMS-ESP32 #2093: hide from Dashboard vs remove from memory

proddy, 2024-10-14:
BTW: We should change the text here:
[image]
`hide from Dashboard` is wrong now, Rename to: `hide from Devices`? Or remove this flag?
And for `mark as favorite? Should we add `show on Dashboard` Or is it clear that only favorites are shown on Dashboard?

_Originally posted by @MichaelDvP in https://github.com/emsesp/EMS-ESP32/issues/2083#issuecomment-2411654757_

mattreim, 2024-10-14:
I'm in favor of renaming `hide from devices` because I use `remove from memory` for the entries without values. When combining, I would always have to include the unimportant ones in the device list.

mattreim, 2024-10-14:
https://github.com/mattreim/EMS-ESP32/blob/1b70b55989a94dac1d46d73894feb46e0a3f9778/interface/src/i18n/en/index.ts#L159

mattreim, 2024-10-14:
https://github.com/emsesp/EMS-ESP32/blob/0f877b67de09dae5b0d20e07c827eb9f5a8499f4/interface/src/i18n/en/index.ts#L150

proddy, 2024-10-15:
I think we go with the rename and re-think Remove from Devices/Remove Memory later. It needs some careful thinking.

closed 2025-03-23 by proddy