emsesp/EMS-ESP32 #2096: Czech translate

proddy, 2024-10-14:
### Discussed in https://github.com/emsesp/EMS-ESP32/discussions/2095

<div type='discussions-op-text'>

<sup>Originally posted by **skylk** October 14, 2024</sup>
Hi,
I am sending a translation into Czech.

[cz.zip](https://github.com/user-attachments/files/17367922/cz.zip)</div>

proddy, 2024-10-14:
let's see if it fits into 3.7.0

SoftVienka, 2024-10-14:
### Update slovak translate
HELP_INFORMATION_5: 'Pre pomoc a otázky, kontaktujte svojho inštalatéra.',

DOWNLOAD_SETTINGS_TEXT: 'Vytvorte zálohu svojej konfigurácie a nastavení',
UPLOAD_TEXT: 'Nahrajte nový súbor firmvéru (.bin) alebo súbor zálohy (.json)',

AUTO_SCROLL: 'Automatické rolovanie',
DASHBOARD: 'Informačný panel', // If it's long, then just 'Panel'
NO_DATA: 'Nie sú k dispozícii žiadne údaje',
DASHBOARD_1: 'Prispôsobte si svoj informačný panel tak, že označíte entity EMS ako Obľúbené pomocou modulu Prispôsobenia.'

proddy, 2024-10-15:
@skylk thanks for the CZ Web UI translations. If you could also take a look at the entity names and use the XLS in https://github.com/emsesp/EMS-ESP32/discussions/2095#discussioncomment-10944440 to assist, and attach it back here.

skylk, 2024-10-17:
Here are the Czech translations of entities.
[locale_translations.xls](https://github.com/user-attachments/files/17407688/locale_translations.xls)

MichaelDvP, 2024-10-23:
@skylk It's now in the dev. please check, if ok we can close this.

skylk, 2024-10-25:
I made some corrections to the text. So ok.
[locale_translations.xls](https://github.com/user-attachments/files/17523438/locale_translations.xls)

MichaelDvP, 2024-10-25:
@skylk  Thanks, i'll put it to dev and make a PR. 
@proddy please wait with release 3.7.0 for the cz optimization PR (next hour)

closed 2024-10-26 by proddy