emsesp/EMS-ESP32 #2099: EMS - Time to next maintenance (difference in EMS app and HA MQTT)

misa1515, 2024-10-15:
### DESCRIPTION

Difference showed in attached picture

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### TO REPRODUCE

Screenshot attached

### EXPECTED BEHAVIOUR

Please correct value in HA MQTT 

### SCREENSHOTS

[image]

[image]

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

MichaelDvP, 2024-10-23:
The time is stored in hours and also published to mqtt in hours. 
For better readable in webUI we convert to days + hours.

misa1515, 2024-10-23:
I found 250*24 = 6000 last weekend.
Thank you @MichaelDvP

proddy, 2024-10-23:
there's an example in the doc [here](https://emsesp.org/Home-Assistant/#solar-pump-working-hour) how to customize an entity in HA

closed 2024-10-23 by proddy