emsesp/EMS-ESP32 #2101: EMULATED Remote thermostat feture: RC20 for  RC30/RC35  Night mode switching to 0

majo551, 2024-10-15:
### DESCRIPTION

I assigned the HC2 to emulated RC20 using  Home Assistant(HA) Controls card for RC35 Thermostat MQTT entity.. All seemed to work ,  remote temperature maintained by HA script appeared in the Thermostat card, it even showed the right Day temperature. But when the master thermostat program switched the HC2 mode from day to night the selected temperature is set to 0 for HC2 and not to the night temperature . I can see the Night temperature for HC2    as a correct value in the  HA Controls for the entity, but the value for HC2 Selected Room Temperature contains 0 and the same is shown by the Thermostat card for HC2. Only the Room temperature is correct.
 When I try to dial manually the temperature in the thermostat card back to the night value from 0, in few seconds it is pulled back to 0. So I rather changed the HC2 assignment back to RC3x not to cause an issue with 0 temp set. But the value 0 stayed in the thermostat till morning change to the day mode .  
The HC1 assigned to RC3x works as expected. See comparison of the selected temp. changes in one day. 

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x ] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [x ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{"system":{"version":"3.7.0-dev.43","uptime":"001+18:12:15.129","uptimeSec":151935,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","freeMem":160,"maxAlloc":99,"freeCaps":110,"usedApp":1690,"freeApp":294,"partition":"app0","resetReason":"Software reset CPU / Software reset CPU","psram":false,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-73,"BSSID":"set","TxPowerSetting":78,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"time.google.com","tzLabel":"Europe/Bratislava"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":79386,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTConnects":4,"enabled":true,"clientID":"ems-esp","keepAlive":60,"cleanSession":true,"entityFormat":3,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":true},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"Buderus","busTelegramsReceived":228303,"busReads":63349,"busWrites":38,"busIncompleteTelegrams":647,"busReadsFailed":35,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"NODEMCU","locale":"en","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":false,"telnetEnabled":false,"maxWebLogBuffer":25,"webLogBuffer":25,"modbusEnabled":false,"forceHeatingOff":false},"devices":[{"type":"boiler","name":"GB*72, Trendline, Cerapur, Greenstar Si","deviceID":"0x08","productID":123,"brand":"Buderus","version":"04.08","entities":81,"handlersReceived":"0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04","handlersFetched":"0x14 0x16 0x33 0x26 0x28","handlersPending":"0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x2E 0x3B"},{"type":"thermostat","name":"RC35","deviceID":"0x10","productID":86,"brand":"","version":"21.08","entities":98,"handlersReceived":"0xA3 0x06 0xA2 0x12 0x13 0x42 0x4C 0x52 0x51 0x5C 0x5B 0x38 0x39","handlersFetched":"0x3E 0x3D 0x3F 0x48 0x47 0x49 0xA5 0x37","handlersPending":"0x53 0x56 0x5D 0x60"},{"type":"thermostat","name":"ES72, RC20","deviceID":"0x19","productID":113,"brand":"","version":"02.01","entities":1,"handlersReceived":"0xAF","handlersPending":"0xA3 0x06 0xA2 0x12 0x13"},{"type":"mixer","name":"MM10","deviceID":"0x21","productID":69,"brand":"","version":"02.01","entities":6,"handlersReceived":"0xAB","handlersFetched":"0xAA","handlersPending":"0xAC"},{"type":"solar","name":"SM10","deviceID":"0x30","productID":73,"brand":"","version":"02.00","entities":16,"handlersReceived":"0x97","handlersFetched":"0x96"},{"type":"controller","name":"BC25","deviceID":"0x09","productID":125,"brand":"","version":"01.05","entities":0,"handlersIgnored":"0x29"},{"type":"scheduler","name":"scheduler","entities":1}]}

```

### TO REPRODUCE

Activate the RC20 as HC2 Control Device from HA and set theHC2 Room temperature from remote . Then wait for nigh mode  auto switch temperatures. 

### EXPECTED BEHAVIOUR

See the correct night mode temperature in the  RC20 emulated Thermostat 

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**
[image]
[image]

MichaelDvP, 2024-10-15:
I guess reducemode is set to outdoor and the outdoortemp is higher that your setting for reducetemp ( off/reduce switch temperature), so the heating switches off as set by the parameters. Set reducemode to `reduce` to always use nightmode temperature or adapt the reducetemp.
The remote thermostat only sends theactual measured roomtemp and does not influence the settings of master thermostat.

majo551, 2024-10-15:
> I guess reducemode is set to outdoor and the outdoortemp is higher that your setting for reducetemp ( off/reduce switch temperature), so the heating switches off as set by the parameters. Set [...]

Well, the Thermostat Card should always show the Actual Selected Temperature. It does for the Physical RC3x (Night or Day selected temperature.  I expect the same behavior for RC20 Thermostat.  If I had a real physical, it would not change my dialed temperatures to any calculated/reduce or whatever temperature neither...   Consumer does not care about system internals and expects to see the values he/she set. Seeing 0  as night selected temperature is rather scary.

MichaelDvP, 2024-10-16:
EMS-ESP only shows the values from the bus. If you don't like it, address your complain to Bosch/Buderus.
For now you have two options:
1. read the RC35 manual and change the settings to fit your expectations
2. read the RC35 manual and change your expectations according to the current settings.

majo551, 2024-10-17:
The issue is in the logic applied in setting the virtual RX20 , which is in the EMSS-EPS32. It has nothing to do with manual for RC35.  It is not correct  if  changing the day to nigh t is shown for RC20 as 0 set .  This does not happen for RC35 physical device, it does only for software emulated RC20.  I believe there is some bug either in EMS-ESP or HA.

majo551, 2024-10-18:
I did the following steps to help finding the issue:

1.  read the manuals for RC35 and RC20
2. tested workaround in HA
3. observed the changes both in RC35/HA and EMS ESP

Findings.

- both manuals confirm my expectation. RC20 can set independently of the RC3x the night/day mode temperatures, day/night modes and even DHW requests.  And of course sets the room remote actual temperature
- EMS-EPS32 has only one entity to set the room actual remote temperature
- unless user select RC35 operating mode for the  circuit as SHUT DOWN (which I did not) the selected night temperature must be used for the system in combination of outside temperature (in my case I have outdoor mode with room temperature impact set)
- Heating circuits with a remote control, e.g. RC2x, RC20/RF: you set the room temperatures at
the remote control, not at the programming unit.
- If “shut-down mode” is selected as the night setback mode, the heating system will be
switched off at night. A nighttime room temperature **cannot** be se

Currently  earlier suggested workaround in HA forcing the correct temp will not work. Changing the requested temperature in night mode  will prevent changing the requested temperature in EMS-ESP32 regardless you set it in the emulated RC20 or EMS-ESP32 itself. There must be some bug or some other setting prevents setting night mode requested temperature (maybe connected to fact there is an option of shut-down -mode which does not require night temperature). In my case I have outdoor mode so the night temperature  must be accepted and expectation is the selected temp should reflect night temp or selected by user..

I also observed the emulated RC20 has only one entity in EMS-ESP32 - the room temperature, whereas physical device has more (including option to change day/nigh /dhw and requested temperatures and   time programs for switching day/night). Maybe it would be good to consider similar approach and feed these to RX35 in EMS-EPS32 same way...

Conclusion: **after the change of day mode to night mode,** it is not possible to change the requested HC2 temperature in EMS-EPS32 to anything but zero. 
 In my scenario I use HC2 operation managed by emulated  RC20.  So I think than  once the RC35 sends the changing mode sequence from day to night,  the selected  temperature should change to the  night temperature, but  this **cannot** be written to the RC35 entity register in EMS-ESP32  and it is kept as  0.  When the mode changes back to day, it is possible to change  the entity back to the day temperature

I believe  this might the issue, because I observed the HA traces and I can see that the actual  RC35 correctly sends the request to change the mode to night with a correct night temperature,  but EMS-EPS refused to change it and set it to 0.

**Changing the mode to night in RC35 for HC2 (either via RC35 or EMPS-EPS32 ) sets the hc2 selected room temperature entity to 0. This number cannot be changed until the  mode is set back to day.**

[image]
[image]
[image]
[image]
[image]

MichaelDvP, 2024-10-18:
The RC35 is the master thermostat and controls the heating. A RC20 only acts as roomtemperature sensor and remote. But all settings you do on a RC20 are only forwarded to the RC35. 
The RC20 emulation only provides the roomtemperature measurement, as emsesp can set all parameters direct on the RC35. It makes no sense to show any other parameter on the RC20, Also the roomtemp is redundant and only shown to have a feedback of working emulation, the RC30 also shows it.
You have to set the working parameters for the nightmode on rc35. Setting reducemode `reduce` will do what you want, sets the night temp and keeps heating on. Reducemode `room` switches the heating off (seltemp 0) until the the room sensor reports the room is at nighttemp, then starts heating to hold it. Reducemode `outdoor` checks reducetemp parameter and if outdoortemp is higher the heating is switched off, on lower outdoortemps the seltemp is set to nighttemp. 
All described in the RC35 manual, chapter 6.3.4.

majo551, 2024-10-18:
Never mind. You are filtering out the essence of my finding and do not accept something not working.  I spent time analyzing potential issue and root cause.  If not useful , the my apology. I found myself a workaround  in HA, just wanted to help the code  be more robust and bug free. If this is not the mutual aim then I will close the ticket and let other complain if they find same issue as this discussion is waste of my time. Good luck

closed 2024-10-18 by majo551

proddy, 2024-10-18:
You can always do a PR and fix the code if you think it's broken. It's free software after all

proddy, 2024-10-18:
Also, looking at the EMS-ESP code I don't know why when changing to Night Mode on the RC35 in HC2 would change the 'selected room temperature' to 0. Does this also happen if you change to night mode directly on the thermostat?

MichaelDvP, 2024-10-18:
@proddy fyi:
seltemp 0 mean heating off. Here the RC35 manual describing the behaviour when switching to night mode:

[image]

majo551, 2024-10-18:
> Also, looking at the EMS-ESP code I don't know why when changing to Night Mode on the RC35 in HC2 would change the 'selected room temperature' to 0. Does this also happen if you change to night mode directly on the thermostat?

As explained in my test, you can try changing the **hc2 selected room temperature** directly in EMS, but ONLY when hc2 mode type is day. Once set to night you cannot change the  **hc2 selected room temperature anymore**, it defaults to 0.   Strange that the value cannot be change only in night mode.   Should this be RX35 feature/bug we still need to maintain the emulated behavior of the RC20 thermostat as the physical one. I understand that for simplicity it is used as external thermometer only but because of incomplete implementation on the side of the UI it fails to show correct night/day set temperatures. 
In physical device you still see Night temperature , not zero. This is my point. 
So if the hc2 selected room temperature at night should, from whatever strange reason, show 0 internally in RC35, we should still be able to see the correct night temp in RC20 thermostat not the zero. 
As a result of 0 set and cannot be changed, the thermostat card shows 0 at night and you cannot change it by changing the thermostat dial /number. 
Now imagine I want to increase the night temperature set. I go and change the value in RC20.  In emulated version you can do it only during day. At night it shows 0 and will be zero till day. How can I increase the desired temperature except of going to RC35?
  So this  emulation is useless from UI perspective and should not be present as HA entity if you do not allow full functionality and want to keep just as a helper thermometer ... This would be pity, because the idea of emulated RC20 is great an appreciated its introduction.  What I pointed out is the functionality is broken at night from UI perspective and usability (changing  set temperature at nigh). And internally , hope setting 0 does not really mean zero...
And I know the RC35  manual by heart, using it for more than 10 years. It does not say, that changing the mode to night implies RC 20 showing 0....

And to your question and comment:

> Does this also happen if you change to night mode directly on the thermostat? 

Yes, either manual or automatic change to night mode cause the 0 and blocking the entity for change

And for changing the code myself:  I trust the authors and expert in their work. If I could do it, I would not seek for your help.  I found a workaround in HA to show the correct temp by keeping the day mode and switching the set temp to desired night temp via actions... This does not remove the root cause though. 
Thanks

majo551, 2024-10-20:
Final comment for reference:  Left thermostat card,  I manually put together with effort, in HA shows correctly the target temperature as expected  in HA for Auto/Day mode and Night modes.  Current implementation shows 0 for night as target temperature.
[image]
[image]

MichaelDvP, 2026-02-07:
@wardwygaerts v3.8.0 has removed climate modes that are not off/heat/auto. For 3.8.1 day mode hasbeen mapped to climate heat again, in 3.8.2.dev2 also thermostat night mode is mapped to climate-off. again.