emsesp/EMS-ESP32 #2103: Implementing a smart thermostat (using SAT)

proddy, 2024-10-15:
Thanks to @HeepH on Discord, I was made aware of this cool project [SAT](https://github.com/Alexwijn/SAT) (Smart Autotune Thermostat) which implements a self-learning heating curve as a Home Assistant component for OpenTherm. We plan to collaborate and extend this to support EMS devices via EMS-ESP.

This issue is to track progress.

proddy, 2024-10-30:
sergeantd, the main developer on SAT recommended I start looking at these files to build the EMS coordinator. There is no documentation unfortunately so need to traverse through a lot of undocumented Python code. If anyone wants to help, please shout!

https://github.com/Alexwijn/SAT/blob/develop/custom_components/sat/coordinator.py
https://github.com/Alexwijn/SAT/blob/develop/custom_components/sat/mqtt/__init__.py
https://github.com/Alexwijn/SAT/blob/develop/custom_components/sat/const.py
https://github.com/Alexwijn/SAT/blob/develop/custom_components/sat/config_flow.py
https://github.com/Alexwijn/SAT/tree/develop/custom_components/sat/translations

proddy, 2024-12-17:
First version is being tested. more details on the SAT GitHub site and discord channel.

olivluca, 2024-12-18:
I'm probably being thick but I cannot see any detail either in the changelog here or in the changelog of SAT (and I'm not keen in joining a discord channel just to ask stupid questions :sweat_smile:)

olivluca, 2024-12-18:
Thank you, so I gather it's in the mqtt-coordinators branch (mind me, I'm not currently using neither ems-esp32 nor sat, I'm just sitting on the fence, waiting for when it's possible to replace my on-off thermostat).

proddy, 2024-12-21:
Adding a link to the GitHub discussion on the mappings: https://github.com/emsesp/EMS-ESP32/discussions/2257

proddy, 2025-01-01:
The SAT just released their Smart Autotune Thermostat Home Assistant HACS component [version 4.0](https://github.com/Alexwijn/SAT/releases/4.0.0) with native integration for EMS-ESP over MQTT.

closed 2025-03-01 by proddy