emsesp/EMS-ESP32 #2104: Bosch Heat Pump error: No communication on EMS bus

bitsbeauty, 2024-10-16:
### DESCRIPTION

I have a new gateway E32 V2 installed since three weeks on the service jack. And it happened today the fifth time, that the heat pump (Bosch Compress 6800i AW) returns a EMS bus error (error code A01-1010, ---6233, A31-3101), and can't access the mixer MM100, and goes to a emergency mode (so my house gets cold). 
The gateway is connected through MQTT to home assistant, where the system reports normal. It also seems to read values from the bus, EMS gateway status is connected. So I can see the boiler temperatures going down, and the EMS TX Fails goes up. But I cant write on the EMS bus anymore (set target room temperature in home assistant or on the webinterface of the gateway returns a RX write error). The ems-esp system log returns a constant RX read error every 10-15sec.  

The only way to fix the issue, is to power down the heat pump internal unit, disconnect the esp-ems gateway from the service jack

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....
[emsesp_system_info (1).txt](https://github.com/user-attachments/files/17402171/emsesp_system_info.1.txt)
{"System Info":{"version":"3.6.5","uptime":"001+07:06:21.802","uptime (seconds)":111981,"platform":"ESP32","arduino":"Tasmota Arduino v2.0.14","sdk":"4.4.6.240105","free mem":162,"max alloc":103,"used app":1591,"free app":4105,"partition":"app0","reset reason":"Software reset CPU / Software reset CPU"},"Network Info":{"network":"WiFi","hostname":"ems-esp","RSSI":-75,"BSSID":"set","TxPower setting":78,"static ip config":false,"enable IPv6":true,"low bandwidth":false,"disable sleep":true,"enable MDNS":true,"enable CORS":false,"AP provision mode":"disconnected","AP security":"wpa2","AP ssid":"ems-esp"},"NTP Info":{"NTP status":"connected","enabled":true,"server":"time.google.com","tz label":"Europe/Berlin"},"OTA Info":{"enabled":true,"port":8266},"MQTT Info":{"MQTT status":"connected","MQTT publishes":47096,"MQTT queued":0,"MQTT publish fails":0,"MQTT connects":2,"enabled":true,"client id":"ems-esp","keep alive":60,"clean session":true,"entity format":1,"base":"ems-esp","discovery prefix":"homeassistant","discovery type":0,"nested format":1,"ha enabled":true,"mqtt qos":0,"mqtt retain":false,"publish time heartbeat":60,"publish time boiler":10,"publish time thermostat":10,"publish time solar":10,"publish time mixer":10,"publish time other":10,"publish time sensor":10,"publish single":false,"publish2command":false,"send response":false},"Syslog Info":{"enabled":false},"Sensor Info":{"temperature sensors":0,"temperature sensor reads":0,"temperature sensor fails":0,"analog sensors":0,"analog sensor reads":0,"analog sensor fails":0},"API Info":{"API calls":2,"API fails":0},"Bus Info":{"bus status":"connected","bus protocol":"HT3","bus telegrams received (rx)":414040,"bus reads (tx)":66307,"bus writes (tx)":227,"bus incomplete telegrams":20,"bus reads failed":3280,"bus writes failed":11,"bus rx line quality":100,"bus tx line quality":96},"Settings":{"board profile":"E32V2","locale":"en","tx mode":2,"ems bus id":11,"shower timer":false,"shower alert":false,"hide led":false,"notoken api":false,"readonly mode":false,"fahrenheit":false,"dallas parasite":false,"bool format":1,"bool dashboard":1,"enum format":1,"analog enabled":true,"telnet enabled":false,"max web log buffer":100,"web log buffer":100},"Devices":[{"type":"boiler","name":"CS6800i/WLW176i","device id":"0x08","product id":8,"version":"00.00","entities":152,"handlers received":"0xBF 0x14 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F","handlers fetched":"0xE6 0xEA 0x048D 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x04AE 0x04AF 0x02CC","handlers pending":"0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x048A","handlers ignored":"0x04C9 0x061E 0x0240 0x0291 0x0292 0x0293 0x0294 0x029B 0x02A5 0x02CD 0x04A8 0x04AC 0x02F5 0x04CA 0x04CB 0xD2 0xE2 0x049F 0x04A0 0x17 0xEF 0xFF06 0xF7 0xF6 0x04 0xFE06 0xFF07 0xFE07 0x01 0xEB 0x2D 0x02D6 0xC6 0xC7 0x0497 0x04B4"},{"type":"thermostat","name":"Rego 3000/UI800/WSW196i/BC400","device id":"0x10","product id":253,"version":"47.07","entities":62,"handlers received":"0x06 0x02BA 0x02BB 0x02BC 0x0267","handlers fetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E","handlers pending":"0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031D","handlers ignored":"0x0507 0x0508 0x0509 0x050A 0x059D 0xBF 0xF9 0xF7 0x1104"},{"type":"mixer","name":"MM100","device id":"0x20","product id":160,"version":"54.02","entities":4,"handlers received":"0x02D7","handlers pending":"0x02CC","handlers ignored":"0x02EB 0x02E1 0x0255 0x02CD 0xBF 0xF7 0xF6 0x01 0x03BD"},{"type":"gateway","name":"MX300","device id":"0x48","product id":252,"version":"08.02","entities":0,"handlers ignored":"0xF9 0x2040 0x0495 0x0494 0x04AF 0x04AE 0x0DE8 0x0DEC 0x0DEF 0x0CBD 0x0CBC 0xF7 0xF6 0x0CBF 0x01 0x0CC2 0x0CC3"},{"type":"connect","name":"Wireless sensor base","device id":"0x50","product id":252,"version":"08.02","entities":0,"handlers ignored":"0x1244 0x1245 0x1246 0x1247 0x1248 0x1249 0x124A 0x124B 0x124C 0x124D 0x124E 0x124F 0x1250 0x1251 0x1252 0x1253 0xF7 0xBF 0x2040 0x10A0 0xF6 0x1230 0x1232 0x1234 0x01 0x1236 0x1233 0x1238 0x1231 0x123A 0x1237 0x123C 0x1235 0x123E 0x123B 0x1239 0x123F 0x123D 0x0B3D 0x0B3E 0x0B3F 0x0B40 0x0B41 0x0B42"}]}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_
I have no idea how to reproduce the failure, how or why it happens.

### EXPECTED BEHAVIOUR
EMS bus without errors and a heat producing heat pump, which i can interact through home assistant

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### ADDITIONAL CONTEXT

_Add any other context about the issue here._
[log (1).txt](https://github.com/user-attachments/files/17402444/log.1.txt)

**(Please remember to close the issue when it has been addressed)**

proddy, 2024-10-17:
Sorry to hear things are not working, but well done with such a concise write-up! Do you have anything else connected to EMS-ESP (like iobroker/home assistant/modbus) that is sending commands to EMS-ESP, just to rule that out? 

A few Rx/Tx errors are normal, but those Tx retries of 10 seconds aren't. I think this is what is tripping up the EMS bus and causing the issues, EMS-ESP is flooding the bus. @MichaelDvP, do you have any idea what could be causing all the HPmode writes?

MichaelDvP, 2024-10-17:
The log shows only read requests to the thermostat not answered. Also boiler goes to safe state, so it's not getting commands from thermostat. Seems to me the bus connection to thermostat is broken, so thermostat can not reach any device and shows error not reaching mixer etc. A full og should show the devices connected and communication from the other devices. 
Normally thermostat is powered by the bus, maybe here this is different. I think a intense look at all cables and cable connections should be done.
For analyzing log it would be better to use the latest dev.

bitsbeauty, 2024-10-21:
Hi @proddy hi @MichaelDvP,
thanks for the fast reply. This is a side project, i can unforatunately only react occasionally.

@proddy: As I wrote in the description, I have HomeAssistant connected through MQTT Mosquito Broker, as described in the manual. And i bought the EMS-ESP to react to energy from the PV system, so yes, I sending commands:

- Mainly I want to use unused PV energy before its exported. And i manually adjustet the temperature from the virtual thermostat in HA. But when i look at the changes and ahen the error appeared, it never is in direct relation.
-And I have one automation to change the heating type from the boiler between "floor" and "radiator". I just have one heating circle, but radiators and floor heating installed. The heat pump dont know about it, but there are time during the day, that all radiators are off, and only the floor heating in the living room needs energy, so the flow can be much lower, to safe energy.

But both are very unfrequent changes. When I look the changes in hA lookbook up, it a few chages during the day - nothing frequent. I attach a statistic to see the corelation:
[image]

bitsbeauty, 2024-10-21:
@MichaelDvP 

> A full og should show the devices connected and communication from the other devices.
Sorry too knew. How can I download a full log? Do I have to select "ALL" and thats it?

> Seems to me the bus connection to thermostat is broken, so thermostat can not reach any device and shows error not reaching mixer etc.
the heat pump is installed since Feb 2024 and never reportet an errror. Since I have ems-esp installed, is almost every second or third day. Yesterday again. So for me it does seems to be in correlation with the ems-esp.

> For analyzing log it would be better to use the latest dev.
I can do, but because its a side project I'm hesitating. Is it possibel to downgade, when i run in to trouble? And what excatly would you need? Just a full Log?

MichaelDvP, 2024-10-21:
> Do I have to select "ALL" and thats it?

Yes, just select loglevel ALL, apply change, wait afew minutes and download this log.
>  since Feb 2024 and never reportet an errror

Maybe something like in #1953.
> Is it possibel to downgade, when i run in to trouble?

Yes.

mroeckl, 2024-10-24:
Hi, I had exactly the same issue with version 3.6.5.
I tried all variants of _TxMode_ and _BUS-ID_ but the issue always occurred again.
On [haustechnikdialog.de](https://www.haustechnikdialog.de/Forum/t/270919/Optimierung-WP-Einstellungen-Buderus-WLW186i?page=23) someone, who had the same issue, solved it by updating to 3.7.0 and setting _TxModus="EMS"_ and _BUS-ID="Modem (0x0D)"_.
I've updated to 3.7.0-dev.39 but kept _TxModus="EMS+"_ and _BUS-ID="Terminal (0x0A)"_ a month ago and, so far, the issue has not occurred again. But I've recognized that ems-esp magically restarts about once per week (not sure if this is somehow related).

bitsbeauty, 2024-10-24:
Hi @mroeckl ,
Thanks for your Experience. I also updated to 3.7.0 dev a week ago, and so far, the error did Not App war again. I also stayed with TxModus = EMS+.
Keep you updated when the error appears again.

bosch-buderus-wp, 2024-10-28:
Bad news! The issue occurred again today 😫

mroeckl, 2024-10-28:
Bad news! The issue occurred again today 😫

mroeckl, 2024-11-02:
Today, the issue occurred again with 3.7.1-dev.3. Afterwards I exported the logs with loglevel=ALL: [log-2.txt](https://github.com/user-attachments/files/17608384/log-2.txt)

@MichaelDvP, any idea what we can try to do? I am a bit frustrated because I really like ems-esp but restarting the heat pump every few days is no possible.

tp1de, 2024-11-03:
I got similar errors on mixer mm100 and whole system when EMS-Bus cabling was causing too many errors.
I would recommend to connect ems-esp by cable to the ems-bus connectors on the heatpump.

bitsbeauty, 2024-11-04:
@MichaelDvP 

> The log shows only read requests to the thermostat not answered. Also boiler goes to safe state, so it's not getting commands from thermostat. Seems to me the bus connection to thermostat is broken, [...]

I upgraded to teh latest dev and two days before: It happened again. Here are the full logs from yesterday and today (i just took two samples in case anything changed, but from what I can see, its very similar)
- [log_2024-11-02.txt](https://github.com/user-attachments/files/17616070/log_2024-11-02.txt)
- [log_2024-11-04.txt](https://github.com/user-attachments/files/17616074/log_2024-11-04.txt)

What I realised is, that the esp-ems uptime is reseting. When I start writing it was a liitle bit above 1 day. And now it resets to zero. Does that mean the ems-esp resets? Can you assume why that happens?
[image]

bitsbeauty, 2024-11-04:
And with the newest dev, its interesting that the rx failures are much less than before. 

Aha interesting. When I look on the rx failure over time, you can see that the ems-esp seems to reboot from time to time (its when the rx failure counter goes to 0). Why could that happen?
[image]

compare the amount of rx failures before and after the firmeware update on 22.10. at 14:00h :
[image]

bitsbeauty, 2024-11-04:
> > since Feb 2024 and never reportet an errror
> 
> Maybe something like in #1953.

Yes, sounds similar, even though its a differnet failure code

bitsbeauty, 2024-11-06:
@proddy @MichaelDvP 
Any idea when looking on the provided logs?

proddy, 2024-12-21:
sorry, forgot about this one!

mroeckl, 2025-01-13:
@bitsbeauty, 2 months ago I changed _TxModus_ to _"EMS"_ and I changed the location of my ems-esp device, which I stored inside the housing of the indoor unit. Now, it lies on top of the housing. Since I made this 2 changes, the error has not occurred again. And, hopefully, it will not occur again right after I write this message as it happened after my last message 🤞

closed 2025-03-22 by proddy