emsesp/EMS-ESP32 #2109: Wrong Uptime compressor pool

Sbried, 2024-10-18:
### DESCRIPTION

Pool Compressor uptime is wrong. But this is not a problem of EMS-ESP.
I saw some problems in the issues/discussion around the interpretation of 4 byte values.

The information sent from the boiler is:
`002+12:18:17.249 N 15: [emsesp] boiler(0x08) -W-> me(0x0B), UBAInformation(0x0495), data: 00 0E 59 CA 00 01 0A D6 00 0D 4E F4 FF FF FF 00 00 00 00 00 0`
`FF FF FF 00`  calculates to those ~49000 days, see screenshot.
Assuming it´s a signed 4 byte integer, it will calculate to -256 (two´s complement). This is also nonsens as an uptime can´t be negative in my opinion.
Maybe a solution is to show other than 0 only in case if the value of the signed 4 byte integer is positive?
Or if compressor pool starts is >0? Or if a PMxx is installed?
Or other ideas?

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{
    "system": {
        "version": "3.7.0-dev.43",
        "uptime": "002+12:22:28.559",
        "uptimeSec": 217348,
        "platform": "ESP32",
        "cpuType": "ESP32-D0WD-V3",
        "arduino": "ESP32 Arduino v2.0.17",
        "sdk": "v4.4.7-dirty",
        "freeMem": 157,
        "maxAlloc": 107,
        "freeCaps": 4174,
        "usedApp": 2333,
        "freeApp": 3363,
        "partition": "app0",
        "resetReason": "Software reset CPU / Software reset CPU",
        "psram": true,
        "psramSize": 4093,
        "freePsram": 4051,
        "model": "BBQKees Gateway Model EMS Gateway E32 V2 v2.0/2024071"
    },
    "network": {
        "network": "Ethernet",
        "hostname": "ems-esp",
        "TxPowerSetting": 0,
        "staticIP": false,
        "lowBandwidth": false,
        "disableSleep": true,
        "enableMDNS": true,
        "enableCORS": false,
        "APProvisionMode": "disconnected",
        "APSecurity": "wpa2",
        "APSSID": "ems-esp"
    },
    "ntp": {
        "NTPStatus": "connected",
        "enabled": true,
        "server": "fritz.box",
        "tzLabel": "Europe/Amsterdam"
    },
    "mqtt": {
        "MQTTStatus": "connected",
        "MQTTPublishes": 47336,
        "MQTTQueued": 0,
        "MQTTPublishFails": 0,
        "MQTTConnects": 1,
        "enabled": true,
        "clientID": "esp32-2cb865e4",
        "keepAlive": 60,
        "cleanSession": false,
        "entityFormat": 0,
        "base": "ems-esp",
        "discoveryPrefix": "homeassistant",
        "discoveryType": 0,
        "nestedFormat": 1,
        "haEnabled": true,
        "mqttQos": 0,
        "mqttRetain": false,
        "publishTimeHeartbeat": 60,
        "publishTimeBoiler": 10,
        "publishTimeThermostat": 10,
        "publishTimeSolar": 10,
        "publishTimeMixer": 10,
        "publishTimeWater": 10,
        "publishTimeOther": 10,
        "publishTimeSensor": 10,
        "publishSingle": false,
        "publish2command": false,
        "sendResponse": false
    },
    "syslog": {
        "enabled": false
    },
    "sensor": {
        "temperatureSensors": 0,
        "temperatureSensorReads": 0,
        "temperatureSensorFails": 0,
        "analogSensors": 0,
        "analogSensorReads": 0,
        "analogSensorFails": 0
    },
    "api": {
        "APICalls": 1449,
        "APIFails": 0
    },
    "bus": {
        "busStatus": "connected",
        "busProtocol": "Buderus",
        "busTelegramsReceived": 779705,
        "busReads": 188590,
        "busWrites": 1,
        "busIncompleteTelegrams": 5,
        "busReadsFailed": 0,
        "busWritesFailed": 0,
        "busRxLineQuality": 100,
        "busTxLineQuality": 100
    },
    "settings": {
        "boardProfile": "E32V2",
        "locale": "de",
        "txMode": 2,
        "emsBusID": 11,
        "showerTimer": false,
        "showerMinDuration": 180,
        "showerAlert": false,
        "hideLed": false,
        "noTokenApi": false,
        "readonlyMode": false,
        "fahrenheit": false,
        "dallasParasite": false,
        "boolFormat": 5,
        "boolDashboard": 1,
        "enumFormat": 1,
        "analogEnabled": true,
        "telnetEnabled": true,
        "maxWebLogBuffer": 50,
        "webLogBuffer": 14,
        "modbusEnabled": false,
        "forceHeatingOff": false
    },
    "devices": [
        {
            "type": "boiler",
            "name": "Enviline, Compress 6000AW, Hybrid 3000-7000iAW, SupraEco/Geo 5xx, WLW196i/WSW196i",
            "deviceID": "0x08",
            "productID": 172,
            "brand": "",
            "version": "02.02",
            "entities": 155,
            "handlersReceived": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
            "handlersFetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x02CC 0x04AA 0x02D6",
            "handlersPending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x04AE 0x04AF 0x04A5 0x04A7",
            "handlersIgnored": "0xF7 0x049F 0x04A0 0x04A3 0x0240 0x02E0 0x02EA 0x061E 0xE7 0x0497 0x2E 0x3B 0x17 0x36 0xF6 0x03 0x04 0x09 0x01"
        },
        {
            "type": "thermostat",
            "name": "RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410",
            "deviceID": "0x10",
            "productID": 158,
            "brand": "",
            "version": "73.04",
            "entities": 67,
            "handlersReceived": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BF 0x02C0 0x031D 0x0267",
            "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E 0x0269",
            "handlersPending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x02F6 0x031B 0x031E",
            "handlersIgnored": "0x0239 0x0508 0x0509 0x050A 0x0507 0xBF 0x0237 0xF7 0x01"
        },
        {
            "type": "gateway",
            "name": "WiFi module",
            "deviceID": "0x48",
            "productID": 252,
            "brand": "Buderus",
            "version": "08.02",
            "entities": 0,
            "handlersIgnored": "0xF9 0x2040 0x0495 0x0494 0x3B 0xE5 0xE9 0x0DE8 0x0DE9 0x0DEA 0x0DEC 0x0DED 0x0DEF 0x0DF0 0x0DF1 0x01"
        }
    ]
}

```

### TO REPRODUCE

n/a

### EXPECTED BEHAVIOUR

Show something that is logical if the information sent from the boiler makes no sense.

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

n/a

**(Please remember to close the issue when it has been addressed)**

MichaelDvP, 2024-10-18:
> Or other ideas?

we have expected a value of 0xFFFFFFFF for invaliid/not set value, but possible that Bosch uses 0xFFFFFF00. 
For the 16bit numbers they use 0x7F00 and 0x8000, indicated that sensor is not connected or short circuit, so last byte zero is not uncommon.

proddy, 2024-10-18:
or we make a special case for DeviceValueType::TIME (which is now an unsigned 4-byte int) to mark it invalid if its > 0xFF000000 ?

MichaelDvP, 2024-10-20:
We have all UINT24, UIN32, and TIME stored as 32 bit uint value and check for all notset values. The times for ems (3byte) and ems+(4byte) are all marked as TIME. But ems+ 4byte values are in seconds (shows with DIV60), so a 24bit limit is too low (half a year).
I've changes in my dev the 32bit notset to 0xFFFFFF00, @Sbried Could you check if this is all we need or do we have check somethng other: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

Sbried, 2024-10-21:
Tomorrow I´ll be back home and check

Sbried, 2024-10-22:
Perfekt. The pool compressor time does not show up anymore in the list of entities.

Sbried, 2024-10-22:
I close this now as it is 3.7.0-dev. 48

closed 2024-10-22 by Sbried