emsesp/EMS-ESP32 #2114: Restart command via REST-API does not have effect

KarstenBr, 2024-10-20:
### DESCRIPTION
Sending the restart command using the REST-API does not reset the EMS-ESP as expected. While there is no error response to the command itself, the board continues to operate without interruption. But the system log shows the following error entry:
[system]  Partition ‘’ not found
Other commands send via REST-API do function as expected

### REQUESTED INFORMATION
I checked the list of issues, discussions and docs, but have not found a related topic for this issue.

Hint: Both partitions - app0 and app1 - have SW 3.6.5 installed

{"System Info":{"version":"3.6.5","uptime":"000+00:13:41.480","uptime (seconds)":821,"platform":"ESP32-S3","arduino":"ESP32 Arduino v2.0.14","sdk":"v4.4.6-dirty","free mem":197,"max alloc":179,"used app":2139,"free app":5989,"partition":"app1","reset reason":"Software reset CPU / Software reset CPU"},"Network Info":{"network":"WiFi","hostname":"ems-esp","RSSI":-69,"BSSID":"set","TxPower setting":78,"static ip config":false,"enable IPv6":true,"low bandwidth":false,"disable sleep":false,"enable MDNS":true,"enable CORS":false,"AP provision mode":"disconnected","AP security":"wpa2","AP ssid":"ems-esp"},"NTP Info":{"NTP status":"connected","enabled":true,"server":"192.168.190.1","tz label":"Europe/Berlin"},"OTA Info":{"enabled":false,"port":8266},"MQTT Info":{"MQTT status":"connected","MQTT publishes":504,"MQTT queued":0,"MQTT publish fails":0,"MQTT connects":1,"enabled":true,"client id":"ems-esp","keep alive":60,"clean session":false,"entity format":1,"base":"ems-esp","discovery prefix":"homeassistant","discovery type":0,"nested format":2,"ha enabled":false,"mqtt qos":0,"mqtt retain":true,"publish time heartbeat":60,"publish time boiler":10,"publish time thermostat":10,"publish time solar":10,"publish time mixer":10,"publish time other":10,"publish time sensor":10,"publish single":false,"publish2command":false,"send response":false},"Syslog Info":{"enabled":false},"Sensor Info":{"temperature sensors":0,"temperature sensor reads":0,"temperature sensor fails":0},"API Info":{"API calls":5,"API fails":0},"Bus Info":{"bus status":"connected","bus protocol":"HT3","bus telegrams received (rx)":3215,"bus reads (tx)":733,"bus writes (tx)":0,"bus incomplete telegrams":0,"bus reads failed":0,"bus writes failed":0,"bus rx line quality":100,"bus tx line quality":100},"Settings":{"board profile":"S32S3","locale":"de","tx mode":2,"ems bus id":11,"shower timer":false,"shower alert":false,"hide led":true,"notoken api":true,"readonly mode":false,"fahrenheit":false,"dallas parasite":false,"bool format":1,"bool dashboard":1,"enum format":1,"analog enabled":false,"telnet enabled":true,"max web log buffer":50,"web log buffer":17},"Devices":[{"type":"boiler","name":"CS6800i/WLW176i","device id":"0x08","product id":8,"version":"00.00","entities":107,"handlers received":"0xBF 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F","handlers fetched":"0x14 0xE6 0xEA 0x048D 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x04AE 0x04AF 0x02CC","handlers pending":"0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x048A","handlers ignored":"0x0240 0x0291 0x0292 0x0293 0x0294 0x029B 0x02A5 0x04A8 0x04AC 0x02CD 0x02F5 0x04CA 0x04CB 0x04C9 0xE7 0x061E 0xD2 0xE2"},{"type":"thermostat","name":"Rego 3000/UI800/WSW196i/BC400","device id":"0x10","product id":253,"version":"47.07","entities":63,"handlers received":"0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267","handlers fetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E","handlers pending":"0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E","handlers ignored":"0x0507 0x0508 0x0509 0x050A 0x059D 0xBF"},{"type":"thermostat","name":"RT800","device id":"0x38","product id":3,"version":"21.03","entities":3,"handlers received":"0x042B 0x047B","handlers pending":"0x0273 0x0A6A","handlers ignored":"0xBF"},{"type":"mixer","name":"MM100","device id":"0x20","product id":160,"version":"24.05","entities":4,"handlers received":"0x02D7","handlers pending":"0x02CC","handlers ignored":"0x02E1 0x02EB 0xE6 0x0255 0x0291 0xBF"},{"type":"gateway","name":"MX300","device id":"0x48","product id":252,"version":"08.02","entities":0,"handlers ignored":"0xF9 0x2040"}]}

### TO REPRODUCE
Send http://<IP>/api/system/restart in case authentification is turned off as in my case or the respective command plus the -H ‘Authorization: Bearer …’ suffix in case it is turned on.

### EXPECTED BEHAVIOUR
After sending the command a reset is performed in the same manner as if the respective button of the Web-UI is being clicked.

proddy, 2024-10-20:
This was fixed in 3.7.0

KarstenBr, 2024-10-20:
Thx for the quick reply - indeed with 3.7.0-dev46 it works :-)

closed 2024-10-20 by KarstenBr