emsesp/EMS-ESP32 #2116: Call read commands from Web

proddy, 2024-10-20:
This has been discussed before and not implemented in favour of the `send` command, but I still think it's useful for novice users.

I've added a new setting in Application Settings called `Developer Mode` which we can use to expose more advanced/debug/developer type features. When enabled you'll get a new command icon in the Web Log where you can send write commands like:

[image]

Delta1977, 2024-10-21:
great feature.
i have another Idea:
It would be great if we could do this also via api or mqtt or is this also implemented?

proddy, 2024-10-21:
> great feature. i have another Idea: It would be great if we could do this also via api or mqtt or is this also implemented?

Yes, possible now. It's a new system command called `read` so you can call it via API and MQTT.  Like

```sh
curl -X POST \
    -H "Authorization: Bearer ${emsesp_token}" \
    -H "Content-Type: application/json" \
    -d '{"value":"8 2"}' \
    ${emsesp_url}/api/system/read
```

Delta1977, 2024-10-21:
Very cool. Thank you

Delta1977, 2024-10-22:
@proddy  i tested the MQTT Variant and the Output is logged in EMS-ESP Log, but no return via mqtt.
is a Mqtt-feedback planned? 
Maybe we can publish the complete Logs via MQTT in an logging Topic.

proddy, 2024-10-22:
> @proddy i tested the MQTT Variant and the Output is logged in EMS-ESP Log, but no return via mqtt. is a Mqtt-feedback planned? Maybe we can publish the complete Logs via MQTT in an logging Topic.

<IP>/settings/version

closed 2024-10-22 by proddy

proddy, 2024-10-22:
@Delta1977  log output isn't sent over MQTT, it wouldn't be very useful as each log line would be replaced every second and unless you're quick enough to catch it you'll miss half the logs. We use the SysLog protocol for that as it's designed for that. Alternatively, you could tap into the SSE event source, which is over HTTP.  

The idea of this feature is for novice users to type in a simple read command, then watch the single line output, copy & paste into GitHub for the developers to work on.

VlastiBroucek, 2024-11-03:
I am using 3.7.1-dev.2 and it is not working. Tried it in Edge, Chrome and Firefox on both Windows and in KALI Linux

proddy, 2024-11-03:
> I am using 3.7.1-dev.2 and it is not working. Tried it in Edge, Chrome and Firefox on both Windows and in KALI Linux

can you be more specific? what doesn't work? what happens? what do you see?

VlastiBroucek, 2024-11-03:
I put in e.g. 0x18 0x06 hit enter and nothing happens, no output in log at all.

proddy, 2024-11-03:
try just '18 6'

VlastiBroucek, 2024-11-03:
Oh, that works...