emsesp/EMS-ESP32 #2118: Menu is automatically changing when getting into Download / Upload

HansRemmerswaal, 2024-10-21:
I'm using the following firmware:

Version: 3.7.0-dev.46
Platform: ESP32-16MB+
Release: Development

I have the following minor issue...

**Issue:**
1. In EMS-ESP I click on **Status** and the Status menu is opened
2. For the Firmware Version I click on UPDATE to see if there are updates
3. The screen is opened with Download / Upload and what we see is that the highlighting in the menu is changed from **Status** to **Settings**
4. If I now click on the arrow back I end up in the **Settings** menu and not in the **Status** menu

**Requested fix:**
Do not change the menu from Status to Settings when clicking on the UPDATE button for Firmware Version

[image]

[image]

[image]

MichaelDvP, 2024-10-21:
This is a mix of old browser cache and new web, press ctrl-R.

proddy, 2024-10-21:
I'm aware of this, one it's a edge case that I hope nobody would notice :-) The code for the 'back' button looks at the URL and steps back, so `/settings/upload` go back to `/settings`. I'll have a think about how best to program this without adding too much extra code. This is the only place it happens

HansRemmerswaal, 2024-10-21:
This was in Chrome... And the CTR-R didn't solve it.
And  I have the same on my iPad and iPhone using Safari.

What I find strange is that as soon as I get in the Download/Upload the menu bottom left is already changing from Status to Settings

HansRemmerswaal, 2024-10-21:
Ah.... now I understand. from the status menu you call an option from the setting menu (Download/Upload). And that's why the back button gets confused. I would not spend to much time on this minor issue...

On this page I click on the UPDATE button...
http://192.168.0.239/status

... and that brings me to...
http://192.168.0.239/settings/upload

... then when I press the back I get in...
http://192.168.0.239/settings

... and on the settings menu we have the Download/Upload.

proddy, 2024-10-21:
Yes, it just traverses back up the tree. I can navigate to the last visited page, but then if you go from say www.google.com to http://192.168.0.239/settings/upload and hit the Back arrow in EMS-ESP, it would take you back to google.com. So also not a good solution. 

I'll start a new discussion on this - need Michael's input on some UI changes inspired by your comment.

mattreim, 2024-10-21:
> @MichaelDvP I was looking at the UI and proposed these changes. What do you think:
> 
> a - In the Settings page, remove the word 'Settings' from all the options. The word Settings is mentioned 9 times on that page. So 'Application Settings' goes to just 'Settings'. b - Same for the Status page, remove the word 'Status' c - Move the version from Status to Settings. Status is a refreshed page showing live data, but Version is static and doesn't need to be shown all the time. People just want a working system and not bothered by the build number.
> 
> votes?

I think all the ideas (a -c) are very good and there may be another one.

d - The icon already says download, the word download can also be removed in these cases.

[image]

[image]

proddy, 2024-10-22:
Thanks for the feedback, I'll make the changes today

proddy, 2024-10-22:
demo at https://demo.emsesp.org

mattreim, 2024-10-22:
It definitely looks better and is clearer. It's just a matter of getting used to checking for the firmware version and uploading them are now in different places.

Just a reminder.
[image]

proddy, 2024-10-22:
clicking on the firmware install button automatically downloads and updates, if that's easier. Most people wont update for months/years so it's really a power-user option for those using the latest dev versions.

I left the Settings and Status in the page titles on purpose, otherwise there would be no context when visiting those pages.

mattreim, 2024-10-22:
> clicking on the firmware install button automatically downloads and updates, if that's easier. Most people wont update for months/years so it's really a power-user option for those using the latest [...]

You're absolutely right. Hardly any user will update as often as we do ;)

HansRemmerswaal, 2024-10-22:
I also think it's more clean now

closed 2024-10-22 by proddy