emsesp/EMS-ESP32 #2130: Buderus MM100 issue

jkj66, 2024-10-22:
### One MM100 shows only 4 instead of 7 values. See screenshot!

_A clear and concise description of what the problem is or the change requested._

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._
[image]
[image]
[image]

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2024-10-22:
support info?

MichaelDvP, 2024-10-23:
Fixed since 3.7.0dev2 (Apr. 2024), see #1679

jkj66, 2024-10-23:
Should be fixed with #1679 but is not visible on my side for HC3. What is to do?

MichaelDvP, 2024-10-23:
> What is to do?

Post the actual support info (3.7.0 and uptime should be > 15 min to see all received telegrams for the devices). And make a full log (Status->system log->loglevel all) and attach it.

jkj66, 2024-10-23:
With 3.7.0 it works, thanks 👍

closed 2024-10-23 by jkj66